ratgdo/homekit-ratgdo #20: Add OTA support

thenewwazoo, 2023-12-06:
I think the ESP8266-based devices' flash is too small (confirm this), but the ESP32 should easily support OTA updates.

So, like, do that.

PaulWieland, 2023-12-06:
ESP8266 supports OTA. Both the mqtt and esphome firmwares have it built in.

dkerr64, 2023-12-09:
This is definitely needed.  It is impractical (impossible) to need to physically connect a USB wire to the Ratgdo to flash new firmware.

PaulM4031, 2023-12-10:
If I understand correctly you're saying the Arduino HomeKit SDK is too large to flash on 8266 SoCs? If so, would moving to ESP HomeKit SDK (https://github.com/espressif/esp-homekit-sdk) enable generating a smaller flash file?

thenewwazoo, 2023-12-10:
@dkerr64 I've updated the README to better set expectations. Hopefully that clarifies things a bit.

@PaulM4031 since I wrote this as a hobby project to scratch my own itch, I didn't perform any kind of rigorous systems engineering in order to design the software so... I don't know, I haven't tried. It's possible that the firmware would be smaller, but it might also be bigger due to including an RTOS. I've been uncomfortable with the state of the [Arduino-HomeKit-ESP8266](https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266) project's maintenance anyway, so I would certainly not be opposed if someone wanted to experimentally port this firmware to use that HAP implementation.

PaulM667, 2023-12-11:
@thenewwazoo Appreciate all the hard work!!

I'm with you on your original comment, I would go to ESP32 long before changing the HomeKit SDK. More i looked into ESP HomeKit SDK it would require refactoring a significant portion of the code.

sstoiana, 2023-12-17:
Apparently all of the ESP8266 modules on the ratgdo are 4m and not 1m flash. PR #50 implements OTA updates via the HTTP server. No code signing, authentication or authorization yet.

closed 2024-01-07 by jgstroud