ratgdo/homekit-ratgdo #26: Document the dev/build environment please

dkerr64, 2023-12-09:
I can help develop/debug if someone documents what the development / build environment needs to be.
Thank you

PaulM667, 2023-12-10:
Looks to be PlatformIO as there is an ini file in src. I've never used it but appears to be well documented https://platformio.org. Good extension for VS Code basically.

dkerr64, 2023-12-10:
Thank you, I was able to install platformio and can now compile the code.  Can't upload it to the ratgdo device yet, need to get that working on the linux virtual machine I'm using to build on.  But progress at least.

dkerr64, 2023-12-10:
And we're in business.  Here is what I did...

1. Clone the GitHub repository.  I forked the code so I can mess with it in my own GitHub account and cloned from there.  Be sure to use `--recurse-submodules` option (I didn't and had to track down why I was getting a build error for missing `secplus.h`).
2. Follow instructions at https://platformio.org/ to install the VS Code extension.  FYI that my cloned source / build is on a Ubuntu Linux virtual machine (hosted on proxmox), but my VS Code is on a iMac desktop.  I use the SSH extension to access the linux VM.  Everything here works fine in that environment (use the VS Code window that is SSH'd into the remote host to install the Platform IO extension).
3. Setup the extension, when you get to the point of creating new project / loading existing project, load existing from the platformio.ini file for this project.
4. You can now build, but cannot upload to the device.
5. If using linux Install the `99-platformio-udev.rules` file as noted [here](https://docs.platformio.org/en/stable/core/installation/udev-rules.html).
6. If you are using Ubuntu there is a conflict for /dev/ttyUSB0 solved with `sudo apt remove brltty` for reasons noted [here](https://stackoverflow.com/questions/70123431/why-would-ch341-uart-is-disconnected-from-ttyusb).
7. If your build system is in a virtual machine, be sure to passthrough the USB port from the host hypervisor to the VM.
8. Reboot so the USB / udev rules take effect.
9. Edit platformio.ini file to include the line `upload_port = /dev/ttyUSB0`.

Build and upload now work.

thenewwazoo, 2023-12-10:
Thanks for reporting back; I didn't have to do any of that for my setup (on macOS), so if you wanted to submit a PR to improve the documentation (even just with a pointer to your comment), that would be greatly appreciated.

closed 2023-12-10 by thenewwazoo