ratgdo/homekit-ratgdo #27: Infinite (maybe) loop when not connected to GDO

dkerr64, 2023-12-09:
I'm testing with just the board itself, not yet connected to garage door opener.  Successfully paired to HomeKit but if I then try and open the door it goes into a loop...

```
!!! [ 227921] RATGDO: transmit failed, will retry
>>> [ 227927] RATGDO: Collision detected, waiting to send packet
!!! [ 227933] RATGDO: transmit failed, will retry
>>> [ 227939] RATGDO: Collision detected, waiting to send packet
!!! [ 227944] RATGDO: transmit failed, will retry
>>> [ 227950] RATGDO: Collision detected, waiting to send packet
!!! [ 227956] RATGDO: transmit failed, will retry
>>> [ 227963] RATGDO: Collision detected, waiting to send packet
!!! [ 227969] RATGDO: transmit failed, will retry
>>> [ 227975] RATGDO: Collision detected, waiting to send packet
!!! [ 227980] RATGDO: transmit failed, will retry
>>> [ 227986] RATGDO: Collision detected, waiting to send packet
!!! [ 227992] RATGDO: transmit failed, will retry
>>> [ 227998] RATGDO: Collision detected, waiting to send packet
!!! [ 228004] RATGDO: transmit failed, will retry
>>> [ 228010] RATGDO: Collision detected, waiting to send packet
!!! [ 228016] RATGDO: transmit failed, will retry
>>> [ 228022] RATGDO: Collision detected, waiting to send packet
!!! [ 228027] RATGDO: transmit failed, will retry
>>> [ 228033] RATGDO: Collision detected, waiting to send packet
!!! [ 228039] RATGDO: transmit failed, will retry
>>> [ 228045] RATGDO: Collision detected, waiting to send packet
!!! [ 228051] RATGDO: transmit failed, will retry
>>> [ 228057] RATGDO: Collision detected, waiting to send packet
!!! [ 228063] RATGDO: transmit failed, will retry
>>> [ 228069] RATGDO: Collision detected, waiting to send packet
!!! [ 228075] RATGDO: transmit failed, will retry
>>> [ 228081] RATGDO: Collision detected, waiting to send packet
!!! [ 228087] RATGDO: transmit failed, will retry
>>> [ 228093] RATGDO: Collision detected, waiting to send packet
```

Eventually I think it crashes and reboots.  Connecting this to my GDO is not practical for testing purposes.

thenewwazoo, 2023-12-10:
There's no infinite loop here (there's no recursion possible), just the main loop spinning.

I'll have to have a think about whether this can be avoided. The RX pin is in input mode with the internal pull-up turned on; the TX pin is an output used by a software serial uart in inverted mode. The two pins are connected through a pair of mosfets, and the bus is asserted by driving it to ground. I think not having a garage door opener connected leaves nothing to hold the RX pin's mosfet's source high.

As a workaround, you can apply VCC to the comms pin.

thenewwazoo, 2023-12-10:
A similar issue is reported [here](https://github.com/ratgdo/esphome-ratgdo/issues/111) in the esphome-ratgdo project.

thenewwazoo, 2023-12-10:
[and here](https://github.com/ratgdo/esphome-ratgdo/issues/47)

dkerr64, 2023-12-11:
Humm, yes.  This may be unavoidable but is unfortunate (especially with lack of OTA). I can't really wire up to my GDO and do dev/test... would require probably a 50 foot run of cable either USB or the 3-wires.

PaulWieland, 2023-12-15:
Two suggestions:

1. Only output obstruction status on obst state changes
2. Only output if x number of low pulses have been detected (indicating obst is hooked up)

closed 2024-03-09 by jgstroud