ratgdo/homekit-ratgdo #30: HomeKit No Response

SausCode, 2023-12-13:
Hi!

Thanks for creating this. I installed successfully, and was able to open and close the garage door a few times using the Home app.

It seems like after an hour or so, HomeKit reports "No response". I've verified that I can ping the IP address of the flashed ESP.

Are there any logs or anything I can do to help investigate?

Also it appears the discord link might be expired.

donavanbecker, 2023-12-13:
Try rebooting from the web server

SausCode, 2023-12-13:
@donavanbecker That's true, and restarting the ESP works fine as well, I'm just trying to see if I can provide any useful data to fix this more reliably.

thenewwazoo, 2023-12-13:
This is, as its core, a dupe of #24, so I'm going to close it and put some thoughts there. "How do I see the serial logs from before something went wrong?" is not a problem I have an solid answer to yet.

closed 2023-12-13 by thenewwazoo