ratgdo/homekit-ratgdo #32: Add dry contact support

jdberry, 2023-12-13:
I have a door opener by Overhead Door, so I need support for the dry contact switches. I'm hoping that can be enabled on this project soon.

dkerr64, 2024-08-10:
@jgstroud what do we need to do to implement this?

seeginator, 2024-08-21:
Would also love to see this. Feature matrix for HomeKit firmware shows "+ - implementation is possible" for dry contact, but doesn't appear to be a Door Protocol option when configuring.

jgstroud, 2024-08-23:
Implementation is definitely possible.  I think the firmware side is fairly trivial to implement here, but the wiring is a little more challenging.  Take a look at the main ratgdo wiring page.  Neither David nor I have a door that requires this.  We could implement it, but won't really be able to test it at all.

jdberry, 2024-08-23:
I'd love to see this happen and would be happy to test. The wiring is pretty simple.

PaulWieland, 2024-08-23:
Testing is no problem.

sullivankyle, 2024-09-24:
I'm setting up my door now--dry contact and HomeKit preferred as I don't have another platform set up at my house.

If I wire for dry contact and add the reed switches for the limit switches, will HomeKit be able to control anything at all?

(Have not added reed limit switches yet) I was able to add the ratgdo to HomeKit and I see it, and it logs actions, but does not control the door at all.

Thoughts?

Screenshot of iOS home app garage controls:

[image]

jgstroud, 2024-09-24:
HomeKit firmware does not yet have support for dry contacts.  It should be pretty easy to add but no one has taken on the task yet.  It's the one remaining major missing feature.

sullivankyle, 2024-09-25:
Yea that's what I understood the more or less by instructions and this thread. Sounds like even though I see the gdo in apple home, no matter what i do with wiring, the app won't be able to get the garage door to do anything.

We shall see what comes first--one of the programmers here adds in the code to support homekit on dry contacts, or me buying the homeassistant computer thing or raspberry pi!

jgstroud, 2024-09-25:
Correct.  At this point even if you wire it all up, it won't do anything.  It's really not much code to write.  I would just do it but I've been extremely tied up over the last 4 months or so.  If I have some time I'll look at it.  Would be nice to close this gap. We also accept pull requests if you want to write it yourself.

tlhagan, 2024-10-07:
I am in the same boat, I have an Overhead Door that requires dry contact wiring and would prefer to use HomeKit directly. Looking forward to this being implemented. The code would probably be over my head but would be happy to help test.

seeginator, 2024-10-28:
Same. I actually have it all wired up for dry contact, and MQTT, but that proved to be fairly unstable for me with XCEL regularly shutting off power after the Marshall Fire a couple of years ago. I’d be happy to test out anything, but the coding is probably beyond me.

flsfx, 2024-10-29:
Adding my $.02 to the comments above, the Genie DO I have will need dry contact support to use HomeKit directly. 
The ratgdo is wired in but not appearing on Apple Home - I'm still working on that.

100ideas, 2024-12-13:
Dang it; I just spent the last few hours wiring up dry contact sensors for my garage door just so I could finally use ratgdo and of course hook it up to HomeKit. But of course this isn’t supported yet. Dang. Is anyone working on a fork to hack this support in? Should I start it? No idea what I’m doing but I’m game to try…

PaulWieland, 2024-12-13:
@100ideas https://github.com/ratgdo/homekit-ratgdo/pull/253

dkerr64, 2024-12-14:
PR #253 merged.  Still some cleanup to do (remove light accessory) but progress.

closed 2024-12-14 by dkerr64

donavanbecker, 2025-02-09:
Anyone willing to share their wiring? And which overhead door opener they have?

tlhagan, 2025-02-09:
The wiring diagrams in the dry contact section here should be pretty clear:
https://ratcloud.llc/pages/wiring

I have a legacy 850, manual [here](https://overheaddoor-production-assets.azureedge.net/assets/docs/default-source/owner's-manuals/legacy-owners-manual.pdf?sfvrsn=3230b530_4)

The white and red wires from the ratgdo go into BWC 3 and 4 respectively, and the white and striped/red wires coming from the wall button that were in BWC 3 and 4 go into the blue input pins, striped/red on pin 1 (top) and white on pin 2. The limit switches follow the dry contact wiring diagram.

I just left the obstruction sensors connected to the GDO and didn't connect them to the ratgdo because the voltages they use are different from what ratgdo expects. The obstruction sensing still works fine.

donavanbecker, 2025-02-09:
This is my GDO,
[image]

tlhagan, 2025-02-09:
So you should take the two wires in BWC and plug them into the blue pins in the ratgdo. The one in black will go in the top pin (R) and the one in white will go in right below (W). Then you plug the white wire coming out of the ratgdo into BWC white and the red wire into BWC black. You can leave the obstruction sensors (STB) alone.

Then you will need to wire the limit switches following the dry contact wiring diagram.

donavanbecker, 2025-02-10:
So I will need limit switches? I can plugin to the close and open?

PaulWieland, 2025-02-10:
If you have a ratgdo32, then you can wire the obstruction sensors. 
The open and close terminals should be checked to see what their function is. They might be limit switches, or they might be discrete controls for the door.

donavanbecker, 2025-02-10:
It's looks like this open and close terminals are just dummy slots and don't actually have a terminal behind it