ratgdo/homekit-ratgdo #36: Wifi times out

klipschthx, 2023-12-14:
I can't get the unit to connect to wifi. Ive tried it two different wireless networks, and it always times out. Any ideas on why that might be? Thx.

eric-ooi, 2023-12-14:
I'm seeing the same.  I can continuously ping the ESPHome firmware with low latency and no packet drops but from the same location I can't do the same with the HomeKit firmware.

eric-ooi, 2023-12-14:
I should mention that I get the same WiFi timeout message, but I can see it does connect to my network.  Once it gets an IP, doing the ping shows the latency and packet loss issues.

klipschthx, 2023-12-14:
Yeah I guess i should add that I know its not the hardware or the network because when i flash it with esphome it connects no problem.

jgstroud, 2023-12-15:
The issues with getting through the initial configuration may be fixed in 0.6.0 released today. I think the Wifi latency and packet loss is different than any issues related to getting connected for the first time.

klipschthx, 2023-12-15:
i installed 0.6.0 and same issue :(

klipschthx, 2023-12-15:
And I guess to be more specific, the issue is that the wifi times out upon connection. Another interesting/weird thing is that there are maybe 3-4 of the same SSIDs. Maybe they're different broadcast bands?

Avi0n, 2023-12-15:
> Another interesting/weird thing is that there are maybe 3-4 of the same SSIDs. Maybe they're different broadcast bands?

They're likely multiple APs broadcasting the same SSID (like mesh nodes). It seems other projects have sorted by signal strength, then removed all but the strongest from the list. I somewhat doubt this is related to wifi latency/packet loss though.

a1224, 2023-12-15:
FWIW, I am getting the timeout, but the device connects and I am able to continue to "Visit Device" by just pressing Skip.

eric-ooi, 2023-12-15:
Just flashed 0.6.0 and while I still get the initial timeout when trying to connect, it does in fact connect to my network (as before) but this time it appears that I can maintain a continuous ping without dropping packets and without high latency. 🤞

klipschthx, 2023-12-15:
you're totally right. it just times out on initial connect. good catch yall. hopefully it all works ok now.

eric-ooi, 2023-12-15:
Yep, I just wired mine up and all functions are quick to respond and work as described.  Awesome work, y'all!!

dmkw01, 2023-12-15:
I’d like to add my thanks as well. Thank you to Paul for the hardware and to all the others for contributing to the HomeKit coding.

eric-ooi, 2023-12-25:
Now seeing what others have mentioned regarding the "no response" message in the Home app after some time.  Almost as if the wifi is sleeping?  I can still reach the device via IP and force a reboot which brings it back up, but it eventually returns to "no response" after a while.

TheFrisco, 2024-01-16:
Despite the timeout message, wifi was stored and on the network for me. I searched for the IP address and it was the device that started with "esp" ... Didn't understand what the docu said, so I thought I write it down here.

kurktchiev, 2024-01-16:
I just installed 0.9 on a brand new board and I am getting Timeout when connecting to my 2.4ghz radio. Any suggestions on debugging?

blueyetisoftware-jason, 2024-02-07:
@kurktchiev I see the timeout as well after repeated attempts, but it is in fact connected. I can see it on the router and you can visit the device. I would just skip the config and on the next screen go to the logs, then reboot the device from there. Your Wifi will likely work fine.

jgstroud, 2024-03-19:
Fixed in #129