ratgdo/homekit-ratgdo #46: Add Security+ 1.0 support

Avi0n, 2023-12-17:


thenewwazoo, 2023-12-18:
I don't personally own a Security+ 1.0 garage door opener, so this is very much going to be a community effort. ;)

dkerr64, 2023-12-18:
How would I know a Security+ 1.0 GDO if I saw one?  I have a GDO that is ~20 year old might it be one?

bguidotti, 2023-12-18:
If it says Security+ and is pre-2011, I think it's a Security+ 1.0.

Avi0n, 2023-12-18:
@dkerr64 There's a table on [this page](https://paulwieland.github.io/ratgdo/01_features.html) that shows how to tell if you have Sec 1.0 or 2.0. But I should probably put that in the description of this issue too.

jgstroud, 2023-12-18:
I think at least some sec+1.0 systems work with Dry Contacts.  If that works, it would be much easier to implement. 

I saw reports in the myq channel of people with purple buttons being able to use dry contacts.

PaulWieland, 2023-12-18:
Dry contacts require two reed switches and more wiring to get door status. 

Sec+1 doors can report their status on the serial line just like sec +2. 

The problem is that it requires a specific myq wall control panel to do so (889lm or similar). I created an emulator that works, but it breaks the light/lock buttons on the 78lm and won’t work at all with the other “kind of smart” wall panels.

PaulWieland, 2023-12-18:
Try to copy my implementation and I will test it here. The protocol is very simple.

jdigangi, 2023-12-30:
My openers are  lift master 3800PLDs jack shaft opened made in 2012 I have the 398LM wall buttons.  What is the best solution ?  I have 2 RATGOs wired up with no success.  Is dry contacts my only alternative ?

is this still the suggested wiring ?

https://user-images.githubusercontent.com/4663918/276749741-fe82ea10-e8f4-41d6-872f-55eec88d2aab.png

pwojt, 2024-01-03:
I have a Purple Security +1.0 opener with a 935Cb remote. Not sure if that remote actually works for reporting status, but I'm willing to try it. Let me know how I can help to test out changes for adding this functionality. Just got my ratgdo today, so I can start testing/making changes.

edrikk, 2024-01-08:
Based on Paul’s comments, would anyone be able to merge in his code?  I honestly don’t know if it’s nearly a drop in or not, but would love to be able to use this fork.  :/

pwojt, 2024-01-08:
@PaulWieland Can you link me to the implementation? If I find some time one of these evenings I would be happy to try and add it here.

jgstroud, 2024-01-08:
https://github.com/ratgdo/mqtt-ratgdo/tree/2.5

edrikk, 2024-01-12:
Looks like the esphome version is almost there with sec plus v1.

https://github.com/ratgdo/esphome-ratgdo/commits/sec_plus_v1/

I do with there was some sort of consolidation/centralization of ratgdo.cpp/.h files so that they could basically serve as interface function providers.  Right now the way I see it the 3 firmware each have to hit the same bug and fix it separately, and the same for general optimizations / enhancements that would otherwise flow through "for free".

An example would be the mqtt firmware's changes to expect 2 "same messages" for sec v1 before actioning so as to avoid corrupt/bad messaging from e.g. opening the door.

rlowens, 2024-01-17:
> How do you wire for dry contacts

See https://paulwieland.github.io/ratgdo/03_wiring.html

sebastiangoldstein, 2024-01-18:
I would be happy to help test this. 
Don't have any coding skills but I have a RatGDO installed in a Security +1 with the requisite Liftmaster 889LMMC.
-Sebastian

pwojt, 2024-01-26:
I was hoping to get some time to code this up for 1.0 to test, but other things got in the way. Instead I set it up with [HomeBridge Ratgdo](https://github.com/hjdhjd/homebridge-ratgdo) plugin and standard MQTT implementation of firmware. So far it works well. Anyone that’s blocked while waiting for native HomeKit Security +1 implementation can get unblocked.

mitchjs, 2024-02-03:
I have implemented 1.0 into this code, more like wedged it in.... i added button on web page to select Secuity+ (1.0) or +2.0
the issues I'm having, is really 1.0 reliability.... since its when to send the message on the buss isn't as elegant as 2.0's clear to send handshaking, I use timing... ie window after last byte received.

I'm not ready to post code to anything official to my GitHub, but if someone wants to test... i  have firmware.bin file to share,
then just use existing http update method.

also right now, only really tested when you have a wall panel 888lm/889lm
the emulation, i need to review it (its in there but for sure needs testing)
i'm very persistent programmer and due to the unknown time to transmit, since we are on the buss with a wall panel that knows the rules
i see received status bytes get corrupted... i tried to filter out the corruption
good thing is the next staus bytes are only a few miliseconds away... as the polling by the 889LM is every 200ms

-Mitch

jgstroud, 2024-02-04:
That's fantastic. Have you looked through Paul's implementation in the mqtt firmware? I think the problems you mention may have been addressed there.

jgstroud, 2024-02-04:
Is it possible to just auto detect what protocol is running instead of adding the web switch?

mitchjs, 2024-02-04:
> That's fantastic. Have you looked through Paul's implementation in the mqtt firmware? I think the problems you mention may have been addressed there.

i did, thats where i started...  the MQTT firmware will have the same issues

to answer your second question, it could be, but its completely differernt protocol... at completely different baud rates
sercuirty+1.0 on the wire isnt secure at all, its just single bytes to get press and releases of buttons, and single byte status
where +2.0 for sure is a real protocol

let me know if you want the fw.bin file, its on my github, but i have to add you as collaborator.

jgstroud, 2024-02-04:
I don't have a sec+1 device, but Paul did say he would test any firmware that was done.

If the two versions run at different baud rates, then I wonder if just running an auto-baud detect would be sufficient to determine which protocol to run.

PaulWieland, 2024-02-04:
I’m afraid  Auto baud detection won’t work since the sec 1 gdos don’t broadcast anything until they are spoken to - and this depends on the specific model wall panel that’s installed. 

I would use ifdefs, generate two separate firmware files and use a selector in the esp web tools for which firmware to install.

mitchjs, 2024-02-04:
@PaulWieland , yea could do that... right now since really its so little code... its 1 firmware with option on webpage
:)

edrikk, 2024-02-04:
@mitchjs unless the firmware is broken it might make sense to post it or allow viewing the changes on your git, and even start a draft PR so that others can provide feedback.

I do like your approach of button vs a separate firmware.  One way or another the user has to know if they have a v1 or v2.  Putting it in the same firmware I think makes it a little easier because it allows a toggle vs a reflash if someone isn't sure/downloads the wrong firmware....

Really great job :)

brianost, 2024-02-11:
@mitchjs Thanks for working on this! I've got a Security+ 1 opener with a 888LM that I could test with. I'd also encourage you to post the code/make a PR so others can help with feedback.

mitchjs, 2024-02-11:
@brianost i did an invite... i will do pr... i haven't worked on this in a week... need to get back to it
i need to finish up and clean up my code
only have really 1 person testing... no issues reported

dkerr64, 2024-02-18:
This is queued up in PR #117 and #113 (they are identical once @mitchjs and I started collaborating on this).

brianost, 2024-02-18:
Awesome! I've been running the private firmware from @mitchjs this past week without issues.

mitchjs, 2024-02-18:
@brianost, the latest .bin is on my github... https://github.com/mitchjs/homekit-ratgdo

mkopelman, 2024-02-18:
> @brianost, the latest .bin is on my github... https://github.com/mitchjs/homekit-ratgdo

I would be interested in testing this but the link is not working for me. Thank you

brianost, 2024-02-18:
@mitchjs I tried the new firmware. I've got 3 garage openers, and 2 of them work great with the latest. 

However, one of them becomes really slow to respond. Too slow for homekit to operate properly, and when I refresh the web interface it takes up to 30 seconds to fill in the data. Rebooting doesn't help. Re-pairing homekit doesn't help. When I downgrade to the previous firmware it goes back to working normally. Then upgrading to the new firmware again gets it back to being slow. I'm not sure what would be different about that device, but it doesn't like something in the new firmware.

mkopelman, 2024-02-18:
@brianost i would still like to try it out, couldn’t hurt, right? Could you point me to a link for the latest .bin file?

brianost, 2024-02-18:
@mitchjs I narrowed it down a bit. Commit 293ec9a is the first one that exhibits the slowness problem for me. The previous commit f401aa3 works fine.

@mkopelman I think that mitchjs needs to invite you to his github for access to the files.

mitchjs, 2024-02-19:
sorry for late response.....  i was out :) @mkopelman  i added you to be able to access the firmware.bin
@brianost the only changes of late all have been web pages... when you testing, dont have browser accessing the device
see if that makes any difference.
the odd thing is you mention other 2 work great... so whats the difference? are they all using security 1.0?

brianost, 2024-02-19:
Homekit would typically show not responding even without the web connected. The garage opener/keypad should be identical across all three. Maybe there is just something corrupted on that one and I should do a full erase and re-flash. I could try tomorrow. If not that I might try swapping two of the ratgdos to see if the problem moves or stays with that opener.

mitchjs, 2024-02-19:
ok, let me know.. we'll figure it out somehow.... i only got 1 garage... my friend who got me started on this... has 2, 1 is 1.0 and the other is 2.0 and he says both working great

you gave the same panels on all 3? 889LMs?

mkopelman, 2024-02-19:
@mitchjs thank you for the invite. I was able to download and install the firmware to the ratgdo, but when I use the settings to switch to security+, it reboots but it is back to security2.0+. is there a way to force it to be sec1.0+?

dkerr64, 2024-02-19:
@mkopelman  That is odd. Can you capture the serial terminal log, from just before the reboot to just after please (you can delete all the non-readable garbage)

mkopelman, 2024-02-19:
@mitchjs my error, I was using safari. using chrome it all works as expected! I have HomeKit control of the garage.  I will post if I find any issues. thank you for this. much appreciated.

dkerr64, 2024-02-19:
> @mitchjs my error, I was using safari. using chrome it all works as expected! I have HomeKit control of the garage. I will post if I find any issues. thank you for this. much appreciated.

It should work on any browser, as long as not years old.

mkopelman, 2024-02-19:
> > @mitchjs my error, I was using safari. using chrome it all works as expected! I have HomeKit control of the garage. I will post if I find any issues. thank you for this. much appreciated.
> 
> It should work on any browser, as long as not years old.

it did now work with the latest safari running on an m2 MacBook running Ventura 13.6.3. not super surprising, I could not run the hard wired firmware update with safari either, had to use chrome as well. 
thanks again for this.

mitchjs, 2024-02-19:
@mkopelman what didn't actually work, the selection? the radio boxes didn't highlight? or when you pressed save
it didn't "stick" after the reboot.

mkopelman, 2024-02-19:
@mitchjs everything on the web page worked as expected, but the settings would not stick after hitting save. For both the password and the security option. Also, the user/password dialog would come up, I would enter the credentials and then it would come up again. I could see behind the dialog that the action was taking place and could cancel the 2nd dialog. Again, on chrome it all worked perfectly.

mitchjs, 2024-02-19:
so after pressing save.... did the the device seem to reboot?

mitchjs, 2024-02-19:
my firmware.bin is all up to date....

dkerr64, 2024-02-19:
@mitchjs I'd prefer you not keep merging my branch into yours.  I've moved on to work on different feature.  I would be best to keep your branch frozen at the update that made passwords optional.  We want that PR accepted into the mainline.

brianost, 2024-02-19:
@mitchjs so the happy, but somewhat unsatisfying conclusion is that it magically just works now. I was preparing to connect it to my laptop so I repositioned it a bit and made sure all the connections were solid. Then I flashed to new firmware one more time just to see if it was still showing the issue, and... it just worked fine. So, maybe a connection was flaky, or maybe there was something interfering a bit with the wifi where it was originally positioned. I think at this point I'll just let it be and see if it keeps working reliably.

mitchjs, 2024-02-19:
@dkerr64, i have no idea what github is doing, when i got to pull the changes you made... it tells me i have to push something... and of course i have no changes... i havent touched code....

nsayer, 2024-03-09:
I'm new to all of this, being a MyQ refugee, but I have a new ratdgo 2.53i hooked up to a sec+1 opener with an 888LM panel. At the moment, the panel itself seems to work, but the HK ratdgo firmware has all the wrong statuses on it and it doesn't work. I gather that I need to use the firmware being developed here for it to work.

mitchjs, 2024-03-09:
@nsayer, you are correct, its been waiting for merge for sometime
if you wan to test, i can add you to my fw.bin repository, you can update existing via the webpage
and then you would have sec1.0 support

nsayer, 2024-03-09:
Yes, please!

nsayer, 2024-03-09:
This mostly worked for me. During my testing my 888LM gave symptoms of the supercaps inside dying again, so I swapped it for the spare I got from LM under their replacement program.

With this all set up, the door operates correctly and shows correct status from HomeKit. This is true both with the web UI buttons and HomeKit itself. However, the GDO light functionality doesn't seem to work correctly. Telling the light to turn on toggles its current state, and turning it off does nothing. So you can work around that using on-offs to force it to do your bidding, but you have to be able to observe the GDO to determine its current state. That's certainly an acceptable fault to live with for now.

mitchjs, 2024-03-09:
@nsayer , oddly the firmware just got merged!
there is no real concept of light on or light off
the code emulates the pressing of the light button on the wall plate
(but it does look at status)
to keep it in sync

so the LM you got now should be 889LM, which is what i got, and light works perfectly from the webpage or the homekit app

nsayer, 2024-03-09:
Actually, I still have an 888LM - I did the freebie thing before the 889LM came out, but I also swapped out the supercaps in the original  one myself just because I could and kept the "new" one on the shelf.

I did order an 889LM yesterday, so I will probably swap it out when that arrives. I'll let you know if that changes the update behavior.

dkerr64, 2024-03-09:
> @nsayer , oddly the firmware just got merged! there is no real concept of light on or light off the code emulates the pressing of the light button on the wall plate (but it does look at status) to keep it in sync
> 
> so the LM you got now should be 889LM, which is what i got, and light works perfectly from the webpage or the homekit app
> 
> is light status accurate in web page?

I’m pleased it is now merged. 

We could changed the web page light on/off two buttons to a single light toggle button for sec 1.0… that would probably be better for users.  Or, the user interface could pay attention to current state of the light (assuming it is always correct) and modify behavior of the on/off buttons accordingly. 

Thoughts?

mitchjs, 2024-03-09:
@dkerr64 , no i dont think we need to change it, it works.... on turns it on, off turns it off
the feedback of the light on or off, is working for me... so code knows to turn on, if off, and off if on
without any logs from @nsayer  its too soon to do anything....

jgstroud, 2024-03-09:
Merged in #117

closed 2024-03-09 by jgstroud

nsayer, 2024-03-09:
I can produce logs. Looks like there's no way to do that OTA and I'd need to take my laptop up on a ladder, but that's not _entirely_ impossible.

mitchjs, 2024-03-09:
@nsayer, ha no, has to be via the usb... comport... you would use terminal software at 115200 baud 8,n,1
i would say watch web page, and turn on/off light from the wall plate 
and see if the status on web page changes

i used a long(ish) usb cable to laptop i have sitting in garage... then remote desktop in to it
still got it set up now :)

nsayer, 2024-03-09:
It's time for bed now, but I'll get something tomorrow. Is the release level of logging good enough (for now)?

[1 more reply]