ratgdo/homekit-ratgdo #59: Discord link on project page is broken

klische, 2023-12-22:
The discord link on the project home page  https://github.com/ratgdo/homekit-ratgdo/blob/2b590057b7081276f0b00bce471a3109112a5aca/README.md?plain=1#L77

Pointing to https://discord.com/channels/432663330281226270/1184710180563329115 is broken. It should be an invite to the Discord server, so tapping/clicking on it will launch the Discord app or website and take you directly to the project’s Discord. Invites should look like https://discord.gg/hTKzmak, hTKzmak, or https://discord.gg/cool-people

thenewwazoo, 2023-12-22:
Thanks! I was just trying to figure this out last night. Can you confirm that this link works?

https://discord.gg/homebridge-432663330281226270

eightnoneone, 2023-12-22:
Worked for me.

klische, 2023-12-28:
Works! Thanks! There are 3 ratgdo channels though. Which is which? https://discord.com/channels/432663330281226270/1184710180563329115

https://discord.com/channels/432663330281226270/1178451927550394368

The above also has 2 threads, one of which is actively being commented in. An explanation of what channel is for what topics would be useful, but we can mark this closed for now.

donavanbecker, 2023-12-28:
> https://discord.com/channels/432663330281226270/1184710180563329115

This one is for native HomeKit firmware as the description of the channel states:
[image]

commit fc43a47e (2023-12-31): Update discord info in readme

Fixes #59
closed 2023-12-31 by thenewwazoo