ratgdo/homekit-ratgdo #60: Warning beep control on or off

vehlajatt, 2023-12-23:
If you can add warning beep as Myq does from app that would be awesome

thenewwazoo, 2023-12-23:
For anyone reading, this requires TTC support.

rlowens, 2023-12-23:
For anyone lost like I was: TTC stands for Time To Close and seems to be a delayed-close feature? I'm not sure what support is needed for this (if it is a software problem or if only some GDOs support it or what).

Also discussed at https://github.com/ratgdo/esphome-ratgdo/pull/43

MustangMan, 2024-01-05:
This was one of the first things I noticed, the lack of delay and beep.  While I would like ot have this feature I am not sure it is critical EXCEPT for remote closing.  Which I have had to do a couple times.  If it is added I will update the device to have it.

jgstroud, 2024-01-06:
I actually really like the fact that it just closes and doesn't beep and flash.  Works just like a remote would.  Am I the only one that likes it better this way?

vehlajatt, 2024-01-06:
There should be option to turn on/off beep

donavanbecker, 2024-01-06:
> I actually really like the fact that it just closes and doesn't beep and flash.  Works just like a remote would.  Am I the only one that likes it better this way?

I agree. Honestly don't get the point.

bguidotti, 2024-01-06:
The beeping flashing was always so annoying to me.-Bruce Guidotti

OleSam85, 2024-01-06:
I used to think that it was just an annoyance, but a big benefit of HomeKit control is being able to operate the door remotely, and as someone with young children I can see why the beep and flash is used (mandated possibly?). I can probably learn to live without this, but having this would add some peace of mind.

donavanbecker, 2024-01-06:
@OleSam85 isn't that what the obstruction sensors are for though?

OleSam85, 2024-01-06:
@donavanbecker yes in theory, though I’ve seen kids playing on something that might straddle it, and some of the assorted neighborhood animals are either small enough to fit below the beam or large enough to straddle it. While I’d love to say we’re always in control, kids and pets are unstoppable agents of chaos. Parenting has broken my brain to the point where I’ve become very good at imagining a myriad of ways for my kids to maim themselves. As mentioned it’s livable without the feature, but any bit of extra insurance for tired/absent minded parents can be welcome. My personal opinion is it’s better left open (albeit as a lower priority issue) but I won’t be offended if anyone tells me I’m nuts and closes this.

donavanbecker, 2024-01-06:
I am not saying that it should never be added, I just don't think it is a priority. If someone wants to take it on an open a PR. PERFECT!

jgstroud, 2024-01-06:
The beeping just never made sense to me.  Even if you were straddling the obstruction beam, even a small child or animal would move out of the way when they hear the garage door coming down on them.  

Also, just to ease your mind a bit, your garage door opener by code is supposed to automatically open if it encounters resistance when closing.  You can safely test it. Close your door and stand near it and hold your arm out.  It should reverse with only a bit of pressure.

jgstroud, 2024-01-06:
I'm really not trying to be a prick about it.  I have kids myself.  I just never saw this as a feature that was doing anything to keep anyone safe.

vehlajatt, 2024-01-06:
Its helps a lot to some,if you dont like it then you don't need to explain.. we should be able to turn on/off for who likes ot or who dont.

thenewwazoo, 2024-01-08:
For the record, I personally do like the warning beeps and flashes, but I was always annoyed by the delay. My GDO is pretty dang quiet, and heavy machinery shouldn't be sneaky. My current thought/plan is to check if the TTC packet supports few- (or zero-) second delay values, and if it does, then send TTC instead of a door command so the control is nicely responsive but still gives warning to anyone around.

lttlrck, 2024-01-08:
> For the record, I personally do like the warning beeps and flashes, but I was always annoyed by the delay. My GDO is pretty dang quiet, and heavy machinery shouldn't be sneaky. My current [...]

This is out of scope of ratgdo project, it'd be neat if it were possible to use (e.g. Ring) motion sensor as an interlock on any switch/door controller. I haven't explored if this is possible. Maybe it is possible to tie a motion sensor into the obstruction sensors.

Why? My garage doors close 1 hour after sunset automatically and it does make me slightly nervous. This was with homebridge via MyQ so it had the flashing light/beep delay. My wife drove home just at that exact moment, opened the door, the automatic close was triggered, the light started flashing, but she kept on driving into the garage, the door started to descend, the door hit the hood, just as the obstruction sensor was broken... the doors are old single panel pivot style, no rollers, so the obstruction sensor is quite far _behind_ the obstruction, plus perspective does hide the movement of the door. So even with beep/flash/motion accidents can happy. I'm sure small kids, sleeping cats won't necessarily understand that light/noise means the door is closing.

thenewwazoo, 2024-01-09:
That’s possible by using the (not visible in Apple’s Home app) lock characteristic of the garage door accessory, but yes out of scope of this discussion ;)

vehlajatt, 2024-01-09:
People who don't want it stay out here it’s discussion for people who wants it..and i if it can happen ….
and if it will happen there should be toggle for on/off.

We don’t need ideas if its safe or not. Thank you

lttlrck, 2024-01-09:
> People who don't want it stay out here it’s discussion for people who wants it..and i if it can happen …. and if it will happen there should be toggle for on/off.
> 
> We don’t need ideas if its safe or not. Thank you

Sorry. I didn't suggest I didn't want it, that it was useless, or dangerous. If it was added I'd enable it.

thenewwazoo, 2024-01-09:
> People who don't want it stay out here it’s discussion for people who wants it..and i if it can happen ….
> and if it will happen there should be toggle for on/off.

With all due respect, I am interested in collecting many user opinions, and welcome all input. No decision has been made about whether it will be supported (likely) or configurable (unlikely).

HughesAndrewC, 2024-01-09:
Just to add another data point: I like the idea of this being implemented but would want it configurable. The fact that I'm able to close the door instantly is starting to grow on me. If implemented, I may try enabling it, but I'm betting I'd prefer to leave it off.

rasod, 2024-01-11:
The delay/beeping/flashing on MyQ are a by-product of US federal safety regulations 1211.14 for "Unattended operation of residential garage door operator".

TL:DR
5 second delay before closing, audible signal (45db at 10ft), visual alarm (360 lumens). Unattended operation is disabled after 2 consecutive unsuccessful closer attempts caused by "entrapment protection", requiring manual operation to be re-enabled.

So "technically" RatGDO (and many other "Smart" GDO accessories for that matter) aren't following safety regulations. Is it a little bit "nanny state" over protection? 🤷‍♂️

thenewwazoo, 2024-01-11:
Thank you for the citation! I’m not (we’re not) bound by that rule but it’s good to have definitive source.

As for the “nanny state” question, the counter to that is that safety regulations are often “written in blood”. Until we get a committee representative in this thread, I guess we’ll never know.

reh8388, 2024-01-11:
Is the TTC communications understood or does that still need to be sniffed and reverse engineered?

PaulWieland, 2024-01-12:
TTC is understood. 

What’s not understood is if there is a different door control command that will cause the beeper to beep. The wireless myq commands cause the gfi to beep. If someone has the hardware, time and energy to use a software defined radio to capture those commands from a wireless myq controller we could try and send the same command via hardwired serial.

reh8388, 2024-01-12:
I do have the MyQ HomeKit hub (819LMB) and a garage door monitor (829LM) which both trigger the flashing/beeping. If I recall both devices communicate to the opener on a ~900MHz frequency band. Is the assumption that the data packet there may be similar/same as the on the wire communication? 

I can try to capture the data from each device. Give me a few days to dig out the RTL and find some time to capture some data. Or if there is someone who knows for certain what they would be hunting for we can arrange lending hardware (829LM)

reh8388, 2024-01-14:
Reading up on the FCC documents for the 829LM (FCC ID: HBW7451) the unit hops frequency across 902-928MHz and has 50 different channels that it hops between. I don't have a ton of experience with the RTL-SDR and I doubt it has the bandwidth to capture the full message across hops. If anyone has any pointers though I am happy to give it a try.

doctorkb, 2024-02-07:
> aren't following safety regulations

This is a consumer product safety issue.  It isn't a requirement of random accessories (which ratgdo is), only of the opener itself if it offers unattended closing directly.

For what it's worth, the folks who promote Tailwind's GDO device use this as the reason to not get a ratgdo -- that it doesn't meet this antiquated safety standard.

jgstroud, 2024-03-20:
https://github.com/ratgdo/esphome-ratgdo/pull/43

dkerr64, 2024-04-02:
I have started to implement this and am doing it within the ratgdo firmware and NOT using the garage door opener's built-in delay function.  The team working on the PR that @jgstroud cited above have not fully reverse engineered how that works and so leaving the GDO's delay feature untouched seems wise.

Here is what I have implemented...

- A setting on web user interface that allows setting a delay from 0 to 60 seconds.  Default is zero (close immediately).
- On requesting door close,  closing will be delayed by X seconds, during which time the GDO's lights will flash on / off.  There is no beeping, but the light on/off has audible relay clicking noise (at least on my GDOs).
- The _current door state_ reported to the web page and HomeKit does not change during delay period... it still reports _opened_ until the end of the delay when it will change to report _closing_.
- If close is requested again during the delay period then the delay is canceled and the door closes immediately.  
- If open is requested during delay period then the delay is canceled and the door remains open, light will be restored to its state (on or off) immediately before close was requested.

Does the above sound right?

dkerr64, 2024-04-02:
> > The current door state reported to the web page and HomeKit does not change during delay period... it still reports opened until the end of the delay when it will change to report closing.
> 
> My only concern with this is that could appear as "unresponsive" to some systems like Alexa or HomeKit. I'm unsure of what their window is for believing that the device has heard their command.
> 
> It would also seem prudent to report that as "closing" unless it was somehow cancellable.

I've not checked yet what HomeKit reports when the myQ app is used to remotely close with a delay.  We should do the same as it does.

If you close from the Apple Home app it moves the door slide to closed position immediately,  it doesn't wait to get a "closing" state back from the door.

dkerr64, 2024-04-03:
The other situation we have to handle is if someone in the garage presses the wall control or another remote button during the delay period.  The garage door will close immediately, but the ratgdo delay loop will still be flashing light on/off.  The firmware needs to cancel delay loop if it detects that someone (something) else has initiated door close.

dkerr64, 2024-04-04:
I have support for this queued up in PR #145

jgstroud, 2024-04-30:
Fixed in #145

closed 2024-04-30 by jgstroud

donavanbecker, 2024-05-06:
> Fixed in #145

I tried this but got no beeping, is that expected?

jgstroud, 2024-05-07:
@donavanbecker yes, that's expected.