ratgdo/homekit-ratgdo #64: Offer a version of the firmware that has the motion detection compiled out

SausCode, 2023-12-26:
My gdo doesn’t have a motion sensor on it, so it is useless for me and just clutters up the home app. 

Would be great to have a way to opt out of this, either just ifdef’ing it out and offering two firmwares, or perhaps a config in the webpage that allows you to turn it off. 

There’s probably a similar use case to remove the light control, but I imagine most people’s openers have a light bulb built in.

SausCode, 2023-12-27:
Thinking about it more would likely be best done at the same time as WiFi config.

tabacco, 2024-01-02:
It seems reasonably easy to not register the motion detector accessory, and to expose a toggle in the web UI. The bit I don't know how to do is to store a preference flag somewhere, since that seems to be delegated to the homekit library right now (for pairing info), and there's no existing config support to piggyback off of.

PaulWieland, 2024-01-02:
This can't be added to the wifi config UI because ESP Web Tools specifically doesn't support any configuration options other than wifi credentials.

The Web UI can be configured to save preferences but requires the web service to store the configuration on the flash filesystem. This configuration then has to be parsed and read into memory at startup.

All of this adds a lot of complexity and overhead to the code base. It might be better to just offer two targets that are compiled using flags, one with and one without the motion sensor.

jgstroud, 2024-01-03:
I had considered just waiting until we see a motion message before advertising the motion service, but wasn't sure if HomeKit would properly handle the new service getting added after the fact.  It would be worth experimenting with.  It seemed pretty unobtrusive having the sensor available, so I didn't spend much time on that.

reh8388, 2024-01-04:
When connecting the device to HomeKit it gives you the option to select which attributes to add so you could deselect the motion sensor during setup. Doesn’t seem like something that needs to be tackled with the FW. 

[image]

dkerr64, 2024-01-05:
Once we pull in #63 we will have opportunity to add configurable settings through web page.  Right now nothing is implemented in the proposed web page design for this, but I think it could be added quite easily.

thenewwazoo, 2024-01-05:
HomeKit handles adding services on the fly badly; it doesn’t work, or didn’t the last time I tried it. 

I feel very strongly that this firmware should take an opinionated stance and not permit configuration. If you want configurability, HomeBridge has a built-in responsive UI toolkit, and Home Assistant is even scriptable. Homekit devices should Just Work.

PaulWieland, 2024-01-05:
I think its fine to not have configurability. it adds a ton of complexity for little gain. I do think its worth trying to find an elegant solution particularly because the majority of people don't have a motion sensing wall panel.

My $0.02 - I think its worth exploring to see if the motion service can be added dyanmically once motion is detected. 
Worst case offering two firmwares using build flags isn't that difficult to do.

thenewwazoo, 2024-01-05:
I’m certain there’s got to be a way to make the wall panel emit statistics or respond to a message somehow, but I don’t know what it might be. If someone with a Chamberlain HomeKit adapter widget could sniff traffic it might reveal itself, but I don’t have one.

lttlrck, 2024-01-05:
I don't have motion but I didn't get a motion widget in homekit...

But you can split a tile, so if you have door+light+motion in one tile, split it and you'll get three. You can then hide them from the home view and summaries so they become hidden. In my case I hid the lights so my home view only has two one-touch door tiles vs. the default the tap and slide on the combined tile.

so I don't know if you have that sensor added whether it is also possible to suppress a motion notification. If you cannot, I could see that being a little annoying.

PaulWieland, 2024-01-05:
> I’m certain there’s got to be a way to make the wall panel emit statistics or respond to a message somehow

There is a possibility that the wall panel encodes its capabilities within the bootup/sync messages. One thing I could do is flash two factory wall control panels with the same client ID and rolling code counter and then capture their startup commands to see if there are any bits that are encoded differently when they boot.

thenewwazoo, 2024-01-05:
John let’s make sure we connect about that, since the espressif HomeKit sdk permits dynamic service addition to the accessory, and (iirc) the Arduino does not.

jgstroud, 2024-01-05:
@thenewwazoo sure, we can sync on that and you are correct that it doesn't seem to handle the dynamic add out of the box.  it would be likely be easier with the espressif sdk, but I'm not sure if and when that will be ready.  I also think it can work with the current sdk, and I'm also somewhat confident the current implementation can be made stable

jgstroud, 2024-01-05:
I have a version that can pseudo-dynamically add the service.  How it works is by default the motion service will be disabled.  Upon detection of a motion event, it sets a flag saying you have a motion sensor.  Then after a reboot, the motion sensor will be there from that point on.

donavanbecker, 2024-01-05:
Nice work @jgstroud!!

jgstroud, 2024-01-05:
@thenewwazoo please review PR https://github.com/ratgdo/homekit-ratgdo/pull/85 I think it resolves this issue and should be relatively portable to a new implementation.  However, I'm not planning on merging this until it's been reviewed and independently tested.

this PR WILL require that you re-pair your device.  I'm not 100% sure why, but I think it's because I had to reorder the services to simplify this.  I verified that after flashing and repairing my device, I did not have a motion sensor anymore.  After triggering my motion sensor, it show up in HK about a minute later.

closed 2024-01-27 by jgstroud