ratgdo/homekit-ratgdo #67: Multiple Doors, Multiple ratgdo Modules

YZFDonor, 2023-12-31:
I just received 2 ratgdo v2.5i boards and flashed them both to "HomeKit-ratgdo" and the seem to work great for my current needs... until they are both plugged in. They have the same HomeKit ID and seem to conflict with each other. Am I imagining this or can you not have two modules in the same home right now? I was having significant delays and unresponsiveness on the one that is hooked into my main door. The other one is just powered on, sitting on my desk. When I removed power from the "Left Door" that is sitting on my desk the "Right Door" became extremely responsive.

Did I miss something? can I change the HK ID somehow? I couldn't find an email address and the Discord link is busted.

donavanbecker, 2023-12-31:
Homekit Pair Code will always be the same and do not cause this issue. This sounds more like a network issue. Others have two setup working just fine.

thenewwazoo, 2023-12-31:
Please make sure you're running the latest version. This issue should have been fixed in 0.4.0.

closed 2024-01-03 by YZFDonor