ratgdo/homekit-ratgdo #68: Release Process Failing

donavanbecker, 2024-01-01:
@PaulWieland anyway you can set the permissions outlined in this answer so that we can get the releases automated?

https://stackoverflow.com/questions/75995802/resource-not-accessible-by-integration-github-action-fails-for-pushed-commit

donavanbecker, 2024-01-01:
@thenewwazoo Once @PaulWieland has changed the permissions you should be able to go tot he failed action and just restart the failed action.

thenewwazoo, 2024-01-01:
Re-running it didn't succeed. I haven't taken the time to diagnose yet, sorry.

PaulWieland, 2024-01-01:
Actually that is the global security setting and it is not changable as the option is disabled.
[image]

I think the permissions have to be set at a more granular level in the workflow configuration.

https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#permissions

donavanbecker, 2024-01-01:
Do you have this set?

https://docs.github.com/en/organizations/managing-organization-settings/disabling-or-limiting-github-actions-for-your-organization#setting-the-permissions-of-the-github_token-for-your-organization

donavanbecker, 2024-01-01:
I know as part of the Homebridge organization we have it set as an org and on the homebridge-config-ui-x for read and write and this action works:
[image]

donavanbecker, 2024-01-01:
@thenewwazoo wanna try #69

donavanbecker, 2024-01-01:
That didn't work @PaulWieland. So I think you will need to check these settings. I was able to get these releases to work on my fork.

donavanbecker, 2024-01-01:
@PaulWieland just tried another one with #70 and still same issues. You will need to set these at an organization level and then it should be available at a repo level.

donavanbecker, 2024-01-01:
#71 should resolve all of this.

closed 2024-01-01 by donavanbecker