ratgdo/homekit-ratgdo #77: Eero 2nd Gen issues

lttlrck, 2024-01-04:
Follow-up to my comment here, because this is potentially useful info 

https://github.com/ratgdo/homekit-ratgdo/issues/24#issuecomment-1877431070

There is a problem when connected to Eero (Gen 2) **satellites** - loss of connection, No Response, pings all over the place (up to 1000ms and drops). I moved the ratgdos next to the master Eero Pro AP, far away from the garage and other satellites, and all these issues vanished.

This is the case with 0.7 & I also built the firmware per https://github.com/ratgdo/homekit-ratgdo/issues/24#issuecomment-1875865540 - there was no improvement.

So, I dug out an ancient OpenWRT reflashed-TP-Link (irrelevant but it was a fun find) and installed it in my garage as an additional AP and now the ratgdos are rock solid, with steady 5ms ping times, no dropouts. It works perfectly.

For what it's worth: my wifi is very stable across the whole mesh, the satellites give me 300Mbps all day long with no issues on my phone, laptops etc. So it really seems like there is perhaps something in Eero mesh management that messes with the wifi stack.

I'll keep an eye out for any improvements in the firmware and give them ago if they might address this.

jgstroud, 2024-01-04:
have you tried the esphome firmware, and does it behave any different?  can you get a pcap of the ratgdo traffic?

lttlrck, 2024-01-04:
No sorry, I really should have done that. I was about to give up and try homebridge-ratgdo (my original plan). ESPHome didn't even cross my mind. I was so happy once it worked after hours of messing around, wondering if my lights/furnace were causing interference that once I got it going I put the cars back in. 

On the next firmware update I'll try to grab a PCAP. Capture off the wifi right? Should be able to do that.

reh8388, 2024-01-04:
I have a first gen Google WiFi mesh and just spent an hour battling it to get my ratgdo reconnected. Similar to the experience above, pings all over the place even with a satellite in the garage 6 ft from the device. Watching ping times and it looked like my unit had 3 ping times (6-10ms, 200-400ms, 600-800ms) that it would do 2-3 pings at a time interval then timeout for a few and then ping 2-3 times at a different one of the ping times. This lines up with 3 mesh nodes I have in the house. In an attempt to reduce hopping I unplugged the unit in the garage closest to the unit and my pings instantly dropped to the 6-10ms range and no more timeouts. Seems like the unit has a hard time when it can see multiple nodes of the same network and is constantly hopping between.

michael-pesce, 2024-01-05:
+1

I am having the same issue with a multi-AP Eero network and a ratgdo (Homekit flashed) that has constant dropouts and returns pings between 4ms to 6-700ms (on the occasions it doesn't timeout). It's unusable and totally unresponsive in Home. 

>  bytes from 10.22.22.119: icmp_seq=4115 ttl=255 time=193.440 ms
Request timeout for icmp_seq 4116
Request timeout for icmp_seq 4117
Request timeout for icmp_seq 4118
Request timeout for icmp_seq 4119
Request timeout for icmp_seq 4120
Request timeout for icmp_seq 4121
64 bytes from 10.22.22.119: icmp_seq=4122 ttl=255 time=208.960 ms
Request timeout for icmp_seq 4123
64 bytes from 10.22.22.119: icmp_seq=4124 ttl=255 time=10.606 ms
64 bytes from 10.22.22.119: icmp_seq=4125 ttl=255 time=8.210 ms
64 bytes from 10.22.22.119: icmp_seq=4126 ttl=255 time=188.177 ms
Request timeout for icmp_seq 4127
Request timeout for icmp_seq 4128
Request timeout for icmp_seq 4129
64 bytes from 10.22.22.119: icmp_seq=4130 ttl=255 time=220.682 ms
64 bytes from 10.22.22.119: icmp_seq=4131 ttl=255 time=77.867 ms
Request timeout for icmp_seq 4132
Request timeout for icmp_seq 4133
Request timeout for icmp_seq 4134
Request timeout for icmp_seq 4135
Request timeout for icmp_seq 4136
Request timeout for icmp_seq 4137

michael-pesce, 2024-01-05:
I last wrote code (Rails, lol) like 15 years ago so ... take this with appropriate caution. Simply trying to help. 

Downloaded the repo, installed VSCode, and monkeyed around enough to figure out how to build and monitor using PlatformIO. 

Here's the log showing disconnects (from the HomeKit class) that perfectly align (well, they seem to immediately precede) ping timeouts. You'll see me turning the light on and off a few times when it stayed connected long enough for a command to go through. 

Going to play with the code to learn more. What could go wrong!?

> --- Terminal on /dev/cu.wchusbserial210 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp8266_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
>>> [    178] RATGDO: Starting WiFi connecting in background
>>> [    186] RATGDO: Setting up pins
>>> [    189] RATGDO: enabling built-in LED
>>> [    193] RATGDO: Setting up comms for secplus2.0 protocol
>>> [    319] RATGDO: id_code doesn't exist. creating...
>>> [    327] RATGDO: writing 00 to file id_code
>>> [    335] RATGDO: id code not found
>>> [    342] RATGDO: writing 9A3539 to file id_code
>>> [    350] RATGDO: id code 9A3539
>>> [    355] RATGDO: rolling doesn't exist. creating...
>>> [    363] RATGDO: writing 00 to file rolling
>>> [    371] RATGDO: rolling code 00
>>> [    375] RATGDO: Syncing rolling code counter after reboot...
>>> [    382] RATGDO: ENCODING 00000000 00000004009A3539 0000008B
>>> [    412] RATGDO: writing 01 to file rolling
>>> [    522] RATGDO: ENCODING 00000001 00000000009A3539 00000080
>>> [    555] RATGDO: writing 02 to file rolling
>>> [    564] HomeKit: Update the CPU to run at 160MHz
>>> [    569] HomeKit: Starting server
>>> [    572] HomeKit: Using existing accessory ID: 86:4A:DE:36:3B:56
>>> [    579] HomeKit: Found admin pairing with DCD2F97A-FE1B-431A-91AC-54F6F2DBFCBE, disabling pair setup
>>> [    588] HomeKit: Configuring MDNS
>>> [    591] HomeKit: Init server over
>>> [    595] RATGDO: HTTP server started
>>> [    599] RATGDO: RATGDO setup completed
>>> [    603] RATGDO: Starting RATGDO Homekit version 0.8.0
>>> [    608] RATGDO: SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
>>> [    681] RATGDO: reader completed packet
>>> [    686] RATGDO: DECODED  00002B36 00000080A600A006 02E07281
>>> [    692] RATGDO: PACKET(0xA006 @ 0x2B36) Status - Status: [DoorState Closed, Parity 0x7, Obs 1, Lock 0, Light 1]
>>> [    702] RATGDO: activating door
>>> [    706] RATGDO: tgt 1 curr 1
>>> [    709] RATGDO: Light Status On
>>> [   2762] HomeKit: WiFi connected, ip: 10.22.22.119, mask: 255.255.255.0, gw: 10.22.22.1
>>> [   2771] HomeKit: Configuring MDNS
>>> [   2776] HomeKit: MDNS begin: Garage Door 18FD9E, IP: 10.22.22.119
>>> [  27095] HomeKit: Got new client: local 10.22.22.119:5556, remote 10.22.22.133:57946
>>> [  27106] HomeKit: [Client 1073692844] Pair Verify Step 1/2
>>> [  27597] HomeKit: Free heap: 27096
>>> [  27696] HomeKit: [Client 1073692844] Pair Verify Step 2/2
!!! [  27704] HomeKit: [Client 1073692844] No pairing for 64AF85D5-CA67-4BD3-9CC8-EF55497F79FE found
>>> [  27714] HomeKit: Free heap: 27728
>>> [  27845] HomeKit: [Client 1073692844] Disconnected!
>>> [  27851] HomeKit: [Client 1073692844] Closing client connection
>>> [  28099] HomeKit: Got new client: local 10.22.22.119:5556, remote 10.22.22.133:57947
>>> [  28109] HomeKit: [Client 1073692844] Pair Verify Step 1/2
>>> [  28600] HomeKit: Free heap: 27800
>>> [  50381] HomeKit: [Client 1073692844] Disconnected!
>>> [  50386] HomeKit: [Client 1073692844] Closing client connection
>>> [  66887] HomeKit: Got new client: local 10.22.22.119:5556, remote 10.22.22.133:57948
>>> [  66897] HomeKit: [Client 1073692844] Pair Verify Step 1/2
>>> [  67388] HomeKit: Free heap: 27744
>>> [  67473] HomeKit: [Client 1073692844] Pair Verify Step 2/2
!!! [  67480] HomeKit: [Client 1073692844] No pairing for 64AF85D5-CA67-4BD3-9CC8-EF55497F79FE found
>>> [  67491] HomeKit: Free heap: 27856
>>> [  67603] HomeKit: [Client 1073692844] Disconnected!
>>> [  67609] HomeKit: [Client 1073692844] Closing client connection
>>> [  67672] HomeKit: Got new client: local 10.22.22.119:5556, remote 10.22.22.133:57949
[... 228 lines cut ...]
>>> [ 327449] HomeKit: [Client 1073694156] Got characteristic 1.11 change event
>>> [ 327456] RATGDO: Light Status Off
>>> [ 327459] HomeKit: [Client 1073694156] Got characteristic 1.21 change event
>>> [ 327466] HomeKit: [Client 1073694156] Got characteristic 1.15 change event
>>> [ 327473] HomeKit: [Client 1073694156] Got characteristic 1.14 change event
>>> [ 337647] HomeKit: [Client 1073694156] Disconnected!
>>> [ 337652] HomeKit: [Client 1073694156] Closing client connection
>>> [ 361609] HomeKit: Got new client: local 10.22.22.119:5556, remote 10.22.22.133:57960
>>> [ 361620] HomeKit: [Client 1073694156] Pair Verify Step 1/2
>>> [ 362111] HomeKit: Free heap: 26632
>>> [ 362237] HomeKit: [Client 1073694156] Pair Verify Step 2/2
>>> [ 362244] HomeKit: [Client 1073694156] Found pairing with DCD2F97A-FE1B-431A-91AC-54F6F2DBFCBE
>>> [ 362274] HomeKit: Call ge_double_scalarmult_vartime_lowmem in ge_low_mem.c
>>> [ 363715] HomeKit: [Client 1073694156] Verification successful, secure session established
>>> [ 363724] HomeKit: Free heap: 26096
>>> [ 363885] HomeKit: [Client 1073694156] Get Accessories
>>> [ 363977] RATGDO: get active: 1
>>> [ 363982] RATGDO: get current door state: 1
>>> [ 363987] RATGDO: get target door state: 1
>>> [ 364070] RATGDO: get obstruction: 0
>>> [ 364075] RATGDO: get current lock state: 0
>>> [ 364080] RATGDO: get target lock state: 0
>>> [ 364206] RATGDO: get light state: Off
Disconnected (read failed: [Errno 6] Device not configured)
Reconnecting to /dev/cu.wchusbserial210 ........

lttlrck, 2024-01-05:
> Seems like the unit has a hard time when it can see multiple nodes of the same network and is constantly hopping between.

it does seem that way, more likely than an issue with mesh management per my original report. All my satellites _are_ reachable throughout my home, including when it works successfully when connected to the master AP - so perhaps there is a signal strength threshold issue that causing flip-flopping, and that in turn causes a more serious resource or other issue in the stack. 

Thanks for everyone's efforts, this is a very cool project. I'm sure this get fixed.

jgstroud, 2024-01-06:
It would be really helpful if you can get a pcap of the traffic when this is happening.  you can apply a host filter for just the ratgdo ip .   But debugging network issues without seeing what's happening on the network is really hard.  When I first connected my ratgdo, I was seeing similar issues and I traced it back to a problem where a device on my network was spamming it with bad mdns packets and overwhelming the network stack.  

You don't need to share your pcap here as it may contain information you don't want on the internet, but you should at the very least scan it for yourself and see if you see anything unusual.  you can send the pcap privately if you need help looking at it.

lttlrck, 2024-01-06:
> It would be really helpful if you can get a pcap of the traffic when this is happening. you can apply a host filter for just the ratgdo ip . But debugging network issues without seeing what's [...]

Is there a generic ESP2566 board I could get from Amazon that I could sue for testing? The. It'd be trivial for me to try out new firmware's and grab a PCAP. 

Something like this?

Hosyond 3Pcs ESP8266 ESP-12E CP2102 NodeMCU Lua Wireless Module Development Board for Arduino IDE/Micropython https://a.co/d/62J6OSw

radford, 2024-01-06:
+1

I too am having similar issues with a Muti-AP Eero network.  I suspect the issue is that I have multiple AP's that share the same SSID but have different BSSID's.   I wonder if we are not storing the BSSID in the flash so that we can't reliably connect to the correct one?  Or maybe the code that removes "duplicate" networks should be taking the BSSID into account?

michael-pesce, 2024-01-06:
> It would be really helpful if you can get a pcap of the traffic when this is happening. you can apply a host filter for just the ratgdo ip . But debugging network issues without seeing what's [...]

I just grabbed a PCAP but it does have a ton of info that's hard to scrub. Is there a more private way of sharing or scrubbing?

jgstroud, 2024-01-06:
The only thing I see there is mdns and arps and they look ok.  The key would be to capture when you see it go unresponsive or if you are pinging it to get a capture when a packet is dropped or when you see really high latency.

edrikk, 2024-01-08:
Not sure how to implement in this firmware, and although not directly related, it can’t hurt if it’s relevant:

https://github.com/ratgdo/esphome-ratgdo/issues/169

jgstroud, 2024-01-08:
Neat.  I'll give this a try.  But probably not before the end of the week.

logjames, 2024-01-10:
I have a pro 6 gateway and that my ratgdo connects to and I am not having any issues. 

--- 192.168.5.157 ping statistics ---
36 packets transmitted, 36 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 4.737/8.211/16.620/3.170 ms

lttlrck, 2024-01-10:
Mine works fine on the gateway. It's as soon as there are satellites close by it gets really wobbly. 

I upgraded to 0.9 yesterday and switched back to my Eero network to grab a PCAP. It connected and I got *nothing*: no ping, no response in HK, no packets in wireshark. 

I reset it multiple times didn't help. Switched back to my non-mesh WiFi and it was back up and running with no issue. To be clear I'm not blaming 0.9, I seem to recall I experience the same issue before. It may have been a glitched set up.  

Then I ran out of time. In the coming days I'll have another go. I wish it were possible to switch WiFi remotely, but IIUC that's not technically possible.

jgstroud, 2024-03-19:
Can someone with an eero network try version 1.0 and report back if there is any improvements

petercockroach, 2024-03-20:
> Can someone with an eero network try version 1.0 and report back if there is any improvements

Very preliminary feedback, but so far results are promising on 1.0. I just re-installed it and I've got an uptime of 26 mins and counting, which I was not able to achieve prior to this release. 

The system also didn't reset during the pairing process which was also very promising. I'll make sure to provide further info on uptime as I monitor.

jgstroud, 2024-03-20:
Bummer. If you can get logs that would be helpful

jgstroud, 2024-03-20:
I wonder what it is about these eero networks. I think this is the biggest outstanding issue.  Some things I can try:

- Try storing BSSID as suggested by @radford 
- Try reducing wifi power to reduce AP hopping as suggested in #116 

On my device with the 1.0 firmware, I now see very responsive pings like you see with the mqtt firmware.  Can someone with eero network see if their pings are back to normal or if you still see the wildly varying results?

Since I don't have an eero network, a serial log with a crash dump would go a long way to helping me find the issue.

petercockroach, 2024-03-20:
> Bummer. If you can get logs that would be helpful

Seems like it survived the night at least. 7 hours and counting at the moment, but it must have rebooted again just before midnight. 

Is there a way to get logs to stream wirelessly or do I have to keep my computer attached and tail USB logs?

eric-ooi, 2024-03-20:
I use two Eero Pro 6 nodes in bridge mode with wired backhaul and haven't had any network issues since v0.9 (and v0.6 prior to that is what fixed my pings).  Before updating to version 1.0 yesterday it had been connected for over 21 days and was working great.  Checking uptime now I see it's been connected for over 16 hours which is when I updated.

I'd check the wiring and power source on the ratgdo to make sure there isn't something that's loose causing it to disconnect abruptly.

jgstroud, 2024-03-20:
Anyone here want to try PR #133

edrikk, 2024-03-20:
> 
> * Try reducing wifi power to reduce AP hopping as suggested in [[FEATURE REQUEST] A few settings to expose if possible / if relevant to this fork #116](https://github.com/ratgdo/homekit-ratgdo/issues/116)
> 

I do think that a few of the items in the feature request will benefit overall useability and stability; if not both.
I just added ability to update wifi info, but generally speaking all wifi settings (SSID, Pass, TX Power) could be user updated, several birds could be hit with the same stone I believe.

petercockroach, 2024-03-24:
I'm happy to report 1.0 has been working well for me. Uptime has lasted about 3 days as of the time of writing this. It's clearly still rebooting, but far less frequently than before and it has not (yet) impacted my usability.

jgstroud, 2024-03-27:
Allow forcing 11G mode in #133 .  Build v1.1.0

jgstroud, 2024-04-30:
Should be fixed as of 1.3.0.  You can now force 802.11g and reduce your tx power.

closed 2024-04-30 by jgstroud