ratgdo/homekit-ratgdo #87: Webpage - Part 2

dkerr64, 2024-01-10:
Opening this Issue to track updates to the web page.  This compliments and replaces issue #56

**Tasks to do (reported bugs):**

- Fix typos/spelling errors, formatting, etc.
- reduce size of favicon / apple-touch-icon

**Still to implement:**
- Password protected with default "admin" that user can change
- Upload new firmware, improve GUI
- Manually set IP/netmask/gateway/DNS
- Update/change SSID/password (maybe too risky?)
- Update/change web page password
- View logs / set syslog server address? (may not be possible)
- Control the door (open/close/etc) -- going too far maybe?

Progress on the above is being committed into this [branch](https://github.com/dkerr64/homekit-ratgdo/tree/webpage-part2), no PR yet.

dkerr64, 2024-01-10:
So, in the development branch I have fixed typos, reduced size of the icons by ~50% and implemented a version check which will report if the user is up-to-date or if a newer version is available.  Examples of the version info...

[image]
[image]

The links will take you to the release page on GitHub.

Version checking is implemented with the Github API and looks at all releases then sorts by created time and uses the most recent.  It does not use the "latest" identifier... partly because the ratgdo does not have any "latest" because all releases are marked as pre-release.  And also because even if we had it, we would still want to offer the most recent pre-release.

We have our versioning a little screwy.  We are tagging everything "v0.0.0" so with a leading letter "v".  However in the code auto_firmware_version we are removing that letter "v'.  This complicates things because it means I have to add back in the latter "v" to do the comparison.  We should consider being consistent.  Either the source code uses the complete version/tag string, or we remove letter "v" from tagging.  Complicated is probably not the right word, but it exposes us to future bugs if tagging suddenly stops using the letter "v".

dkerr64, 2024-01-10:
On the icons, I found a tool, [ImageOptim](https://imageoptim.com/mac), that compressed the two icons.  That has improved things greatly.  To get any further improvement we probably have to use the same icon for both favicon and apple-touch-icon but that has complications due to transparency.  I think how we have it now is good enough.

dkerr64, 2024-01-10:
Ok, now for the really cool part.  I have a _TEST_ (emphasis on test) code that will download the latest release from GitHub to your browser and upload it to ratgdo without any intermediate step.  Click the `TEST` button to try it...

[image]

If you do try it, keep the serial console open because that is the only way to reassure yourself that it is working... a series of dot-dot-dot to report upload progress.  At the end it reboots.

BUT... there is a big hack in here.  Because of CORS (Cross-Origin Resource Sharing) this shouldn't be possible.  A browser will block attempts to load a resource from a 3rd party site (a URL that is different from the URL that the page was loaded from).  As far as I can tell, there is no workaround browser/client side... it needs to be done by the (or a) server.

The hack is to go through https://cors-anywhere.herokuapp.com/ but that exposes us to this server going offline.  We could...

1. host our own intermediate server that mimics what the above server does.
2. download from GitHub directly to the ratgdo

Neither of these are much fun.  I spent hours last weekend trying to get https/SSL access working in ratgdo and gave up.  The packages required take up a lot of memory, and besides, I couldn't get it to work.

Here is the client-side code that is working...
```javascript
      console.log("Download firmware from: " + status.downloadURL);
      let response = await fetch("https://cors-anywhere.herokuapp.com/" + status.downloadURL, {
        method: "GET",
        cache: "no-cache",
        redirect: "follow",
        headers: {
          "Accept": "application/octet-stream",
        },
      });
      const blob = await response.blob();
      console.log("Download complete, size: " + blob.size);
      const formData = new FormData();
      formData.append("content", blob);
      response = await fetch("./update", {
        method: "POST",
        body: formData,
      })
      console.log("Upload complete");

```

donavanbecker, 2024-01-10:
Awesome work @dkerr64!

dkerr64, 2024-01-10:
Flash of inspiration overnight and a solution to the CORS problem.  Turns out that GitHub Pages does not block cross-origin.  We already host the binary file there for the flash tool so...

```
let response = await fetch("https://ratgdo.github.io/homekit-ratgdo/firmware/" + status.filename, {
        method: "GET",
        cache: "no-cache",
        redirect: "follow",
        headers: {
          "Accept": "application/octet-stream",
        },
      });
```

works. !!!!

This means we can directly download firmware from GitHub (via browser) which will greatly simplify things for most users.  We should still offer ability to upload a firmware binary from local file system as well.

I'm traveling for a couple of days, so won't make any progress, but I'm really happy to have found a way to do this.

donavanbecker, 2024-01-10:
Dreaming homekit-ratgdo!!! Tried out the branch. Tried the `test` but didn't do anything as expected since it is on the same version.

dkerr64, 2024-01-10:
> Dreaming homekit-ratgdo!!!
> 
> Tried out the branch. Tried the `test` but didn't do anything as expected since it is on the same version.

It is still very rough.  Much to be done to e.g. advise user that they are already up to date.

donavanbecker, 2024-01-10:
Correction it reverted me back to the latest published pre-release on repo. Makes sense.

dkerr64, 2024-01-12:
Latest on firmware update.  I now have web page of our own design rather than use the default...

[image]

Both update directly from GitHub and from a local file are working.  Right now I'm piggybacking on the ESP8266HTTPUpdateServer code but I am very tempted to eliminate that and move the code into our own source files.  Main reason is that it gives me more control over what happens at the end... right now it just reboots, but I cannot intercept that to first change the screen to display a "completed" message.  Nor can I show progress in the browser screen... which at the moment goes to console log.

donavanbecker, 2024-01-12:
> Latest on firmware update.  I now have web page of our own design rather than use the default...
>
>
>
> [image]
>
>
>
> Both update directly from GitHub and from a local file are working.  Right now I'm [...]

Looks awesome! I would be all for that! I think it would be good to have control over what is displayed.

donavanbecker, 2024-01-12:
I just upgraded and noticed that in safari desktop browser it looks like this: 
[image]

dkerr64, 2024-01-12:
Humm, that is odd because I use safari to dev/test myself.  You did prompt me to try it on Chrome and Firefox and I found a problem with the chose file button, so looks like I have to accommodate browser issues.

Your problem looks like the old CSS was still in your browser cache.  Hold down the shift key and reload, that should fix it.

donavanbecker, 2024-01-12:
Holding down shift and reloading didn't help also tried in private browsing mode with same results.

dkerr64, 2024-01-13:
> Holding down shift and reloading didn't help also tried in private browsing mode with same results.

That is really odd.  The copy/paste you sent clearly indicates that the latest `style.css` is not loaded.  Can you look at the browser debugger and watch for error messages and check that style.css it is using matches what is in GitHub.

Thanks

lttlrck, 2024-01-13:
Also whilst dev tools is open you can check Disable Cache and be 100% sure it's loading from the server.

Chrome's caching is really aggressive and I have up on the shift reload trick years ago because it really doesn't seem to work sometimes.

dkerr64, 2024-01-14:
Added light and door controls...

[image]

donavanbecker, 2024-01-14:
> Added light and door controls...
> 
> 
> 
> [image]
> 
> 

I am totally not against this, but I think if you are going to add these controls to the Web UI, their the UI needs to have a username and password to secure it.

thenewwazoo, 2024-01-14:
Web based controls guarded by an authentication scheme are far beyond what I envision for this project, and I am not a fan. There are alternatives if one wants to duplicate HomeKit functionality via other means.

dkerr64, 2024-01-14:
> I am totally not against this, but I think if you are going to add these controls to the Web UI, their the UI needs to have a username and password to secure it.

Authentication is in the to-do list.

dkerr64, 2024-01-14:
> Web based controls guarded by an authentication scheme are far beyond what I envision for this project, and I am not a fan. There are alternatives if one wants to duplicate HomeKit functionality via other means.

The webpage should be guarded by authentication whether or not we allow control buttons.

donavanbecker, 2024-01-16:
> Added light and door controls...
> 
> <img alt="image" width="565" src="https://private-user-images.githubusercontent.com/17837612/296527160-bd9a66ca-9a39-4fc5-b252-da4e211d7b1d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDUzODUwMzIsIm5iZiI6MTcwNTM4NDczMiwicGF0aCI6Ii8xNzgzNzYxMi8yOTY1MjcxNjAtYmQ5YTY2Y2EtOWEzOS00ZmM1LWIyNTItZGE0ZTIxMWQ3YjFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE2VDA1NTg1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIyNzM5ZThlMzJlYzlmNTc4YzEyNmY5OTE4OGY3YmE2MWFkZTkzODY5NDQ2N2FhZDE1MWQ4ODE4ZDc4YmQ5ZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m5P_oPcBuiybyXJdMePDU_L2Frl1ruNSgS4Ot0KKCxc">

Any way to add the lock function to this as well?

dkerr64, 2024-01-16:
> > Added light and door controls...
> Any way to add the lock function to this as well?

Yes, done in my current version.  However...

In testing the control buttons, it is not stable.  Using the light as example, if I turn it on/off many times in a row the server will stop responding and after a delay crash and reboot itself.  This does not occur if I repeatedly turn the light on/off from Apple Home app.  I do not know how to interpret the crash dump but I'm suspecting the ESP8266WebServer code.

I'm eagerly waiting @thenewwazoo 's updated ESP-based HomeKit, because I suspected that this would require a change to other components including the web server.  And lo, last night I see @thenewwazoo committed into his branch a switch from ESP8266WebServer to esp_http_server.  Woo Hoo.

So, @thenewwazoo how best to coordinate? I'd like to migrate my work over to the new base.  At what point does it make sense for me to rebase my work onto your branch?  THANKS !!!

dkerr64, 2024-01-16:
A note on authentication.  I have added DIGEST_AUTH to the code, this is a bit more secure than BASIC_AUTH.  I have created a "settings" page which is password protected where user can change the password.  I have also wrapped the GDO controls and firmware updates in authentication as well.

Now... this has been working really well with Chrome and Firefox, but I have run into some issues with Safari which was not prompting for userid/password after changing it.  I think I've sorted it, but there are widespread reports (thank you google) of this safari behavior... not necessarily on password change, but on fetch API to a protected sub-resource.

dkerr64, 2024-01-16:
And next... Semaphores.  I have wrapped fetch API calls in a Semaphore.  I did this because I have run into issues on other projects when talking to smart home devices where it was important for one request to complete before another started.

So, I use a semaphore to serialize access/requests to the server with Javascript fetch API. Why bother?  Well an abundance of caution... fetch is asynchronous and I have an interval timer running to keep the web page up-to-date on server state (I should really replace that with web socket?)  It could yield while waiting for response from server, during which time another fetch request could be sent to the server.  Because the ESP8266WebServer code is primitive, and because of the problems with light on/off I noted above, I decided to add this.

But, I hope this is overkill and unnecessary... that the esp_http_server solves the problem and I can remove this.

dkerr64, 2024-02-07:
closing as complete, part 2 merged into main.  Will open new issue if required for any remaining work.

closed 2024-02-07 by dkerr64