ratgdo/homekit-ratgdo #88: Lock and Unlock exposed to HomeKit, but not Open and Close

mikemormino, 2024-01-11:
When attempting to create an automation related to the door opening, Open/Close isn’t available but Lock/Unlock are.  They don’t work the same way.
[image]

HughesAndrewC, 2024-01-11:
I had the exact same problem. I was able to use a suggestion to get access to the door opening/closing as a trigger. 
[image]

What's interesting is that now that I have created an automation using that suggestion, I no longer have access to the lock/unlock triggers.
[image]

mikemormino, 2024-01-11:
Yup, that worked for me too.  Interesting workaround.  Basically, the solution is to take the HomeKit suggestion and slice it and dice it to your needs.

HughesAndrewC, 2024-01-12:
That leads me to believe it's a problem on the HomeKit side instead of the ratgdo firmware end. Is open/close now available when creating an automation from scratch?

mikemormino, 2024-01-12:
No, just lock and unlock.

thenewwazoo, 2024-01-12:
This is super weird and strikes me as an iOS Home app problem. I actually can't repro on the macOS version:

[image]

It offers me the open/close trigger straight away (I have no other automations for the accessory set).

thenewwazoo, 2024-01-27:
Going to close this as a bug in the iOS Home app and not the firmware. I'd say I'll file a RADAR but... 🤣

closed 2024-01-27 by thenewwazoo