ratgdo/homekit-ratgdo #89: Activity light stays on all of the time.

Cristov9000, 2024-01-11:
This was mentioned in Issue #25 that was closed but the issue persists.   Per PaulWieland from Issue #25 "fwiw, the mqtt firmware turns the led on when searching for wifi. Once connected it shuts off. It then flashes whenever it receives a command.  I would suggest having the same behavior for all firmware versions to help people trouble shoot issues in a common way."

Currently the activity LED stays on as default instead of turning off after connection to Wifi and flashing on activity.  

Tested on 5x RATGDO v2.5i running 0.9.0.

dkerr64, 2024-01-13:
Maybe make it an option?

The vast majority of devices use LEDs to indicate status / activity.  Think the lights on a ethernet port, disk drive, etc.  The flashing is an indicator that all is good.  Whereas, in my mind, a constant on light suggests the device is hung, as no activity is visible.  And off is of course off.

PaulWieland, 2024-01-13:
The reason to make it flash on activity is that garages are not know for having the strongest WiFi coverage. If the gdo isn’t responding to commands it’s nice to easily tell whether the commands are being received by the ratgdo control board vs it being another issue related to client id/rolling code/etc.

Cristov9000, 2024-01-15:
Agreed with Paul.  The status light flashing on received command to confirm its receiving is helpful for troubleshooting.  So it the light turning on for boot and off when connected to wifi.  This is the way the MQTT firmware is setup.

The constant on light just makes my neighbors ask me why the garage glows blue every night....

jgstroud, 2024-01-15:
The LED works opposite of the MQTT firmware.  It's on by default and blinks on activity.  when the feature was implemented, all those that commented agreed that they like the LED on by default showing power and blinking off for activity.  This is similar to something like a network device where it's steady on for link and blinks off for activity.  If your garage is glowing blue, you might consider putting some tape over the LED.

thenewwazoo, 2024-01-15:
This is one of those issues with no objectively correct answer but lots of preferences. I prefer on-by-default and off-when-active because it provides an indication that the system has booted and is ready, without having to communicate with the system successfully. I’m going to close this as “working as designed” with apologies to everyone who prefers it the opposite way.

closed 2024-01-15 by thenewwazoo