ratgdo/homekit-ratgdo #91: Wireless Lock

IWCaldwell, 2024-01-11:
It would be fantastic to have a Lock/Unlock feature for the remote lockout. Imagine the convenience of being able to disable your garage remotes or the outdoor keypad remotely via HomeKit, especially when you're away from home for an extended period.

donavanbecker, 2024-01-11:
This is always possible, but not visible in the Apple Home app. You have to control it with Eve, Controller, Home+ (most 3rd Party HomeKit Apps)

IWCaldwell, 2024-01-12:
Uncertain why we couldn't simply integrate a lock within the HomeKit app for seamless management. It's as straightforward as incorporating a lock, akin to a door lock, ensuring native compatibility with HomeKit.

I can achieve this effortlessly through MQTT with Homebridge, would be awesome to make it native though.

donavanbecker, 2024-01-12:
@IWCaldwell I thought that same thing as well but this is native HomeKit. There is a lock characteristic designed specifically for garage doors, which is being used here. Apple however didn't choose to reveal this lock. You can ask Siri to lock and unlock your garage door, and it will work.

Here is an example of the garage accessory and it's characteristics: https://developers.homebridge.io/#/service/GarageDoorOpener

thenewwazoo, 2024-01-27:
Closing as resolved. Thanks @donavanbecker!

closed 2024-01-27 by thenewwazoo