ratgdo/homekit-ratgdo #92: USB Port soldered shut

Anthwerp, 2024-01-14:
[image]

Posted on reddit and they mentioned to try and post here. I can't connect the USB because the port is covered. Is there a way for me to fix?

OleSam85, 2024-01-14:
Seems defective if it came that way, I would contact @PaulWieland and reference your order #  and I’d imagine he’ll help you with an exchange.

Anthwerp, 2024-01-14:
How to do it.. Just email? 
Yeah got it yesterday, opened it today and was looking up the documentation thinking.. "Well maybe we don't use this port anymore, but how do I load the firmware" lol.

rlowens, 2024-01-14:
> Is there a way for me to fix?

Do you have a soldering iron and some desoldering braid?

But mostly you just need to get it replaced by @PaulWieland

PaulWieland, 2024-01-16:
GitHub is not the place for this type of issue. Please reply to your receipt and it will be taken care of.