ratgdo/homekit-ratgdo #94: Door shows as "No Response" in HomeKit until rebooted

lancep, 2024-01-16:
I can successfully add the ratgdo to HomeKit, and everything works great, but after a day or so, it shows as "No Response" until I reboot the device.

drewskix86, 2024-01-17:
Me three, it seems to hang and will occasionally come back on its own after a couple of minutes, other times it requires restart. Running version 0.9 with a Chamberlain CS105MYG with obstruction sensor installed.

egrim, 2024-01-17:
I thought it was just mine and thought maybe my RATGDO's hardware had a problem.  I'm planning to try to debug at some point, but not sure where to begin with that yet so any tips are appreciated.

OleSam85, 2024-01-17:
I'm in no way qualified to offer much technical guidance here, but I documented how I was able to log from the device in Issue #24: 
https://github.com/ratgdo/homekit-ratgdo/issues/24#issuecomment-1854591497
There might be other ways and other tricks needed to get the necessary data to fully debug the issue, but it seemed to work fine for logging from the device. Of course I had to keep my computer awake and connected to the ratgdo in the garage the entire time, and of course as soon as I started logging, I was unable to reproduce the problem! Anyway, hopefully that helps.

pattont, 2024-01-17:
> @thenewwazoo <https://github.com/thenewwazoo> is working on replacing the
> Arduino-based HomeKit stack with the ESP-HomeKit stack. We are hopeful that
> this will bring more stablity to the connections.

I read in another Issue that with UniFi the Ratgdo roaming between APs was
causing an issue. Neither of my Ratgdos have went “no response” since I
forced connection to a single AP in the UniFi settings. I will report back
if it happens again, but looks like that was the issue after a few days of
it working.

egrim, 2024-01-17:
Oh!  I'll give that a try as I am indeed on UniFi.

sttonev, 2024-01-17:
I'm on UniFi too and have been fortunate in having really good uptime. My configuration is to have a separate (2.4Ghz) SSID broadcasting *only* from the closest IP. Not sure if there's a material difference between that and forcing a client to a given AP, but I wanted to offer another (working) solution.

lancep, 2024-01-17:
Ah, I'm also on UniFi, I'll try the roaming lock

pattont, 2024-01-18:
Crazy how many of us with these home automation devices also has Unifi
setups!! :-)

jgstroud, 2024-01-18:
I'm on unifi and I locked my device to the closest AP.  I have twice hit this issue where I had to power cycle the ratgdo to get it back alive.  And it was several weeks apart.  So, this helps, but don't expect it to be a magic bullet.  I've also seen where if I power cycle my AP, it may not reconnect.  Which seems like it might be similar to moving between APs.  I think once it's connected it doesn't like getting disconnected.

dmsinger, 2024-01-18:
Also Unifi, have also run into this issue twice. However, I can see the ratgdo has not connected to another AP within Unifi's console, so that wasn't issue. I've had no AP resets either. ratdgo was also still connected to the AP when it was reset according to the connection log.

lancep, 2024-01-18:
I did the AP lock yesterday, woke up this morning to "no response" :(

dmwilson1990, 2024-01-19:
Same issue here - UniFi and locked to AP. Still no response.

infamous-pattern, 2024-01-19:
Been having the same issue with my RATGDO on 0.9 and also running UniFi. Going to lock to a single AP just to be safe. Looking forward to the next firmware update. Thanks.

sttonev, 2024-01-19:
> I'm on UniFi too and have been fortunate in having really good uptime. My configuration is to have a separate (2.4Ghz) SSID broadcasting *only* from the closest AP. Not sure if there's a material difference between that and forcing a client to a given AP, but I wanted to offer another (working) solution. 

In case it matters, I'm on 0.6.0. It doesn't seem like any thing was changed in the interim that would affect connectivity, though.

jim-wong, 2024-01-20:
I'm on 0.8 and am seeing this problem, as well. In the most recent occurrence, it happened after just a few hours: HomeKit couldn't see the garage door, the web interface wouldn't load, and the device wasn't reachable via ping.

If there's anything I can do provide data to help troubleshoot this issue, I'd love to know.

thenewwazoo, 2024-01-20:
I appreciate everyone continuing to submit bug reports. Thank you! And I'm not kidding - it helps me get a sense of the scope of the problem.

I do have good news, though! I have made significant progress in re-implementing the firmware to avoid the problem, which is instability in the Arduino core. I've replaced Arduino with the HomeKit SDK written by Espressif (who makes the ESP8266 chips used in the ratgdo), and my prototype is working well. 

Stay tuned, I expect to be able to publish a version for intrepid users to test in the next few days.

tadaschi, 2024-01-20:
Thank thenewazzo!
I am another one with unifi and HomeKit 0.9.0

same problem

we gonna be waiting for a new firmware!

keep the great work!

excluding this bug the HomeKit is by far the easiest and most responsive of all three firmwares for me

Guy559, 2024-01-22:
I'll also confirm issue with Unifi and 0.9.0 firmware. Three other devices using ESPHome work fine via HASS. Only this device fails to update, even when connecting to known good AP.

Appreciate acknowledgement and I, too, am looking forward to f/w update.

blakej, 2024-01-24:
Unifi user here who did a good bit of troubleshooting before. Three APs, three ratgdos.

v0.8 has definitely corrected the "No Response" issue I was very very frequently getting cause by hopping between APs.

What v0.8 did not (and could not) correct was WiFI signal issues that only my ESPs were seeming to have. Despite "No Response" being corrected (when WiFi was working), I would still randomly and frequently get drop offs from WiFi entirely - some ratgdos more than others. At least two of them would not reliably stay connected for even an hour.

About two weeks ago I made two changes in Unifi, and since then only one device has reconnected one time (and only a 5 sec disconnect) - the other two devices haven't lost connection a single time.

Change #1: Stopped using "Minimum Data Rate Control" on my 2.4 GHz network. It was previously set to "5.5 Mbps" as a minimum, and I think this was forcing hopping as well as disconnects from not meeting the minimum.

Change #2 (and probably more impactful): I increased AP 2.4 GHz signal strength. Yes, I know just increasing AP signal strength doesn't mean the devices can respond stronger, but it has helped - massively. Before this, each ratgdo was typically under 10 Mbps on rx & tx. After this, right now, they are all 52-54 Mbps on rx & tx. "AP/Client Signal Balance" are all "Good". "WiFi Experience" are all "Excellent".

For everyone else on Unifi, if you're experience high latency / low data rates / full wifi disconnects - check the settings I listed above. Looking forward to @thenewwazoo's HomeKit rewrite, but I suspect it will have no impact if you can't get WiFi stable first like I was experiencing.

I'm using v0.9, and since making the listed Unifi changes above, all three ratgdo's have been flawless - instant responses and never showing "No Response".

DMBlakeley, 2024-01-24:
I do not have a UniFi system but am using a Velop system. One setting on the Velop which I have found made a difference for HomeKit device connectivity was to turn off "Device Steering". This lets the device connect to its preferred node rather than the Velop system steering which can result in a disconnect. Not sure if similar setting on UniFi but one to also try if available.

Guy559, 2024-01-24:
Went ahead and reloaded f/w, reconnected to closest UniFi WiFi and all is good in Home at the moment. Will keep an eye on it in days to come. 

Looking forward to next release. Impressive to see how quickly this went from concept to reality.

Minnow1986, 2024-01-24:
I had the same problem with Unifi (UAP-AC-HD). I downgraded WPA2/WPA3 (combination) to WPA2 (only) and it seems to be working well. I do use Band Steering and Fast Roaming. I don't use QOS. It doesn't seem to be causing an issue at the moment.

DMBlakeley, 2024-01-24:
Also recommend having your 2.4GHz network configured for WPA2 and not WPA2/WPA3 (combination).

thetvguru, 2024-01-24:
Are you guys resolving this issue by rebooting from the web interface or removing power? Are you needing to re-pair to HomeKit?
I have been unable to fix this issue my rebooting both ways, but haven't tried to re-pair to HomeKit.

egrim, 2024-01-24:
I've been able to resolve by rebooting in the web interface, but it sounds like others weren't able to reach it so we're forced to power-cycle it.

jim-wong, 2024-01-24:
@thetvguru For me, power cycling the box has done the trick.

thetvguru, 2024-01-25:
Mine is working again. For myself, I had to remove and reinstall to HomeKit.

thenewwazoo, 2024-01-25:
The underlying problem is one of instability in the HomeKit implementation, unfortunately. I don't have any insightful analysis about environmental factors (network topology, mDNS chatter, etc) that contribute to failure, but I would not put any faith in changing configurations to have any lasting impact on reliability.

I'm actively working on a re-implementation based on the Espressif HomeKit SDK, but am currently struggling with the reality that the ESP8266 really is just underpowered for HomeKit applications. There's a reason they never took off as a preferred platform and Espressif recommends against them. The ESP32-based families work great! It's downright _easy_ to get them working, but I'm trying to get the ESP8266 working in order to support the devices that are out in the field today. I'm no longer very optimistic I'll be successful.

(this is your reminder that I'm just a dude doing this as a hobby and I don't speak for Paul or RATGDO or, basically, anyone else, and have no inside information)

So anyway, please accept my apologies for the problems you're all having. I wish I had better news.

blakej, 2024-01-25:
> The underlying problem is one of instability in the HomeKit implementation, unfortunately. I don't have any insightful analysis about environmental factors (network topology, mDNS chatter, etc) that [...]

Understood that your hands are fairly tied on the current HomeKit implementation - that project is no longer maintained and has issues. That being said, I believe the current implementation *can* work well if environmental factors are eliminated. I feel like I'm proof of that, since v0.8 along with a couple AP tweaks has taken me from less than 1 hour stability to more than 2 weeks now.

Everyone's setup is a little different though - VLAN differences, HomeKit hub differences, etc. At this point I doubt anything more can be done to "correct" the current HomeKit code (the issues are in the dependencies), but maybe we can at least sort out the environmental issues that are causing problems for some people and find workarounds. If anyone else with Unifi continues to have issues and wants to compare notes, feel free to ask me.

> I'm actively working on a re-implementation based on the Espressif HomeKit SDK, but am currently struggling with the reality that the ESP8266 really is just underpowered for HomeKit applications. [...]

Sounds like you hit some roadblocks on the latest efforts, but hopefully some good ideas come up to resolve them. I wish there was an easy way to swap out the 8266 board to an ESP32, but I don't believe there's any good options for that.

> (this is your reminder that I'm just a dude doing this as a hobby and I don't speak for Paul or RATGDO or, basically, anyone else, and have no inside information)
> 
> So anyway, please accept my apologies for the problems you're all having. I wish I had better news.

As a reminder for you: you've done a great job with this project and built a very solid product in a short amount of time, as a hobby. Sure there's some issues out there still, but I suspect a large number of people are using it silently and without any issues. I'm in that camp now - everything is working reliably without intervention. Thanks for all your work on it - between Paul's initial design and your HomeKit release, I'm left in a better place than before Chamberlain started this whole mess (being on the other side now, I'm even glad they did).

lttlrck, 2024-01-25:
I had the same issues 2-3 weeks back with Eero. My conclusion through testing (removing/moving satellites) is that it is caused by the mesh, multiple visible APs on the same WLAN. Eero how no way to bind/block clients to a certain AP, but I'm not certain that is 100% reliable solution (even if work for some) as the clients will still see the beacon, they just be steered (coerced) to a specific one and I think that could still cause oscillate between APs and trigger a problem.

Anyway, my solution was to hook up an old TP-Link AP in my garage as a second separate WLAN (on the same LAN) and use that just for ratgdo. Instant stability.

Of course that may not be possible or acceptable to everyone, but it been 100% stable since I did this. I can trust it to the point that I'm about to disconnect the openers from MyQ because I don't want them to see my comings and goings anymore.

I already have Homebridge running for other purposes, so I _could_ have gone that route (it was my original plan) but homekit-ratgdo + an old AP is the superior solution for me.

ThePostman22, 2024-01-26:
Is there a way to know/tell which ESP I have? (ESP8266 vs ESP32)

thenewwazoo, 2024-01-26:
All shipped ratgdos are ESP8266-based.

atmendez, 2024-01-28:
If we have instability issues running v0.9 of Homekit on Ratgdo it sounds like the best course is to try to provide better wifi environment, e.g. switch from 5 Ghz to a closer 2.4 Ghz and strengthen signal. I can access Ratgdo via the local IP address to reboot but it shows up in Homekit as not responding.

Is there any other option for us if we want to use Homekit with Ratgdo at this point?

pattont, 2024-01-28:
Since I set my ratgdos to a single AP I haven’t had a single issue with my
unifi setup.

Just my experience but it has been a week or so now.

leesonthenet, 2024-01-28:
> I read in another Issue that with UniFi the Ratgdo roaming between APs was causing an issue. Neither of my Ratgdos have went “no response” since I forced connection to a single AP in the UniFi [...]

I've got an ASUS Mesh, and I've locked it to my node closest to the gdo. I still lose connection after a few days and usually have to re-pair it.

atmendez, 2024-01-28:
My home setup doesn't include any access points - just the Unifi Dream Machine for 5 Ghz coverage plus an old router connected in bridge mode to provide 2.4 Ghz coverage close to the garage. I'll report back if I get stability from putting the Ratgdo on the stronger 2.4 Ghz network.

dibsies, 2024-01-29:
Forgive my ignorance on the subject, but does this mean the current no-response issue is tied specifically to current limitations of the homekit-ratgdo firmware? Is the mqtt-ratgdo / homebridge-ratgdo route subject to the same stability concerns?

Damien514, 2024-02-02:
Unifi setup here, running ratgdo on 0.10.0
Everything was working well, and like everybody here, I have the "No response" in my Homekit and I can't even connect  to the web interface.

Thinking out of the box, would it be possible to add any mechanism to check hourly if the wifi connection is OK, and of not, force reboot the device? Or even an auto-reboot system everynight, at 3AM for example? Seems to be a savage trick but it could to save my mariage ;-)

thetvguru, 2024-02-02:
Haha!
Mine continues to run without issue since 0.10.0. I would imagine it’s a
matter of time base on the comments before. Like you I understand what beta
means and am impressed with the progress being made on the project, my wife
less so. :)

Damien514, 2024-02-02:
For the context, I did some firmware upgrades on my APs yesterday, might be related.

lttlrck, 2024-02-02:
> What was confusing was that my last drop the ratgdo also un-paired from Homekit.

ahhh... so I am not alone.

I am on a dedicated 2.4Ghz AP and it's been excellent... **but** it seems like a restart of the AP/loss of signal can trigger de-registration. I am on 0.9.0.

**Both** my ratgdo's have become unregistered from homekit twice in the last two weeks at approximately the same time.

iyerusad, 2024-02-02:
No issues in my testing, has been stupidly rock solid experience. Magically shows up in carplay, low latency response etc. Tested for few weeks, including upgrading access points.

U7 Pros, multiple
SSID: WPA2, 2.4/5GHZ (other SSIDS are 2.4/5/6, and WAP3)
RatGDO 2.5i, in plastic case (firmware: 0.9.0, successfully OTA'ed from 0.8.0)
Hubs: ATV4k and multiple HomePod Minis, Threads enabled, only Homekit no Homebridge or Zigbee hubs
Mostly bone stock network, some settings enabled:
- Fixed IP reservation for ratGDO
- mDNS enabled
- IPv6 LAN side enabled
- No VLANs

atmendez, 2024-02-03:
Spoke too soon… RatGDO is currently unresponsive through HomeKit but shows up if I go to its local IP address in a browser from my home network. Using its ip interface to reboot the router  it it is still unresponsive in HomeKit. 

Do the non-HomeKit firmware options have this issue? I would be willing to try homebridge. 

> On Feb 2, 2024, at 9:48 AM, iyerusad ***@***.***> wrote:
> 
> ﻿
> No issues in my testing, has been stupidly rock solid experience. Magically shows up in carplay, low latency response etc. Tested for few weeks, including upgrading access points.
> 
> U7 Pros, multiple
> SSID: WPA2, 2.4/5GHZ (other SSIDS are 2.4/5/6, and WAP3)
> RatGDO 2.5i, in plastic case (firmware: 0.9.0, successfully OTA'ed from 0.8.0)
> Hubs: ATV4k and multiple HomePod Minis, Threads enabled, only Homekit no Homebridge or Zigbee hubs
> Mostly bone stock network, some settings enabled:
> 
> Fixed IP reservation for ratGDO
> mDNS enabled
> IPv6 LAN side enabled
> No VLANs
> —
>

lttlrck, 2024-02-03:
> Do the non-HomeKit firmware options have this issue? I would be willing to try homebridge.

I switched one of mine to MQTT and Homebridge today. I had to use an old AP to get ratgdo-HomeKit stable but it still deregisters itself from HomeKit. 

It's been perfect so far and conmects fine to my mesh network which ratgdo-HomeKit never did. Time will tell. 

I'll keep an eye on this project though because when it works it's great and I'd rather not have the HB dependency. Deregistering is particularly annoying because it breaks automations and shortcuts.

petercockroach, 2024-02-11:
I've been having similar issues and I'm begin to suspect it's because of my eero network. I'm also getting other symptoms including false notifications (details: https://github.com/ratgdo/homekit-ratgdo/issues/105)

For anyone running 0.10.0 or greater, can you check your uptime? That's one clear indicator for me. The uptime never lasts more than 20 mins or so.

DMBlakeley, 2024-02-11:
Have been following this thread but currently using ratdgo with homebridge hoping that the native HomeKit version will become stable. I have noticed on the ratgdo homebridge version will also disconnect very randomly but reconfigures and reconnects in a matter of a few seconds.

One item I do not see anybody mentioning is what they are using for their Apple hub? My experience has been that a wired Apple TV provides the best stability followed by WiFi connected HomePods. Then there is the question of OS version. I am currently running 17.3 as well as 17.4 beta on one of my Apple TVs. 17.3 is reasonably stable but occasionally I find that a camera in my setup will disconnect and then reconnect a couple minutes later. 17.4 beta is not quite there yet and have added a plug to be able to turn off if needed at which time one of the other Apple TVs will take over.

Wonder if some of the disconnect issues that are being observed with the HomeKit ratgdo may also be related?

iyerusad, 2024-02-11:
> I've been having similar issues and I'm begin to suspect it's because of my eero network. I'm also getting other symptoms including false notifications (details: #105)
>
> For anyone running 0.10.0 or [...]

v0.11.0/RatGDO 2.5i -- `Up Time: 4:14:58:53`
Prior to that v0.9.0 was rock solid as well. Upgrading from v0.9.0 to v0.11.0 had to unpair/repair due to motion sensor change. Device shows consistently in CarPlay on approach, connecting via IP to status page has been solid

Homekit hubs list HomePod mini as current, but there are multiple HomePod Minis and ATV 4k in the mix. WIFI has been Ubiquiti 6-Long Range and upgrade to 7-Pro. No homebridge or any extras - vanilla Apple HomeKit setup.

The only quirk I've seen with ratgdo-homekit has been potential code overlap -- opening with RatGDO then attempting to close via car's homelink button sometimes seems to want 2-3 presses on the homelink. Only happened twice but can't be certain I didn't click wrong homelink button. Might be a bug here, but haven't repo'ed it consistently enough to pursue.

jgstroud, 2024-02-12:
> I've been having similar issues and I'm begin to suspect it's because of my eero network. I'm also getting other symptoms including false notifications (details: #105)

eero networks seem to be a bit of a common thread with regard to these issues.  I don't think the Eero's are to blame, but there is something about these networks that the ratgdo doesn't seem to like.

atmendez, 2024-02-12:
I have a Unifi Dream Machine and had connectivity troubles on 5G. Switching
to a (non-Unifi) access point with a stronger 2.4 Ghz signal seems to have
fixed the problem for me.

Firmware: 0.9.0
Up Time: (dd:hh:mm:ss) 9:04:44:18

On Mon, Feb 12, 2024 at 2:08 PM jgstroud ***@***.***> wrote:

> I've been having similar issues and I'm begin to suspect it's because of
> my eero network. I'm also getting other symptoms including false
> notifications (details: #105
> <https://github.com/ratgdo/homekit-ratgdo/issues/105>)
>
> eero networks seem to be a bit of a common thread with regard to these
> issues. I don't think the Eero's are to blame, but there is something about
> these networks that the ratgdo doesn't seem to like.
>
>
> Message ID: ***@***.***>
>

dkerr64, 2024-02-12:
> I have a Unifi Dream Machine and had connectivity troubles on 5G. Switching to a (non-Unifi) access point with a stronger 2.4 Ghz signal seems to have fixed the problem for me.

THe ESP8266 does not support 5GHz, it can only connect on 2.4GHz.

dmsinger, 2024-02-12:
I used to have an Eero setup and the switching was very frequent. If there's any way to lock to one Eero, that would probably make it a lot more stable for you.

I haven't had to do anything with the Unifi setup outside of lock it to an AP (which it was doing outside of resets).

> One item I do not see anybody mentioning is what they are using for their Apple hub? 

My hub has switched a few times, and I think it could have been one of my drops, but it was only once (the drop). I've done multiple upgrades. Luckily I'm on a wired Apple TV right now, but you never know when that'll flip. Hopefully we get control of that one day.

jgstroud, 2024-02-12:
I use a wireless HomePod mini for my hub and mine has been very solid.  At this point I wouldn't rule anything out though.  I am also on a Unifi AP and I locked my device to the nearest AP.

I have a spare ratgdo with the serial console being logged.  I might try forcing it to roam and see if I can trigger this issue.  Based on https://github.com/mrthiti/Arduino-HomeKit-ESP8266/commit/3825ef4d1c11c79d242c843c094285cbab15ea57 I'm wondering if there is still something bad happening in the mdns stack when the ratgdo roams.

billy27607, 2024-02-29:
Like many here, I have a couple of ratgdo devices and a Unifi Network. The devices go non-responsive all the time. Nothing I do with the network helps at all. Even when they are non-resonsive in homekit, I can reach the web management page and use it responsively.  I have written my own homekit appliances a few times and have always had trouble when tying to use the Arduino-Homekit-Esp8266 library. Some kind of behavior. So I am convinced that this is something in the timing of the protocol between homekit and the device. 

While I hate to do it, I have a home bridge instance running already for some other stuff so I loaded the mqtt-ratgo firmware, used the homebridge-ratgdo plugin and everything runs well and is responsive.

I have had much better luck with the Espressif Homekit SDK in terms of performance and stability, but it is a lot harder to get going with it. So I hope that @thenewwazoo has better luck with that attempt. I hearby volunteer for any testing you want to run when you are ready.

iyerusad, 2024-02-29:
I've been running v0.11.0 for 22+ days uptime without any of these issues. I use it just about everyday and working consistently (haven't noticed a failure/issue).

Some speculation:
- **HomePod Mini related strength**? `I use a wireless HomePod mini for my hub and mine has been very solid.` I have 3-4 as hubs, as well as wired ATV4k (isn't chosen as primary hub as often as you'd like)
  - I have additional Eve Water plug that includes a Threads radio as part of my HomeKit. I don't run any of the bridges etc, just vanilla Homekit, and just RatGDO, Eve Water sensor, ATV, and the handful of homepod minis.

- **RatGDO hardware issue?** Mine is a v2.5i, ~2023-Dec batch.

- **Wiring issue?** I used recommended solid core wire (not stranded).
  - [Super unlikely but a thought] Maybe wiring is a vector for intentional/unintentional glitching. Silly but a thought. My GDO was never registered to MyQ or whatever their cloud thing is.

- **Environmental issue causing CPU issue (heat? interference?)?** Mine is wrapped in a Etsy case, maybe that helps. Unit is in garage, temps are 40-60 degrees Fahrenheit (maybe summer heat will introduce failure condition)

- **Network issue?** This one increasingly seems less likely to me, particularly if getting unpaired from Homekit. FWIW, multiple Unifi 7 Pros, static leased IP, pretty standard network config (no VLANs or tinkering, no pinning to specific AP, multicast DNS enabled). I do run separate SSID for these IOT like devices, but it all feeds into same network.

Not sure if this helps anyone else but some datapoints.

billy27607, 2024-03-01:
For what it is worth, I have a bit of data to add to the discussion. When my ratgdo devices were flashed with ratgdo-homekit, ping times to the devices were all over the place ranging from 10ms-60ms. There would be bursts of long latency times. Now that they are flashed with mqtt-ratgdio the ping times are constantly under 5ms. Same location, same network settings, etc. This suggests that the chip does not have enough juice to run the tcp stack and homekit in the Arduino framework.  Note that I have like 50+ accessories in this particular home and there are 5 potential hubs to talk to.

I have a number of devices with 8266s that work well in this environment, but all are running code based in the Homekit SDK rather than the Arduino framework.  These are nice little boards and there is no reason they be able to handle this with a good framework.

thenewwazoo, 2024-03-02:
@billy27607 I believe you're basically correct, as evidenced by my work porting this codebase to use the Espressif HomeKit SDK. The ESP8266 is (in my opinion) simply not powerful enough to meet the use cases of this firmware.

I wrote a blog post about it, announcing that I'd be retiring from supporting this firmware, and recommending alternatives:

### [On Abandoning my RATGDO Native HomeKit Users](https://thenewwazoo.github.io/bye-bye-ratgdo.html)

Not every maintainer of this repo shares my pessimism! @jgstroud deserves crazy applause for his work to improve reliability, and he's previously stated that he's going to keep working on it. Choosing to stick it out is a reasonable option, so long as it's done after having expectation set accordingly.

billy27607, 2024-03-02:
@thenewwazoo,

I certainly understand and appreciate the position that you are in. Sometimes when the juice ain't worth the squeeze enough is enough. There is working solution for those of us who have purchased the boards.

I for one, appreciate all of the effort that you put into this. You should not worry about anything at all.

dkerr64, 2024-03-03:
The no-response error occurs for me "reliably" if the ratgdo is left to run for ~3 days, at which point it spontaneously reboots and on startup believes that no HomeKit device is paired.  If I do not pair HomeKit and leave the ratgdo running, it runs for many more days (I don't know how many, I reset it after 6 days).

To me this smells awfully like a memory leak within the HomeKit code.

It will likely take time to find the cause of the problem, or may never be found.  But what works reliably for me is to reboot the ratgdo before the error occurs.  We could do this automatically.

In the PR #117 that is awaiting merge, I have added an automatic reboot every 24 hours (by default, configurable between 1 and 72 hours, with 0 meaning never).  This is of course a nasty hack / work-around... but you would be surprised at the number of devices I have had to set to automatically reboot on a regular interval... not necessarily daily, but maybe weekly... commercial supported devices like cable modems, routers, etc. that would hang for no apparent reason, but work great as soon as rebooted.

iyerusad, 2024-03-03:
> @billy27607 I believe you're basically correct, as evidenced by my work porting this codebase to use the Espressif HomeKit SDK. The ESP8266 is (in my opinion) simply not powerful enough to meet the [...]

Completely understandable and a big thank you for all the hard work, this clearly is what a lot of people were interested in and you definitely got it quite far.

If there are any more expensive options on market would be helpful to list those in the blog post for those who prefer not to trade one buggy tool for a time-consuming homelab setup. Likewise, a fancier high end RatGDO that has more powerful SOC/memory/etc. would be also nice if it exists.

----

There is some curiosity/merit into trying to conclusively pin down what produces a "working" unit vs a crashy/buggy unit and/or root cause. It irks my brain to think there are, for a lack of a better word, 'golden' units vs general units, like an math formula that shouldn't work but does, and consistently. 

**Does the flasher/flashing method matter?** This might be a common denominator that would explain larger majority of people having issues vs handful of people without this issue. I don't personally know if the initial flash on this platform matters, but my understanding is initial firmware flash is different than firmware upgrade.
- To get on the HomeKit-RatGDO, I believe I flashed using a locally cloned version of https://esphome.github.io/esp-web-tools/
  -  Have to edit a couple of lines that reference the firmware file to get it working, as well as providing actual homekit firmware bin and manifest file
  - I think I tried another flasher app at some point but it didn't do wifi provisioning so I moved to the ESP Web Tools option
- My guess is majority of people are flashing using Paul's hosted ESPHome flasher and then manually flashing from web interface there. I did run into the wifi bug multiple times using Paul's flasher but I think that is in general a ESPhome bug.
- My very very limited understanding is the flasher does include some helper components from wifi and those versions may differ from what's [hosted on Paul's site](https://ratgdo.github.io/esphome-ratgdo/) vs the latest from the [repo](https://esphome.github.io/esp-web-tools/)

Probably nothing but I think that covers all the possible variations my consistently working build/setup could have from majority of everyone else's builds.

edrikk, 2024-03-04:
I’ve been reading tonight, and this seems like something to consider given that it’s a further optimization (with a focus on ram) vs what thenewwazoo attempted in the other branch:

https://www.blessuisse.net/?_=%2FJKoss2%2FArduino-HomeKit-ESP8266%23KJWqMdlUlBnsIvkdRR%2BuhIT4

Some interesting reading there as well within the readme…

[11 more replies]

closed 2024-03-19 by jgstroud