ratgdo/homekit-ratgdo #97: Unable to Add Accessory: The setup code is incorrect.

nickv2002, 2024-01-28:
I just use the flash page to upgrade my `ratgdo_v2.5` device to `homekit-ratgdo 0.10.0`.

I am attempting to pair it with my home via pointing my camera at QR code shown in the webUI. Home.app starts walking me through the adding process and warns me about the the unverified accessory. I continue through that and after a short delay it errors out with the message:

Unable to Add Accessory: The setup code is incorrect.

nickv2002, 2024-01-28:
Hmm, I tried downgrading to 0.9.0 and got the same results, but it _was_ working with 0.9.0 in the past so I'm confused about what is wrong now. I have no problem flashing or navigating to the web UI. I attached the logs (but they don't output anything when I'm trying to pair from Home.app).
[esp-web-tools-logs.txt](https://github.com/ratgdo/homekit-ratgdo/files/14078115/esp-web-tools-logs.txt)

jgstroud, 2024-01-29:
I've had this issue a few times, and I'm not sure it's always the same issue.  I've definitely seen this when my wifi signal was marginal.  The pairing process seems to be the process most susceptible to poor quality wifi.  I assume you have already made sure to remove the device from the Home app.  You might also hit the un-pair option again in the web ui.

nickv2002, 2024-01-29:
Thanks @jgstroud 

I'm within eyesight of my access point so I don't think signal strength is the issue (the web UI comes up quickly).

I did remove the device from my Home.app, then hit the unpair option in the UI, restart the rated, retry but still get the same error.

nickv2002, 2024-02-04:
Update, I was able to set up "ratgdo v2.57 for hardware v2.50" using the [MQTT firmware](https://paulwieland.github.io/ratgdo/flash.html) and this [Homebridge ratgdo plugin](https://github.com/hjdhjd/homebridge-ratgdo#readme). This seems to be working well enough but I'm still confused why this HomeKit adapter doesn't work.

Some things I've also tried that still all produced the same error about an incorrect setup code.

- A variety of SSIDs for my home Wifi, some are 2.4Ghz-only some are 2.4Ghz+5Ghz.
- Different ratgdo hardware (a brand new 2.45i board).
- Creating a new Test home in the Home.app and adding the HomeKit adapter there.

Since these all failed in the exact same manner while the MQTT firmware works without issue, this leads me to believe there's some bug in this homekit-ratgo firmware that is incompatible with some part of my home. I don't have any good idea of what to try next, but I would still like to switch to using this project's native HomeKit firmware, so please let me know what other debugging I could do to help diagnose this issue.

jgstroud, 2024-03-20:
@nickv2002 would you be willing to test this bug with the latest release?

jgstroud, 2024-03-20:
Actually, never mind @nickv2002 .  A user just reported a similar issue on discord.  It seems like sometimes you have to try several times before it pairs.  And the problem seems to occur most after a full reflash of the device.  I'll try to reproduce locally.

jgstroud, 2024-03-20:
Recent report on discord says:
> got this twice when trying to add. The page seems to be working though.  I’ll try to reboot
>  "Accessory Not Found"
> after reboot i got this instead:
>  "Unable to Add Accessory"
> Ok fourth time is a charm. It added but took a while.  I’ll go test it out now

Will try to reproduce locally.

nickv2002, 2024-03-20:
Thanks for checking in. I saw the new firmware had some HK related changes so I was planning to retry with it again soon.

I have definitely tried more than 4 times with older firmware version but maybe not sequentially. 😆 I can try again on the newest firmware at some point soon, but likely not in the next week because of other personal obligations.

jgstroud, 2024-03-20:
@nickv2002 no worries.  I think this is probably reproducible, so I'll try to recreate and hopefully find a fix.

jgstroud, 2024-03-22:
Pairing reliability improved in #135 
That said, I just helped someone that was hitting the issue reported here and it was the HomeBridge mDNS client spamming the device with oversized corrupt mDNS packets.  Make sure you don't have Bonjour-HAP enabled on your network.  The webUI was only showing <5k free heap and was getting an OOM error right away during pairing.

nickv2002, 2024-03-22:
I do have Homebridge with Bonjour-HAP enabled! I will try with a different mDNS Advertiser to see if that changes anything next time I test with the new Ratgdo HK firmware.

jgstroud, 2024-03-22:
@nickv2002 I recommend using avahi

seanwalter, 2024-03-24:
I am facing this issue when HomeBridge is running but it doesn’t matter which mDNS advertiser I use. I have tried avahi, bonjour-HAP, ciao and the experimental systemd-resolved. If I shut down HomeBridge, the ratgdo on HK firmware 1.0.0 seems stable and responsive. But the minute I start HomeBridge again, even accessing the ratgdo web UI becomes unreliable.

seanwalter, 2024-03-24:
Quick update…today I enabled IGMP Snooping on all VLANs on my network. That plus using avahi as my HomeBridge mDNS advertiser has given me “pretty reliable” access to the ratgdo web UI for the past hour or so (previously, access would become unreliable within several seconds of restarting HomeBridge, so this feels like a fix — but I will update in the next few days if stability drops over time).

I should note during my troubleshooting I also did two other things: I fixed an issue with a different network device that HomeBridge couldn’t access and was polling every ~20 seconds in the logs, and I enabled WiFi 6 on one of my WiFi networks where I hadn’t done so yet. I don’t think either should have been affecting the ratgdo, but am mentioning just in case.

jgstroud, 2024-03-25:
Curious to hear more on how your testing goes.  It seems apparent you can kill the device by overwhelming it with mDNS traffic.  I'll see if there is anything we can do in the mDNS stack to keep from overflowing our memory with mDNS packets.

seanwalter, 2024-03-26:
One more update…on a whim last night I disabled the Homebridge _Airport Express Connected_ plugin, which had been throwing errors due to some other issues with my home setup. I was guessing it might be the source of excessive mdns traffic. Since disabling the plugin, the ratgdo has been highly responsive.

Sharing this as a possible workaround for others facing this issue…you might try disabling Homebridge plugins one at a time to see if you can get a temporary solution that way.

jgstroud, 2024-03-26:
That's good feedback.  I'm having trouble reproducing the issue now, but I have a build that might help prevent this problem in the future.  I just have no idea if it works or not.  I need a beta tester.

seanwalter, 2024-03-26:
Happy to test

jgstroud, 2024-03-26:
> Happy to test

I just put a test binary in my local branch.  You can grab it here: [homekit-ratgdo-v1.0.1_debug.bin](https://github.com/jgstroud/homekit-ratgdo/raw/mdns_debug/docs/firmware/homekit-ratgdo-v1.0.1_debug.bin)

seanwalter, 2024-03-28:
I’ve been running that binary for ~26h and I’m getting very high reliability. I’m judging this based on web UI availability upon refresh and not HomeKit availability itself, but the two have seemed pretty highly correlated.

The web UI is still more stable and responsive with my errant Homebridge plugin completely disabled, but with the plugin enabled I get an unresponsive UI about once in ~40 refreshes when hammering it from my phone browser. And, in cases when the UI is not responding, it reliably comes back within a few seconds.

if helpful I have noticed the freeheap/minheap have settled around 16,000/6,000 after several hours. For the first few hours after installing it was more like 19,000/16,000. With the Homebridge plugin disabled it’s more like 20,000/1,000.

closed 2024-04-30 by jgstroud