ratgdo/homekit-ratgdo #98: Notify HomeKit of garage door open status after door is fully open

leesonthenet, 2024-01-28:
Current code triggers a notification from HomeKit that the Garage Door is Open as soon as the gdo button is pressed.  The door is still in the opening process.  The notification should happen after the door is finished opening.

However, the code does seems to work correctly for closing the door. When the door is open, and the gdo button is pressed, HomeKit does *not* notifiy you that the Garage Door is closed until it is actually fully closed. This is working as intended.

Caesar113, 2024-01-28:
Just curious… why do you prefer this and think that it’s the “intended behavior?” 
#“The notification should happen after the door is finished opening.” 
As another non-contributing user, I prefer the current notification behavior, because I’d like to know ASAP that the door is opening when I’m not home.

leesonthenet, 2024-01-28:
HomeKit doesn't have a "Garage Door is opening" notification I believe. I'm referencing intended behavior more of a past precedent.  Before MyQ messed up their APIs, they would notify by their app (or via Homebridge/MyQ) the door was open as soon as the door sensor detected it was open.  Theoretically, this would work if you manually opening the door with the gdo and as soon as it was fully open, MyQ would trigger to HomeKit/MyQ app that the door was open.

chrisparana, 2024-02-03:
I'm pretty sure HomeKit has an "Opening…" and "Closing…" status. It should show "Opening" or "Closing" during that time and "Open" or "Closed" when it's complete.

PaulWieland, 2024-02-05:
The HomeKit notification is based on the start and end of the physically insecure state of the door. 

Closed = Secure. The other three states are not secure.

Therefore `Opening` is the start and triggers the notification. Closed is the end and triggers the follow up notification.

blueyetisoftware-jason, 2024-02-15:
Totally agree. Partially open = open. If the door is stopped halfway, a status of open makes the most sense from a security perspective.

thenewwazoo, 2024-02-15:
In any case, it’s done at the HomeKit app level and out of our control :). I’m going to go ahead and close this ticket since it works as intended.

closed 2024-02-15 by thenewwazoo