ratgdo/homekit-ratgdo #103: Can't connect to WiFi after v0.10.0 OTA update

nyjklein, 2024-01-30:
I'm not sure this belongs here. But I'm not sure where else to go. So ...

I've been running on v0.9.0 and it's been completely stable and reliable with no interruption for over 9 days. Yesterday I upgraded OTA to v0.10.0 using the web page. While everything appeared to go fine, when the device came back up it wasn't connecting to my non-mesh, single router, 2.4 Ghz only Wi-Fi network.

After waiting for quite some time I disconnected the ratgdo from the garage door and brought it inside close to my router. That didn't help. Here's where it gets weird. I connected over the serial connection from my PC (tried both Win 11 and Mac OS with the same results). The flasher tool works fine and I re-flashed the firmware (tried with and without erase). But when I get to the page to connect to Wi-Fi, even though it lists my network in the dropdown, the connection always times out no matter how many times I try. To eliminate my router, I tried two other 2.4Ghz networks on separate routers. Same results, timeout!

Is there some process to completely reset the board? Is there an older flasher I can use to try to load the older v0.9.0 code? Any other ideas?

Jeff

rogersgw, 2024-01-30:
Had this same issue with the latest version, but the device was actually connected.   

There is a problem with the firmware not reporting back a connected WiFi signal.  I had to hard reset the device and reconnect to Apple Home. 

Try to ping the IP of device.  You can find the device name loaded in the router table to get the IP address. I reserve the IP so I can ping from my phone to verify connectivty.

nyjklein, 2024-01-30:
That's not my issue. It's not even connecting to the router. I see no attempts in the router log. Can't ping what should be its DHCP reserved ip address. Something really weird is going on. 

BTW, what did you mean by a "hard reset"?

simar7, 2024-02-02:
Same issue. Brand new ratgdo 2.53i and wifi cannot connect.

simar7, 2024-02-02:
FYI the other firmwares for ratgdo don't seem to have this issue. I've tried MQTT and ESP both and they're fine.

weyemar, 2024-02-02:
> Same issue. Brand new ratgdo 2.53i and wifi cannot connect.

Same here. Brand new board and I cannot use it. Every once in a while I can get it to come online and access the web GUI. But then when I try to connect it to the garage, it never seems to come online. Spent a couple hours troubleshooting.

simar7, 2024-02-02:
The best workaround at the moment seems to be to flash the MQTT firmware and the setup homebridge-ratgdo plugin if you happen to have homebridge.

weyemar, 2024-02-02:
> The best workaround at the moment seems to be to flash the MQTT firmware and the setup homebridge-ratgdo plugin if you happen to have homebridge.

I tried that, the device gets a 169. address and when I reboot it, I cannot access it. I see my DHCP server gives it an address but it just doesn't work. idk if I have a bad device or what.

simar7, 2024-02-02:
> I tried that, the device gets a 169. address and when I reboot it, I cannot access it. I see my DHCP server gives it an address but it just doesn't work. idk if I have a bad device or what.

Do you have a wifi access point name that has any special or non English characters?

weyemar, 2024-02-02:
> Do you have a wifi access point name that has any special or non English characters?

Nope. I am thinking of booting up an older AP to see if the ratgdo just doesn't like my equipment. It has been on the wifi once or twice during all the hours of trying to get this to work. It's just never able to be reliable.

simar7, 2024-02-02:
> > > > The best workaround at the moment seems to be to flash the MQTT firmware and the setup homebridge-ratgdo plugin if you happen to have homebridge.
> 
> > > 
> 
> > > 
> 
> > > I tried that, the device gets a 169. address and when I reboot it, I cannot access it. I see my DHCP server gives it an address but it just doesn't work. idk if I have a bad device or what.
> 
> > 
> 
> > Do you have a wifi access point name that has any special or non English characters?
> 
> 
> 
> Nope. I am thinking of booting up an older AP to see if the ratgdo just doesn't like my equipment. It has been on the wifi once or twice during all the hours of trying to get this to work. It's just never able to be reliable.
> 
> 
> 
> Additional deals about my wifi. I do have two APs, 2.4Ghz and 5Ghz are separate SSIDs. They're TP-Link, WPA2/3, auto channel width, one AP is channel 1 and the other 11.

Oh good point! I forgot to mention that I had to create a separate 2.4Ghz only access point for it. 

The dual 2.4Ghz and 5Ghz AP wouldn't work for some reason. Maybe band steering was the cause.

nyjklein, 2024-02-02:
Just to re-iterate my issue. My ratgdo was working JUST fine for weeks on my existing router setup until I tried updating to v0.10.0. Since I did that, I can't get it to even attempt a wifi connection. Just timeout or can't connect. Don't even see it attempt to get an IP address from the router. Tried HK-ratgdo, HB-ratgdo  and even ESPHome with same results. 

I don't know if my device somehow failed. 

As a last ditch effort, I've ordered a new ratgdo board. Hopefully that one will work.

nyjklein, 2024-02-05:
Received my newly ordered ratgdo board today. Set it up and installed it just fine on my network using a fresh install of v0.10.0.

Works fine! I guess the old board went bad.

nicklayb, 2024-02-05:
Same issue for me. Fresh open box, I received it about an hour ago and been trying to connect to Wifi ever since, only getting Timeout.

My Wifi network has two seperate bands 2.4ghz/5ghz. I have a PiHole but I disabled it during the install (just in case)

weyemar, 2024-02-05:
> Received my newly ordered ratgdo board today. Set it up and installed it just fine on my network using a fresh install of v0.10.0.
> 
> Works fine! I guess the old board went bad.

Mind sharing details about your wifi? Mine _has_ worked up until I power cycle it, then it never seems to come back my wifi. Have you power cycled a couple times to see if it's reliable?

I pulled out an old TP-Link Archer A7 just to see if I could get it working and still had no luck with the HK build. I did end up getting the MQTT firmware to work on this old router but still not my normal wifi.

One thing that may be unique about my network is I have two APs that broadcast the same SSID for my 2.4Ghz band. Maybe the device gets confused seeing two broadcasts for the same SSID? Though all my other cheap ESP and Chinese 'smart' devices don't seem to have any issues.

> I have a PiHole but I disabled it during the install (just in case)

I also use PiHole on my network. I use it for both DNS and DHCP with "DHCPv4 rapid commit (fast address assignment)" enabled. Maybe a commonality that could be an issue?

simar7, 2024-02-05:
FWIW, disabling dual band broadcast and limiting to only 2.4Ghz seemed to fix this for me.

nicklayb, 2024-02-05:
> I also use PiHole on my network. I use it for both DNS and DHCP with "DHCPv4 rapid commit (fast address assignment)" enabled. Maybe a commonality that could be an issue?

I only use it for DNS on my end

> One thing that may be unique about my network is I have two APs that broadcast the same SSID for my 2.4Ghz band

I'm using a 3-router Asus AiMesh, I wonder if that is related. If only I could just stick an ethernet into and voilà

matthewprenger, 2024-02-06:
I just converted three ratgdos to the homekit firmware. They had previously connected without issue using the esphome firmware. All three reported the timeout error during configuration, but I found that they did actually connect to the WiFi network and were pairable in HomeKit. The flash/configuration tool just wasn't able to detect this.

2.4 and 5 GHz networks using the same SSID, Unifi UAP-AC-Pros

nicklayb, 2024-02-06:
Surprisingly, this morning my ratgdo is connected to Wifi 🤔. I think I better understand the following from the README

> There is currently a known issue where the RATGDO will timeout on connect. Just retry the connect. This is only known to happen on initial provisioning.

I'm guessing, once you attempted one connection, it keeps the wifi info in there and just wait for it to be connected. I just open the flasher this morning hoping to try again and saw a new option "Visit device" which shows the devices info (ip and all).

vassilmladenov, 2024-02-25:
Does anyone have the 0.9.0 binary link handy? Mine worked for a day but now disconnects and reconnects every minute, curious to try the old firmware before breaking out Homebridge.

jgstroud, 2024-03-19:
Fixed in #129

nicklayb, 2024-03-20:
I might be the unlucky one but I'm still experiencing the "Unable to connect" thing after updating to `1.0.0`, what could I provide to help the troubleshooting?

jgstroud, 2024-03-20:
can you try installing the update from usb and checking the box to erase the flash on update.  then try and connect to the wifi.  it stores some information to flash and I'm wondering if yours is corrupt.  I've also seen that the eero users are still having issues, so maybe still something unique to meshing networks.

nicklayb, 2024-03-20:
> can you try installing the update from usb and checking the box to erase the flash on update. then try and connect to the wifi. it stores some information to flash and I'm wondering if yours is [...]

This is how I updated to 1.0.0 actually, through USB, I tried a direct update plus a reset. I'm indeed in a meshing network (AiMesh)

jgstroud, 2024-03-20:
Sorry, can you describe your issue in a bit more detail.  exact procedure you did that led to the error and what exactly the error is.  Is the problem in the initial wifi config?  We had a buffer overflow that was causing it to fail to connect during the initial config.  I was able to verify this fixed the issue for me.  I'm not sure if it's just another manifestation of the same bug (maybe the buffer is still too small) or something completely different.

jgstroud, 2024-03-20:
I believe the original problem reported in this ticket is fixed.  There was a buffer overflow that I just mentioned that would cause it to always timeout trying to connect to wifi.  So, I definitely don't want to dismiss your issue as I do believe it's a real problem, but I also think the original problem posted here is fixed.  Which is why I wanted to get more details on your problem exactly.

nyjklein, 2024-03-20:
I originally solved my issue by purchasing a new RATGDO board. The new board worked fine with 0.10.0 on the first try (of course with a couple of initial timeouts). The original board was having trouble connecting with MQTT and ESPHome variants also. So something really weird happened to that board. 

Also, some additional details. I do not have a mesh network. It's a single quad band Asus router with the 2.4 Ghz band on its own SSID. It was completely stable on 0.10. for a couple of weeks. I updated to 0.12.0 a few days ago with a network upload. It was also fine. Then yesterday I successfully upgraded to 1.0. via github and it's also running fine.

Thanks for the great support of this.

Jeff

jgstroud, 2024-03-20:
so, wait @nyjklein, I'm reading your comment and I'm not getting to the part where you say everything is broken.  are you actually saying its working?

nyjklein, 2024-03-20:
One last note here. Just for the heck of it, I tried loading v1.0.0 with the flash installer on the old board. Still gets the "Unable to connect" error after trying to connect to wifi. So I guess the board is fried!

jgstroud, 2024-03-20:
@nicklayb wonder if PR #133 might help.  There are some reports on the esphome firmware that this setting helps.

nyjklein, 2024-03-21:
> so, wait @nyjklein, I'm reading your comment and I'm not getting to the part where you say everything is broken. are you actually saying its working?

Sorry for the confusion. As I noted above, I bought a new board weeks ago and that "fixed" my problem but I neglected to post an update here at that time. So with the new board, I'm all good.

nicklayb, 2024-03-21:
@jgstroud I uploaded your PR and it connected instanstly without any hiccups! I might've been lucky but I refuse to believe it's a coincidence.

closed 2024-04-30 by jgstroud