ratgdo/homekit-ratgdo #105: False door open/close notifications from HomeKit

petercockroach, 2024-02-06:
I've recently been getting a lot of false notifications from HomeKit. [image] 

Without logs from ratgdo, it's hard to understand what's happening, but one giant red flag is the uptime from the admin page. Right now it shows it's only been up for 4 minutes which doesn't meet my expectations. I only restarted it last week when I upgraded to FW 0.10.0.

If there's any more info I can provide to help debug, please let me know.

petercockroach, 2024-02-07:
I just updated to 0.11.0. I'll see how it goes.

petercockroach, 2024-02-07:
Still persists in 0.11.0. [image]

petercockroach, 2024-02-07:
I erased and re-flashed today and set it all back up. I'm still seeing discrepancies with uptime though so I really get the sense that something is crashing.

jgstroud, 2024-02-07:
I have not personally witnessed this uptime issue, but I'm also not entirely surprised.  we know there are instabilities.  However, I haven't heard of this specific issue, so I might suggest you inspect your setup and verify you have good power.

jgstroud, 2024-02-07:
We keep adding minor features and really no bug fixes at all.  So no amount of upgrading will fix any issues you are having.  There are several folks working on a complete rewrite that would hopefully stabilize the platform.  but that's still a ways away from release.

jgstroud, 2024-02-07:
You can, but it's currently a bit of a PIA.  You have to connect the USB port on the ratgdo to a pc, or even something like an rPI.  then connect to the serial console and capture that output.

jgstroud, 2024-02-07:
I guess I missed the fact that you didn't have this problem before.  would be interesting if you downgrade to the last known good build if you still see the issue.

petercockroach, 2024-02-11:
I tried downgrading and found I was hitting the same problem now with 0.8.0. I managed to get logs from the console and it does seem like something is crashing. 

Another data point - I updated back to 0.11.0. Still had the same issue so I unpaired HomeKit. The ratgdo has now been up for 12+ hours and counting which is a record. 

I wonder if perhaps this is just an issue with a faulty board? Perhaps the processor is getting overloaded?

jgstroud, 2024-02-11:
That's interesting.  So having it paired to HomeKit causes it to crash. Have you tried looking at your network traffic? I've seen some malformed traffic cause issues with the network stack on the ratgdo.

HyruleWarrior, 2024-02-22:
I have hit this same issue but only maybe twice so definitely not happening as frequently as this. But I do notice my uptime never gets above 20 mins or so on either of my ratgdo devices. I have Eero mesh so seems like the same thing @petercockroach and others are seeing in #94 .

petercockroach, 2024-02-22:
I've had to move to HomeBridge recently since this is just not sustainable at the moment.

jgstroud, 2024-02-22:
I'm going to have to buy an Eero network just so I can see what the heck is going on.  seems like just about anyone with Eero has a problem and its really intriguing me

HyruleWarrior, 2024-02-26:
Not for me, I have had it off since I set up the network.

HyruleWarrior, 2024-03-02:
@jgstroud is there any serial logging around this that would be useful? If so I can capture logs for some time and grab a timeframe when the device uptime resets.

jgstroud, 2024-03-20:
@petercockroach would you be willing to test this bug with the latest release?

petercockroach, 2024-03-20:
I have client steering off. I just bumped to 1.0 yesterday and I'm living on it now (see my comments in

petercockroach, 2024-03-24:
Small sample size, but so far I have not seen this issue on 1.0.

closed 2024-04-30 by jgstroud