ratgdo/homekit-ratgdo #106: Can't add second RATGDO - accessory already added to this home

chbuilds, 2024-02-07:
Firmware: 0.10.0

We have 2 RATGDOs. Something went a bit haywire a few days ago and both were showing No Response. I unpaired both from HomeKit in the RATGDO web interface and removed them from the home app. I can now add one, but the other displays the attached screenshot error - Accessory Already Added / Garage Door has already been added to this home. 

I've tried downgrading to 0.9.0, reupgrading to 0.10.0, tried rebooting, tried unplugging and replugging.

[image]

bob3k3, 2024-02-08:
Experienced this in going from .8 to .9 releases. I flashed the device several times and then also removed it from my HomeKit and I was able to re-add it. In the beginning, I thought it was never going to happen. 

FYI there is a .11 update. for me - it is rock solid - but I only have one door.

donavanbecker, 2024-02-08:
Are you just scanning the QR code? Or are you click passed the camera and selecting the device?

chbuilds, 2024-02-09:
First, THANK YOU for working on this project. It's awesome. To answer your question, I tried both the QR code and manually clicking through to enter the code. Same issue. I just took it down, erased it and refreshed to 0.11 and reconnected. It seems to be working now. I'm going to close this as can't repro but please feel free to let me know if there is anything I can do to help. You all rock.

closed 2024-02-09 by chbuilds