ratgdo/homekit-ratgdo #108: README.md not updated with releases

KevinRemhof, 2024-02-07:
Suggestion: if you make changes in a release such as adding a login, please update the README.md file with the information. The release notes for v0.11.0 do specify the default username and password which is necessary. The README.md does not have this info yet.

And yes, I know I could put in a PR to address this. :)

Damien514, 2024-02-07:
I tried "admin / password" and it worked. Clicking the setting icon, I was able to update to my own password.

KevinRemhof, 2024-02-07:
Yes, admin/password works. My issue is that the README.md should have been updated to specify this. You had to read the release notes to find that.

My suggestion is an improvement to the overall process when creating a new release.

jgstroud, 2024-02-07:
Yeah, this occurred to me right after I merged the PR that we should have included a README update with it.

jgstroud, 2024-02-07:
LoL.... Then shortly after that I remembered that @donavanbecker loves updating the readme file and anticipated a fix by the next morning.

donavanbecker, 2024-02-13:
> LoL.... Then shortly after that I remembered that @donavanbecker loves updating the readme file and anticipated a fix by the next morning.

Done: [#115](https://github.com/ratgdo/homekit-ratgdo/pull/115)