ratgdo/homekit-ratgdo #109: Way to be notified there has been software update without checking website.

bob3k3, 2024-02-08:
Noticed in the .11 update it states  "Automatically detect new releases on github". Not sure what this means but it would be nice to somehow be able to know when updates are released rather than having to go to the site to look. Not a big deal but just curious if it's possible or there's something I don't understand about Github.   My device is rock solid. Very impressed and appreciative. TIA

SShah7433, 2024-02-11:
Not ratgdo specific, but you can enable notifications in GitHub for releases. To do this:

1. Navigate to the landing page for this repository: https://github.com/ratgdo/homekit-ratgdo
2. Click the **Watch** button, click **Custom**, then check **Releases**, followed by **Apply**.

You can manage how you receive these notifications in the Settings section of your account.

bob3k3, 2024-02-11:
Thank you so much!

dkerr64, 2024-02-18:
This should be closed.  Nothing to be done in the code.

closed 2024-02-18 by bob3k3