ratgdo/homekit-ratgdo #110: Change Wifi SSID with Flashing Board

grouchystitch, 2024-02-08:
Good Morning. Could there be an option to change the SSID / Network without having to flash the board? I am trying to segregate my IoT Devices to a different SSID and can't change the RATGDO wifi unless I flash the complete board.

thank you.

blueyetisoftware-jason, 2024-02-08:
There is a way to access settings by visiting the IP address, but they are behind some sort of username and password. I don’t recall setting those up, so they may be defaulted in the firmware

grouchystitch, 2024-02-08:
I can access the device via the web with the IP Address but there is no option to change WIFI settings.

dkerr64, 2024-02-18:
How strongly do people feel about this?  I could add it to the settings page,

grouchystitch, 2024-02-19:
> How strongly do people feel about this? I could add it to the settings page,

It would be amazing if possible . Thank you for your amazing work and support.

dkerr64, 2024-02-19:
> > How strongly do people feel about this? I could add it to the settings page,
> 
> It would be amazing if possible . Thank you for your amazing work and support.

The user interface side is easy enough now that I have settings page.  I need to find out where the SSID/password are stored though.

mitchjs, 2024-03-03:
its done by a sevice called INPROV
the webpage you loaded the firmware.. .you connect to the board... and you should be able to change wifi

dkerr64, 2024-03-03:
> its done by a sevice called INPROV the webpage you loaded the firmware.. .you connect to the board... and you should be able to change wifi

The problem is that this requires connecting the ratgdo board to a computer with USB cable.  Once installed at the GDO it is kind of awkward to have to go out to the garage with your laptop and a long USB cable... or to disconnect the ratgdo and bring it inside to flash.

While likely rare, I can see the need to be able to update WiFi settings without having to connect to USB... ie, from a web browser.

mitchjs, 2024-03-04:
thats true... maybe we can put it in the web, course one makes a mistake... it prob needs to timeout.. maybe it can default back
else then one would need the USB cable

jgstroud, 2024-03-19:
Don't think we will implement this in the webUI, but you can do this using Improv by vising the web flash page in chrome.  You need to be connected to the device over USB.
https://ratgdo.github.io/homekit-ratgdo/flash.html

closed 2024-03-19 by jgstroud