ratgdo/homekit-ratgdo #111: Remote lockout isn't firing when controlled from HK

blueyetisoftware-jason, 2024-02-08:
When engaging/disengaging the remote lockout from the wall pad, the lock state is changed and a lock notification is sent for the door. When engaging/disengaging the remote lockout from a scene or via HK controls, the lockout does not occur.

To create the scene and directly control the lockout, I used the Eve app. I did try to fire the scene from both Eve and the Home app with the same result of nothing occurring.

Hardware Version: v2.5
Software Version: v0.11.0

jgstroud, 2024-02-09:
Did you by any chance recently upgrade firmware?  I recently updated mine and the door and obstruction sensors worked fine, but the light and lockout did not.  I unpaired and repaired it and everything works again.

blueyetisoftware-jason, 2024-02-09:
This is a brand new unit that was flashed with 0.11 when I installed it. I did verify that the remote lockout is actually working if you try to use a standard remote. The wall panel is just not showing that it is locked. The wall panel still says that the lock is disabled. Is this consistent with the current expected behavior?

jgstroud, 2024-02-09:
ive had mixed results with what my wall panel shows.  probably depends on when it checks that status.

blueyetisoftware-jason, 2024-02-09:
Makes sense. I'm guessing they assumed it would only ever be enabled from the wall panel and aren't expecting it to occur from the GDO itself.

mitchjs, 2024-02-12:
where in HomeKit do you enable/disable the LOCK (lockout)? i cant seem to find it
do i have to have homekit hub? me just using iphone/ipads

thanks
mitch

blueyetisoftware-jason, 2024-02-12:
> where in HomeKit do you enable/disable the LOCK (lockout)?

Apple Home doesn't expose it. You need to use a second app like Eve to control those. After you setup a scene or routine, Apple Home can run it.

jgstroud, 2024-02-12:
You can also ask Siri to lock / unlock it.

cvkline, 2024-02-18:
Whoa, confirmed that you can say "Hey Siri, lock the garage door" and according to the ratgdo web interface (and the door's own behavior) it definitely works to lock the door...    BUT there's no indication of this in the Home app!!

It seems very pathological that HomeKit allows functions to be accessed via Siri that there is no way to do or even monitor from the app itself.

blueyetisoftware-jason, 2024-02-20:
> Whoa, confirmed that you can say "Hey Siri, lock the garage door" and according to the ratgdo web interface (and the door's own behavior) it definitely works to lock the door... BUT there's no [...]

HomeKit is a framework for home automation/control. The Home app is just one of the many clients that support it. Home chose not to show that feature, but others like the Eve App and Siri do allow it. Each app figures out what they want to support for the user experience, but HomeKit is much more powerful than these apps typically let on.

jgstroud, 2024-02-20:
Is there still an issue here or can this be closed?

cvkline, 2024-02-20:
From what I've read here and discovered myself, ratgdo is properly exposing the "lock/unlock" functionality to HomeKit, and certain apps such as Eve for HomeKit are able to properly use it and the garage door responds correctly.

It's still surprising to me that two of Apple's OWN products (Siri and the Home app) don't have feature parity on this (Siri can lock the door but Home cannot), but that's hardly a problem for ratgdo.

I vote to close.

cvkline, 2024-02-20:
> HomeKit is a framework for home automation/control. The Home app is just one of the many clients that support it. Home chose not to show that feature, but others like the Eve App and Siri do allow [...]

I understand, and I get that HomeKit and the Apple Home app are distinct things. My surprise is merely that two of Apple's _own products_ (Siri and the Home app) don't have feature parity here.

cvkline, 2024-02-20:
> ive had mixed results with what my wall panel shows. probably depends on when it checks that status.

I too get inconsistent indications on the wall panel about the lock state when controlled by ratgdo, not sure if there is anything that can be done in ratgdo to remedy that.  The true state is displayed in the ratgdo web interface, and apps that expose that such as the Eve App.

jgstroud, 2024-02-22:
Maybe I'm misunderstanding, but this sounds like an issue with HomeKit and not the ratgdo firmware.   The fact that you can control the lock from HomeKit sounds like the firmware is fine.  Not being able to control it from a scene sounds like an HK limitation.  Is there something that makes you believe otherwise?

blueyetisoftware-jason, 2024-02-23:
You could definitely be correct. I'm not familiar with what HomeKit does under the hood from a scene and if the actions are identical to manual control. For example, are scene commands received differently from direct controls? I'm not sure, so I'm just reporting the behavior. It could also be that scenes send commands in a different order. Just tossing out ideas, but I can say that actual door locks function as expected in scenes, so this is either unique to garage door lockouts or unique to the RATGDO.

jgstroud, 2024-02-23:
I'm going to close this as it appears to be a limitation of HomeKit.  Per HomeKit gdo service, the lock state is a characteristic of the door.  You would have to expose it expose the lock characteristic as an actual lock service to make it work like other locks.   This is possible with a minor code change, but you would need to do that yourself and rebuild the binary.  You might be able to script up some other automation to control it.  Also, does the lock actually do what you want?  It's not obvious to most users, but the door can still open and close, it just disables any of the classic wireless remotes.

closed 2024-02-23 by jgstroud

blueyetisoftware-jason, 2024-02-24:
> Also, does the lock actually do what you want? It's not obvious to most users, but the door can still open and close, it just disables any of the classic wireless remotes.

Yes. I am only referring to locking out the traditional RF remotes that people have on their visors. Thanks.