ratgdo/homekit-ratgdo #112: Recover or reset password

Landonthi, 2024-02-09:
Wondering if there is a way to reset or recover the password for my RATGDO that is setup to HomeKit. I don't remember setting a password for the device. Im unable to get into the settings or use any of the commands on the device webpage. Any help would be greatly appreciated. Thanks

DesertHillsautomation, 2024-02-10:
I am in the same boat looking for a password I never set up to the IP address of the board for HomeKit. When I click on the gear icon on the HomeKit IP page for the board, it asks me for a user name and password.

idarrenly, 2024-02-10:
Username/password implemented in v0.11.0. 

Release notes says default username/password is admin/password

DesertHillsautomation, 2024-02-11:
That worked. Appreciate the info.

Landonthi, 2024-02-11:
Thanks for the info! This worked for me too. Appreciate it.

dkerr64, 2024-02-12:
We need to update the readme.  And/or allow for no authentication required, and make that the default.

dkerr64, 2024-02-18:
Pull request #113 (and #117... they are identical) provides for the web password to be optional and displays the default (admin/password) on the settings page.

First time this is run the default is to NOT require a password, user has to go into settings and enable it.  Thoughts as to whether that is the correct default?

closed 2024-03-20 by jgstroud