ratgdo/homekit-ratgdo #114: Login dialog disappears after two seconds on windows 11

smcahee, 2024-02-12:
when i try to use the 'buttons' on the device page to control the opener (i.e light on/off) the page asks for the username and password, but the login box disappears after two seconds.  this does not happen when you click the 'gear' icon.  if you log in after clicking the gear and then back out of the password change screen you can control the opener without re-entering the password.   this happens with both chrome and edge on two different windows 11 pcs. it does not seem to occur on an iphone running ios 17.3 using safari or chrome.

jgstroud, 2024-02-12:
I have seen this behavior as well on Chrome.  I don't see this in Firefox or Safari.  @dkerr64

mitchjs, 2024-02-15:
i have fixed this is my fork... its relayed to the post with a Timeout set to 2 seconds
i not a expert web programmer i just added a check to make sure we are authorized before the post
my fork also has Security +1.0
i have only been able to test my gdo which is 1.0, so im waiting for a firend to test it with +2.0 then ill do PR
but i think anyone can get my fork... not sure how github works... very confusing :)

dkerr64, 2024-02-15:
@mitchjs I like what you have done.  I added some comments to your javascript changes.  Can you look at my PR #113 please and consider rebasing to that, we could take your authentication changes into that PR.

I like removing dependencies on outside commands from the build web content python script.  But my PR #113 has notable changes to it... specifically adding versioning to sub-resources pulled into the main html.... it has a multi-search/replace sed command that would need to be replaced with native python.

THANKS !!!!

mitchjs, 2024-02-15:
@dkerr64 , ill take a look at your PR #113
im all for making it better...

mitchjs, 2024-02-15:
@dkerr64 take a look at my fork... i merged your PR #113 
course i dont know really how to do that via git hub...  i just downloaded it, and used file compare and merged it in
and incorporated your crc32 (base64 encoded) to the build webpages script
i think it works right... :)

jgstroud, 2024-02-16:
We should do a proper git merge so the commits all retain proper authorship

jgstroud, 2024-02-16:
I can probably merge everything if you need help

dkerr64, 2024-02-16:
> We should do a proper git merge so the commits all retain proper authorship

I agree we should do that.  But @mitchjs fork is is not ready for that yet.

dkerr64, 2024-02-16:
> I can probably merge everything if you need help

Not yet, needs code review and testing.  I see problems.

dkerr64, 2024-02-16:
> @dkerr64 take a look at my fork... i merged your PR #113 course i dont know really how to do that via git hub... i just downloaded it, and used file compare and merged it in and incorporated your crc32 (base64 encoded) to the build webpages script i think it works right... :)

Hi @mitchjs  I have had a quick look at your code.  You have misunderstood the web file versioning / CRC logic and so your updates to build_web_content are wrong.  You replaced two passes over all the files with a single pass that will not work...  you have to calculate a CRC for each file and save it first.  Then when you replace the CRC placeholder in the html files you have to search on strings that include both the filename and the CRC placeholder.... and do it for every filename.  See how I built the "sed_param" variable.

mitchjs, 2024-02-16:
@dkerr64 , i dont think i misunderstood, i dont think need to do it in 2 passes
i get a crc32 of the file, and put it in file replacing ?v=CRC-32 and turn it in to gz
now if you ask me, that changes the CRC32 of the file, and that value also is put in the webcontent map
the crc is saved in varnames, thats what i use to put in map

dkerr64, 2024-02-16:
Reference this section of index.html...
```
  <link rel="stylesheet" href="style.css?v=CRC-32">
  <link rel="icon" href="favicon.png?v=CRC-32" type="image/png">
  <link rel="apple-touch-icon" href="apple-touch-icon.png?v=CRC-32" type="image/png">
  <script src="functions.js?v=CRC-32"></script>
```
Each of these files have their own unique CRC.  Your code inserts the same CRC for all.

mitchjs, 2024-02-16:
hahah did i put the same crc for each :) hahah
will fix

dkerr64, 2024-02-16:
You need to keep the whole [loop](https://github.com/dkerr64/homekit-ratgdo/blob/dd6fa487c5adc76309d36224e5fefd4e476123fe/build_web_content.py#L48) that builds a map of `file_crc[file]`

You can dump the part that builds the `sed_param`

and instead loop over file_crc where I otherwise used [SED](https://github.com/dkerr64/homekit-ratgdo/blob/dd6fa487c5adc76309d36224e5fefd4e476123fe/build_web_content.py#L82).

mitchjs, 2024-02-16:
course its much harder now... not the 1st loop... the find/replaces :)
for me to novice python programmer

dkerr64, 2024-02-16:
> course its much harder now... not the 1st loop... the find/replaces :) for me to novice python programmer

I'm not a python guru either, but should be simple enough to loop over `file_crc`

mitchjs, 2024-02-16:
YUP :)

dkerr64, 2024-02-16:
Try here.... https://stackoverflow.com/questions/3294889/iterating-over-dictionaries-using-for-loops

mitchjs, 2024-02-16:
i think i got it, testing now... thanks for steering me

dkerr64, 2024-02-16:
Make sure you append the right CRC to the `varnames` list

dkerr64, 2024-02-16:
Looks okay, I will test it later in the day.

Now for the GDO security type...

1. User interface should be in settings.html and not index.html.
2. The function in web.cpp is a toggle, not sure I am a fan of that.

In other words, should the user interface setting be a list of valid options (1.0, 2.0, whatever) which user selects, and when saved, the requested value is posted up to web.cpp.

dkerr64, 2024-02-16:
@jgstroud does GitHub allow me to pull in part of @mitchjs changes and retain his authorship.  I know about cherry picking, but in this case I want to, e.g. just merge the `build_web_content.py` file into my branch, then later turn to adding the GDO security changes as a separate merge?

mitchjs, 2024-02-16:
@dkerr64 , yes i want the settings there... i dont really know html well enough, i just copied what was originally there
i was gonna try to do a dropdown list or maybe we do radio buttons, but on the main page wouldnt look right, so it was a button, and like i said i copied an existing input
if you can do the settings.html for me with the selector... ill will code it in to web.cpp :)
but since there is only 2 options button does work....

and PLEASE by all means cherry pick :) im fine with what ever direction we go

dkerr64, 2024-02-16:
ok, I'll add the user interface to settings.html.  Again will be late today/tonight.

dkerr64, 2024-02-17:
@jgstroud and @mitchjs what do you think the right user interface should be for GDO security type?  We can have either a radio button with two options, or a dropdown list with two options.  And what is the right way to name them, is it "Plus 2.0" or "+2.0" and "1.0" or "+1.0" or "Security+ 2.0" and "Security+"

I've pretty much got the user interface done, just on the fence on what looks best

dkerr64, 2024-02-17:
I went with radio buttons for now.  I have pushed the user interface code into my branch that is in PR #113...
@mitchjs you will not have to touch any of the browser-side files, take everything I have unchanged.  You will need to change a couple of lines in web.cpp to actually change the security type... search on "TODO"

The code defaults to +2.0

mitchjs, 2024-02-17:
@dkerr64 , ok... ill get to it (if wife allows me on puter over weekend... you know the usuall... you spend too much time...)
also... maybe we need a better way to work together on the fork
my friend tested the 2.0 portion of the code, and he said its working fine (aka i didn't break anything)
so maybe after this web merge, do a PR and get it in the official build @jgstroud 
(i whish i knew github better, learn as i go)

mitch

mitchjs, 2024-02-17:
@dkerr64  something wrong... i merged,... i have it set to security  +1.0 so the front page says (and it works as 1.0)
but when i go to settings, the opposite radio box has selected... where does settings.html get set to right selection?
(could be missing)
[image]

i  pushed the changes to my fork

mitchjs, 2024-02-17:
@dkerr64  i added you as a collaborator to my fork, maybe that makes it easier for us to work on it, just a guess since i dont know github well enough at this point in life :)

mitchjs, 2024-02-17:
@dkerr64 , i found the code that sets it, its for sure STYLE... to me the solid one is selected...
just need a CSS tweak

dkerr64, 2024-02-17:
@mitchjs It is being set correctly.  I am not changing how the radio buttons look, it should match the default for your browser. 

I accepted your invitation to collaborate on your branch.  On my local copy of it I have squashed all the commits into one, and rebased it onto my webpage branch.  Now, with your permission, I'll push that up to GitHub.  It will be a "force" push and when I do that you will need to fetch onto your local copy and "git pull --rebase" to sync things up.  The squashing and rebase makes the changes we made much clearer.

The only other question I have is why you changed the upload speed from 115200 to 921600.  Much as I love the increased speed, there was a comment that it needed to be 115200 to support improv (which is how WiFi SSID and PW are initially set).

mitchjs, 2024-02-17:
@dkerr64 
to me the solid white radio box... due to me running dark theme in edge make it look as that was selected
i suggest some COLOR (the gold works) to indicate which is selected.

its the upload speed when sending the code down to the esp8266 when programming.
has nothing to do with the improv etc
just the speed the flashing happens at when programming and developing right from the IDE

ok, your gonna push to my fork? i been using Github Desktop to do the push/pull etc will it know to rebase...

dkerr64, 2024-02-17:
I have pushed to your fork.  You should be able to fetch/pull.  You may have to specifically tell the desktop to rebase, not sure how that client works (I use Sourcetree).  It will likely initially give you an error and tell you that you need to rebase instead.

dkerr64, 2024-02-17:
If you want to play with radio button colors, go into style.css and mess with this...
```
input[type="radio"] {
    vertical-align: middle;
    accent-color: darkgoldenrod;
}
```
Right now the vertical-align setting is there, but not the accent-color.  Play around and let me know what you think works best.

mitchjs, 2024-02-17:
the gold is better, at least i know that is the one selected... its good enough
i ended up deleting local copy and cloning...

dkerr64, 2024-02-17:
SO I started testing your code.  First thing I noticed that it defaulted to Security +1.0 when I was using +2.0. I had to set it to +2.0 before it worked.

I think we should change that default, can you do that?  Logic being that everything out in the wild today is only working with +2.0 and to make sure everyone keeps working on upgrading their firmware, it should be set to +2.0.

I'm loving the faster firmware upload speed!

dkerr64, 2024-02-17:
Bug is in `read_gdo_security_from_flash` where you actually do write 2 to flash if it is not already set, but are returning 1 from the function.

mitchjs, 2024-02-17:
ha change to return secType;

dkerr64, 2024-02-17:
I pushed a fix into my branch.

mitchjs, 2024-02-17:
k, i pushed it to mine... are you ok with the gold?

dkerr64, 2024-02-17:
I pushed it into your branch as well.  Along with the radio button color.

mitchjs, 2024-02-17:
and now i got conflicts, with no idea how to resolve :)
ie use yours
it wants me to do some command line sh*t

mitchjs, 2024-02-17:
ok, i got it figured out... in vscode :) let me chose yours

dkerr64, 2024-02-17:
> k, i pushed it to mine... are you ok with the gold?

yes, gold is fine. Testing now on my iPhone and the formatting of the radio buttons is not ideal, so I need to tweak that.

mitchjs, 2024-02-17:
ok... it never ends :)

mitchjs, 2024-02-17:
id by ok with a single radio button turned in to a switch (via css) i always like how they look    +1.0  (  0 )   +2.0  (bad ascii gfx)

dkerr64, 2024-02-17:
I'm happy with radio buttons for now.  I have fixed the formatting and pushed to both your and my branches, so we are in sync.

mitchjs, 2024-02-17:
looks good to me... now @jgstroud  should i do a PR?

mitchjs, 2024-02-17:
@jgstroud  @dkerr64 , i did the pull request :)

david, if we change my branch does the pull request update? (before merge to his main)

dkerr64, 2024-02-18:
> @jgstroud @dkerr64 , i did the pull request :)
> 
> david, if we change my branch does the pull request update? (before merge to his main)

Yes, any additional commits are automatically included in the PR.  So if you want to start working on additional feature you should first create another branch.

Opening your PR was not necessary... because I added your changes into my branch which already had a PR open.  So @jgstroud will pick one and ignore (close) the other.  Doesn't matter which as authorship attribution remains correct and attached to each commit.

@jgstroud I do think this is now ready to merge.

mitchjs, 2024-02-18:
oops :)
yea.. no new features... but if we found a late issue
i just flashed and played with it, it looks good to me

jgstroud, 2024-03-09:
Fixed in #117

closed 2024-03-09 by jgstroud