ratgdo/homekit-ratgdo #119: Obstruction status not working

HyruleWarrior, 2024-02-22:
So I set up two Liftmaster Security+ 2.0 garage doors a couple weeks ago and everything seems to be working as expected except the obstruction detection. When something blocks the door from closing HomeKit doesn't show any obstruction warning. Checking the device page on my network also shows `Obstruction: false`. Is there anything I can check to troubleshoot why this wouldn't be working? Wiring is correct as far as I can tell from the setup docs:

[image]
[image]

jgstroud, 2024-02-23:
the symptoms sound like the sensors are mis-wired, but your pictures appear correct to me.  I would check that you are actually getting good contact on each wire inside the terminal block.  Also on the GDO side, I can't see any exposed copper, so double check those are actually pinching down on the copper and not the wire covering.  Though, I don't think that side would cause this error.

jgstroud, 2024-02-23:
I actually didn't wire my obstruction sensors into the RatGDO.  I just connected the 3, red, white and black wires.  I think the terminal blocks for the obstruction sensors was just to make connecting everything easier.

HyruleWarrior, 2024-02-23:
>Also on the GDO side, I can't see any exposed copper, so double check those are actually pinching down on the copper and not the wire covering.

If these wires didn't have a good connection wouldn't the obstruction sensors not be working at all? I thought this is just passing through the board to the GDO. I can block the sensors when doors are closing and the doors reverse, it's just that nothing shows up in HomeKit showing obstruction which is what I would expect.

jgstroud, 2024-02-23:
Ah, yeah, that definitely sounds like it's all wired properly.  I just double checked mine to make sure we didn't have a firmware regression, but it seems to be working.  I'll have to think about this a bit more

HyruleWarrior, 2024-02-24:
Are there any logs I can get from device or other things I should check?

jgstroud, 2024-02-24:
Yes there are logs but they can be a bit tricky to get.  It logs to the serial console. So you need to connect the usb port to a PC / laptop / rpi / anything that can monitor the serial port.

HyruleWarrior, 2024-03-02:
Attached logs from a session:
1) Opened door via control panel
2) Closed door via HomeKit
3) While door closing I tripped the sensor
3) Door reversed and opened up
4) Closed door again via HomeKit without obstructing 

I see some evens where `Obst1` and `Obst2` packets are being ignored which seems suspicious:
```
>>> [ 218808] RATGDO: DECODED  00032490 000000308D5B5006 0000C084
>>> [ 218814] RATGDO: PACKET(0x5B5006 @ 0x32490) Obst1 - NoData: [Zero: 0x00000000, Parity: 0xC]
>>> [ 218823] RATGDO: Support for Obst1 packet unimplemented. Ignoring.
>>> [ 218900] RATGDO: reader completed packet
>>> [ 218904] RATGDO: DECODED  00032490 000000308D5B5006 2000F085
>>> [ 218911] RATGDO: PACKET(0x5B5006 @ 0x32490) Obst2 - NoData: [Zero: 0x20000000, Parity: 0xF]
>>> [ 218919] RATGDO: Support for Obst2 packet unimplemented. Ignoring.
>>> [ 218996] RATGDO: reader completed packet 
```

[device-monitor-240301-234245.log](https://github.com/ratgdo/homekit-ratgdo/files/14470287/device-monitor-240301-234245.log)

jgstroud, 2024-06-17:
Recently found an issue causing some characteristics to not update properly.  Fixed in #204 .  Not sure if this is the same problem or not, but it may be fixed in 1.6.0.

jameyers4, 2024-07-23:
Installed a ratgdo v2.5 running fw 1.6.0 last week.  All works very well except for no notification from Apple HomeKit when obstruction occurs.  If I look at device webpage, obstruction briefly flashes true during beam incursion.  So for me, webpage reports (very briefly) correct status.  I'd expect a Home Notification as well at this time.  Please LMK if I can provide more info.

Also - wiring should be correct as when sensor beam incursion occurs, door does reverse, and LiftMaster GDO lights flash as per normal.

dkerr64, 2024-08-10:
@HyruleWarrior please test with v1.7 (tagged as pre-release so select that checkbox on the firmware update dialog)

@jameyers4 v1.7.0 now has setting to trigger the motion sensor if obstruction is detected.  We cannot add a separate "obstruction sensor" beyond what is built in to the HomeKit garage door opener accessory.  There are reports that 3rd party HomeKit apps will let you trigger an automation (which could include a notification?) from the obstruction sensor.

HyruleWarrior, 2024-08-12:
@dkerr64 I actually switched off this firmware so I can't test it out. If I switch back I will post an update

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.