ratgdo/homekit-ratgdo #120: Garage door opens by itself

atmendez, 2024-02-24:
My garage door has opened twice by itself in the past two weeks. The first time I thought it was a fluke but the second time I checked the HomeKit log and it shows that I opened the garage door via HomeKit two minutes after closing it via the external keypad.

I was probably a quarter of a mile away at that point (my phone was probably connected to CarPlay but I’m not certain) and can’t figure out why the door mistakenly received a HomeKit request to open 2 minutes after closing it. Anyone else experience this or have suggestions for how to deal with it?

Running HomeKit native firmware 0.11 on board 2.5i.

jgstroud, 2024-02-26:
Does the HomeKit log just say it was opened, or does it say it was opened by "some user"?  From what I can tell, if the door just opens and the ratgdo reports the event, then it will just say opened.  But if you request that it open from a device, then HK told the GDO to open.

atmendez, 2024-02-26:
HomeKit log says it was opened by some user (in this case, me). I can't tell if this means this is a problem with HomeKit or with RatGDO.

jgstroud, 2024-02-26:
I think this means something you are logged into told HomeKit to open the door, which then told the ratgdo to open the door.  So, this would indicate the Ratgdo just did as requested.

dsp4144, 2024-02-28:
You may want to rule out the possibility of any automations that trigger the garage door to open.

mdreizin, 2024-02-29:
I can verify that it occurred to me. I'm currently on version 0.11.0.

mdreizin, 2024-02-29:
Allow me to share some details—I'm experiencing the problem mentioned in #94, and the garage door is inaccessible through HomeKit. There might be a connection between these two issues.

jgstroud, 2024-03-09:
Closing based on the information reported here, it sounds like HomeKit told the ratgdo to open the door.  If you see something that says otherwise, please reopen.

closed 2024-03-09 by jgstroud