ratgdo/homekit-ratgdo #121: QUESTION: How to pair?

russellquinn, 2024-02-25:
I have flashed my new RATGDO with homekit-ratgdo. I can access the web interface. But, when I tap on "Pair to HomeKit" I get "Pair to new HomeKit, are you sure?" but when clicking OK it just downloads a zero kb file named Reset.

I've tried this on Safari and Chrome on macOS and in Safari on iOS (all devices connected to HomeKit) and get the same thing.

I tried scanning the QR Code from the Home app instead, and this did successfully add Door and Light to HomeKit but they are both immediately non-responsive. Furthermore, the homekit-ratgdo web interface still shows the QR code and "Pair to HomeKit," which from reading the documentation means it doesn't think it's paired.

The buttons in the homekit-ratgdo web interface work fine.

So, my question, what is "Pair to HomeKit" supposed to do? What's going wrong in my flow?

Thanks!

Russell

jgstroud, 2024-02-26:
The problem with it downloading an empty reset file is a bug and will be fixed in the next release.  

It sounds like your pairing failed in some way.  Try removing the device from HomeKit, rebooting the ratgdo and repair it using the QR code.

russellquinn, 2024-02-26:
Thanks for the note. I couldn't get it to pair at all and I tried multiple methods. I ended up reflashing with MQTT and connecting via Homebridge.

jgstroud, 2024-03-20:
@russellquinn would you be willing to test this bug using the latest firmware?

russellquinn, 2024-03-20:
Sorry, I went with MGTT and Homebridge now, which has been super stable for me, so I don't want to switch firmware flavors again.

closed 2024-03-20 by jgstroud