ratgdo/homekit-ratgdo #122: ratgdo forgets that it's already paired to HomeKit

jangellx, 2024-02-26:
After a few weeks, it seems that both of my ratgdo modules forget that they're connected to HomeKit.

They appear in the Home app, but show up as non-responsive.  They are connected to a specific UniFi AP with a static IP, and I can get to the web interface without issue.  However, both show the HomeKit QR code that implies that they do not think they're paired, and the "Pair HomeKit" button is present (as opposed to the "Un-Pair HomeKit" button).

I can re-pair them in the Home app, which adds them as completely new devices.  I then have to set up my scenes and names again so that everything works as it did before.

This is the third time I've done this;  I was running v0.9.0 until today; I've now upgraded to v0.11.0.

To sum up:
- HomeKit thinks the device is paired, but it's non-responsive.
- The device is on wifi and visible to the AP, and the web interface works. 
- The web interface suggests that the device is NOT paired, and is showing the QR code to pair it.
- I can re-pair it to HomeKit as a new device, which then means I have to set it up again in the Home app (scenes, etc).

Has anyone else experienced this?

Thanks!

-- Joe

atmendez, 2024-02-26:
I also continue to have to unpair and pair with HomeKit. Sounds like this is the same issue as here: https://github.com/ratgdo/homekit-ratgdo/issues/94

jangellx, 2024-02-26:
I was wondering if it was the same as issue #94.  What I find interesting is that I don't have to unpair it from the ratgdo web interface first -- it seems to think it's _already_ unpaired.  I can just scan the QR code and add it back to HomeKit (although HomeKit thinks it's a different device).

The fact that it thinks it isn't paired implies to me that ratgdo is itself forgetting that it's paired.  My loose understanding of HomeKit is that it should remain paired even if it's not in range of the network unless explicitly reset.  Or at least that seems to be how other HomeKit devices I have work.

I did find it interesting how many people use UniFi and seemed to encounter issues.  I have had to power-cycle some of my HomeKit wifi devices, including once the ratgdo modules, but most of the devices work, so it's hard to say.

Honestly, I'd like to get everything onto Thread at some point, but it's going to be a while before that's a reality.

atmendez, 2024-02-26:
You raise a good point - my RatGDO is currently listed as "No Response" in Homekit and when I go to the webpage the option to Pair with Homekit is there instead of the option to "Un-pair Homekit."

I just added RatGDO to HomeKit via the webpage and now I have two Garage Doors in Homekit - the previous one that is listed as "No Response" and the one that I just added that I called a different name.

Now my RatGDO webpage has the option to "Un-pair Homekit" as I would expect it to... until the next time it decides to unpair itself!

jangellx, 2024-02-26:
That's the exact same thing I'm seeing.  Glad I'm not alone, at least. :)

logjames, 2024-02-28:
I’ve reverted to v0.10 and it’s no longer occurring. The connection to HomeKit would be lost within a few hours in my case. Restarting doesn’t help.

tmparks01, 2024-02-28:
Same issue. Unifi - no static IP, but locked to AP with strong signal to avoid roaming. Mine would periodically go back and forth from normal to 'unresponsive'. Just reverted to v.10 so we'll see if that helps.

doctorpangloss, 2024-02-29:
I have the same issue. MikroTik, static lease, strong wifi.

Damien514, 2024-02-29:
Same here. I have to re-pair to HomeKit almost once a week. Following this thread hoping to find a solution!

apearson, 2024-02-29:
Had this issue on v0.9.0

a-rodriguez, 2024-02-29:
I am having the same issue also with a UniFi network.

dangerusty, 2024-03-02:
Same issue, unifi extender in garage, tried with and without static IP.

tmparks01, 2024-03-02:
So far (3 days), seems to be pretty solid on v.10. Have not seen the 'no response' issue in the home app.

dkerr64, 2024-03-03:
Someone with powers please close this as a dup of #94  @thenewwazoo @donavanbecker @jgstroud

thenewwazoo, 2024-03-03:
Closing as dupe

closed 2024-03-03 by thenewwazoo

logjames, 2024-03-03:
I don’t think this is a dup of 94. That issue mentions a restart fixes the connection issue. In this case, reboot doesn’t help, only re-pairing the device with HomeKit will fix it, temporarily. It will eventually lose the connection again and need to be re-paired. I have down rev’d to 0.10 and it is stable.

dkerr64, 2024-03-04:
> I don’t think this is a dup of 94. That issue mentions a restart fixes the connection issue. In this case, reboot doesn’t help, only re-pairing the device with HomeKit will fix it, temporarily. It [...]

What you describe is the same as my experience... but it all boils down to the same problem which is why I don't think we need multiple issues to track.  Best to keep all discussion in the one thread.