ratgdo/homekit-ratgdo #124: Obstruction sensor issues

robotron-smith, 2024-03-05:
**Behavior**

With the ratgdo wired in, all functions work in homekit (and with hardware controls), EXCEPT the 'door close' function

When attempting to close the door, it starts down the track for about 1/2 second, then flashes the light and stops as when there is an obstruction

Note that this happens when trying to close the door by any means

- Firmware web page
- Homekit
- wall button
- arrow button on opener itself
- opener remote

In addition to the light flashing, the up and down arrows on the unit flash as well, which indicate a short or missing motion sensor

**Troubleshooting so far:**

- Disconnected ratgo and re-wired as original, in case something had actually misaligned the detectors, or if there was a coincidental break/failure : GDO functioned as expected -- opened and closed without issue

- Assuming it was a wiring issue, I trimmed all of the wires to reduce the chance of a short or cross-wire induction.

**The reason I think this is a firmware issue and not anything else**:

- I re-flashed the ratgdo with the latest firmware for HomeAssistant, reconnected the unit, and the issue did NOT occur : all functions operated as expected

- I flashed AGAIN with the latest Homekit firmware, rewired the unit, and the issue re-appeared.

To be clear on that point: I thought it might be a voltage issue or the like, but the issue disappeared when switching to the Home Assistant firmware, and re-appeared when switching back to the Homekit firmware 

I am happy to provide pics, or logs , or anything else that might help!

**ratgdo info:**
hardware : 2.52
firmware : homekit v0.11.0

**GDO info:**
Liftmaster 1D7675 w/Yellow learn button

jgstroud, 2024-03-09:
Does HomeKit report the door is obstructed?  This is strange because the obstruction sensor logic was copied almost exactly from the mqtt firmware.  Your reasoning does seem to point to something in the firmware though.

jgstroud, 2024-03-09:
I wonder if this is possibly related to #119.

jgstroud, 2024-03-20:
@robotron-smith This is odd.  The code in the HA firmware and HK firmware are nearly identical as we just copied their implementation.  It's a long shot, but we fixed a bunch of memory issues.  I wonder if you could try again with the HK firmware and report back if you still have the same issues.  I'll review and see if anything new was added in the HA firmware and see if we are missing a patch.

walking, 2024-06-30:
have the same issue with version 1.6.0. Everything works well with version 1.4.3, but when I upgrade to 1.5.0, I start to encounter problems. For example, in HomeKit, it keeps showing "closing" even when the door is already closed. So, I upgraded to 1.6.0. This issue is fixed in 1.6.0, but now, almost every time I open the door, it reports an obstruction and opens the door again. I'm downgrading right now to 1.4.3 to see if it resolves the problem.

If I disconnect the ratgdo, the door works fine, even with the wires connected, so there is no contact issue between them.

jgstroud, 2024-07-01:
"almost every time I open the door, it reports an obstruction and opens the door again."

I don't understand this comment.  Do you mean when you close the door?  Can you attach logs?

jgstroud, 2024-07-01:
Did you have the obstruction problem in 1.5.0?  The obstruction code was tweaked in 1.6 to filter out spurious detections. Really need logs to understand what is going on. You can viewlogs.sh script or monitor the logs in the javascript console.

jgstroud, 2024-07-01:
1.6.0 fixed a problem with the wrong status being reported in HomeKit. But I'm really curious about this obstruction regression and would like to try and fix it.  If you could load 1.6.0 and capture the logs, that would be fantastic.   Then you could revert back to the build that works for you until I can release a fix.

dkerr64, 2024-08-10:
@robotron-smith and @walking  please test with v1.7 (tagged as pre-release so you need to check that box on the firmware update dialog).

rcabell, 2024-08-20:
I'm also seeing this exact same behavior with v1.7.0 and a Sec+ 1.0 opener. Light control, sensors, locking, and _opening_ door work fine. If I try to close the door by any means (Home app, device web app, physical controllers) the door moves for less than a second and then the light blinks as if there's an obstruction. 

Changing over to the ESPHome firmware and using Home Assistant to re-export the device to HomeKit works just fine.

igniscyan, 2025-02-14:
I'm getting odd sensor behavior also - after rewiring the sensors are working correctly (interrupts down if obstruction is in the way), yet the HomeKit sensor always shows "No" for obstruction detected. Wouldn't the obstruction sensor just not work if the ratgdo32 was not actually getting the signals?

dkerr64, 2025-02-14:
> I'm getting odd sensor behavior also - after rewiring the sensors are working correctly (interrupts down if obstruction is in the way), yet the HomeKit sensor always shows "No" for obstruction [...]

Is this with Sec+ 1.0 or 2.0 ?

igniscyan, 2025-02-17:
> Is this with Sec+ 1.0 or 2.0 ?

This is in 2.0

HankMardukas19, 2025-04-20:
I’m also having this issue

jgstroud, 2025-06-27:
Did you put an oscilloscope on your obstruction sensor or just measure it with a voltmeter?  Do you know you aren't seeing pulses?

frankea, 2025-06-27:
Nope, don't have one and certainly not gonna go out of my way to buy one for this.

I reported the issue originally over a year ago and nothing has happened since.  I decided to attack it myself with what I had available.  No matter what changes I made in the firmware, ratgdo never picked up any pulses.  Clearly a ratgdo hardware/firmware issue as otherwise the opener works perfectly as if the ratgdo weren't there.  Sensors trip, doors stop closing, etc.  The literal *only* issue I've had is ratgdo thinking the sensors are tripped when they aren't.  This happens across multiple firmwares (as reported both here and in esphome-ratgdo), so my assumption is either this opener is doing something entirely different from other Chamberlain openers, or my hardware has been screwed since day 1.

After finally finding a solution for myself after over a year, I honestly don't care to troubleshoot anymore.  It works for me, and that's all I really care about here.  I figured I'd share the knowledge for anyone else interested, but beyond that, I think I'm done here.

frankea, 2025-06-27:
Alright got the scope and hooked it up between the sensor wires coming into the ratgdo and the other sensor wires going into the GDO.  From what I can tell, this should be theoretically working.  I see pulses on the scope, but when I add debug code to the firmware to log pulse count it stays at 0.

When the sensor path is clear, the signal is normally HIGH at ~6.25V with brief, low-going pulses to ~0.15V. The frequency is ~152Hz, meaning there is a pulse approximately every 6.6ms.

[image]

When the sensor path is blocked, the signal is a steady, HIGH DC voltage at ~6.2V.

[image]

The code in `src/ratgdo.cpp` expects a normally-high signal with low-going pulses and uses a `FALLING` edge interrupt to count them. My sensors are providing this exact signal, so the firmware should be working correctly.

I'm gonna try messing with the thresholds in the firmware, but I'm starting to think more and more this is a hardware issue.

jgstroud, 2025-06-28:
Sounds like you did all the right debug steps. Not sure where you are probing but the next step would be to check the signal on the ratgdo itself.  I agree it sounds like a hardware issue based on your observations.

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.

closed 2025-09-13 by dkerr64