ratgdo/homekit-ratgdo #125: WiFi Timed Out

rickoramaa, 2024-03-07:
Been running into issues connecting the ratgdo v2.53i to my ASUS Router's (RT-AX58U) WiFi. I'll flash the homekit firmware > Connect to WiFi > Timed Out. I've tried updating to the same version when it prompts to see if that would even do anything, but same issue.

I've also tried re-flashing the homekit firmware, but I continue to get the same error. I've tried flashing the ESPHome Firmware as a test, but I get an Unable to Connect error.

I've tried changing the SSID and password to something more simple in case it couldn't recognize a special character (a space or apostrophe), but I'll get the same issue. Both 2.4GHz and 5GHz bands are separated with their own SSIDs.

There were times when it would connect to WiFi. However, as soon as I disconnect it from power and reconnect at the location (or any other location) I would like to install, it would not connect. Pinging IP Address and web interface were giving no response. I've tried reserving an IP for it, but same issue. Even set it up right next to the router, and would receive the same error.

I've seen some reports of users running into the same error, but would see that it's connected on the router and web interface accessible. Unfortunately, that's not the case for me.

Finally, a different PC, micro USB cable, and power brick were used, but did not make a difference for this error.

Any help would be much appreciated!

JonVisc, 2024-03-08:
I agree, haven't been able to connect the ratgdo to WiFi since I updated it past 0.8.0.  Asus GT-AXE11000 for me.

jgstroud, 2024-03-19:
Please do an erase and upgrade to v1.0.  I believe this should be resolved.

jgstroud, 2024-03-20:
Oh, I just saw that you are having the same issue with ESPHome firmware.  Sounds like maybe you have a bad module.   Going to close this since it doesn't appear specific to the HomeKit firmware.  Feel free to reopen if you think otherwise.

closed 2024-03-20 by jgstroud