ratgdo/homekit-ratgdo #126: Reboot caused by exception when HomeKit WiFi client apparently has zero for localIP and remoteIP

dkerr64, 2024-03-09:
Error creating new wifiClient in HomeKit.  For some reason localIP and remoteIP are zero which leads to an exception and reboot...

```
>>> [86950558] RATGDO: PACKET(0x521910 @ 0x2B1E) Motion - NoData: [Zero: 0x00000000, Parity: 0xF]
>>> [86950567] RATGDO: Motion Detected
>>> [86950572] RATGDO: ENCODING 00000EFA 00000000006E8539 00000080
>>> [86950613] RATGDO: writing 0xEFB to file rolling
>>> [86950997] RATGDO: reader completed packet
>>> [86951002] RATGDO: DECODED  0001A757 000000C0BC1A200A 46609481
>>> [86951008] RATGDO: PACKET(0x1A200A @ 0x1A757) Status - Status: [DoorState Opening, Parity 0x9, Obs 1, Lock 0, Light 1]
>>> [86951019] RATGDO: tgt 0 curr 2
>>> [86951022] HomeKit: [Client 1073693196] Got characteristic 1.12 change event
>>> [86951030] HomeKit: [Client 1073693196] Got characteristic 1.11 change event
>>> [86951037] HomeKit: [Client 1073693196] Got characteristic 1.15 change event
>>> [86951044] HomeKit: [Client 1073693196] Got characteristic 1.14 change event
>>> [86955571] RATGDO: Motion Cleared
>>> [86955575] RATGDO: broadcast status change to client IP 192.168.17.241 on channel 0 with new state { "garageMotion": false, "upTime": 86955575 }
>>> [86956395] HomeKit: [Client 1073693196] Get Characteristics
>>> [86960541] RATGDO: reader completed packet
>>> [86960546] RATGDO: DECODED  0001A758 000000C0BC1A200A 42608181
>>> [86960552] RATGDO: PACKET(0x1A200A @ 0x1A758) Status - Status: [DoorState Open, Parity 0x8, Obs 1, Lock 0, Light 1]
>>> [86960563] RATGDO: tgt 0 curr 0
>>> [86960566] HomeKit: [Client 1073693196] Got characteristic 1.12 change event
>>> [86960574] HomeKit: [Client 1073693196] Got characteristic 1.11 change event
>>> [86960581] HomeKit: [Client 1073693196] Got characteristic 1.15 change event
>>> [86960588] HomeKit: [Client 1073693196] Got characteristic 1.14 change event
>>> [86960603] RATGDO: broadcast status change to client IP 192.168.17.241 on channel 0 with new state { "garageDoorState": "Open", "upTime": 86960602 }
>>> [87153960] HomeKit: Got new client: local (IP unset):0, remote (IP unset):0

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4023a24e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000009 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4023a24e in WiFiClient::keepAlive(unsigned short, unsigned short, unsigned char) at ??:?

>>>stack>>>

ctx: cont
sp: 3ffffdf0 end: 3fffffd0 offset: 0150
3fffff40:  00000000 65736e75 00000000 00000000  
3fffff50:  65736e75 00000000 4023b184 00000000  
3fffff60:  4023b184 00000000 3fff4590 4021da8d  
3fffff70:  3fffff40 00000000 00000001 3fff1920  
3fffff80:  3fffdad0 3fff21ac 3fff18f4 3fff1920  
3fffff90:  3fffdad0 3fff429c 3fff18f4 4022ac60  
3fffffa0:  3fffdad0 00000000 3fff18f4 4021e89f  
3fffffb0:  00000000 00000000 00000001 402326d4  
3fffffc0:  feefeffe feefeffe 3fffdab0 401007d5  
<<<stack<<<

0x4023b184 in DhcpServer::isRunning() const at ??:?
0x4023b184 in DhcpServer::isRunning() const at ??:?
0x4021da8d in comms_loop() at ??:?
0x4022ac60 in homekit_server_process(homekit_server_t*) at ??:?
0x4021e89f in loop at ??:?
0x402326d4 in loop_wrapper() at core_esp8266_main.cpp:?
0x401007d5 in cont_wrapper at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```

dkerr64, 2024-03-16:
Closing as fix merged with #127

dkerr64, 2024-03-16:
Reopening as this may not be fixed yet

jgstroud, 2024-03-16:
#129 might help with this one.

Also, I see that the HomeKit library includes its own WiFi library which is also quite old.  I just removed it altogether from my builds.

jgstroud, 2024-03-19:
Fixed in #129

closed 2024-03-19 by jgstroud