ratgdo/homekit-ratgdo #128: Unable to Add Accessory to HomeKit

malotoom, 2024-03-10:
I got the v2.53i board, and installed the Native HomeKit firmware v0.12.0 and using iOS 17.3.1. After flashing the firmware, asked to connect to WiFi, kept refusing the password, but it seems it accepted it since I am able to control the garage door opener through the Dashboard from the Chrome browser. When I try to add the accessory to my Home app, it goes through "setting up" then " adding to home", at that time the QR code on the Dashboard on Chrome browser disappears for few seconds, then comes back, and on my iPhone it say "Unable to Add Accessory", when I try to do that again, it does not allow it, saying "Accessory Already in Another Home". To restart the cycle, I need to reset HomeKit (which I tried few times).

Tried a bunch of things, reset HomeKit (on board), refreshing it again, reboot phone, kill and launch Home app.

One thing is puzzling me, I see old garage door accessory in the Home app as "not responding" from the old days when myQ allowed to connect through my Homebridge device, and I am unable to remove it.

Any thoughts? Tried reading other threads, but not sure if there was an answer.

jgstroud, 2024-03-20:
You might want to try a full erase and upgrade to the latest build.  The initial Wifi connect error should be fixed.  There are a number of other fixes that improve overall stability and responsiveness, so hopefully you can get past the Unabled to add accessory issue.  

Your problem not being able to remove the old device is likely an Apple HomeKit bug, so I can't help you there.

jgstroud, 2024-03-20:
Duplicate of #97 
Feel free to add any additional comments to that issue.

closed 2024-03-20 by jgstroud