ratgdo/homekit-ratgdo #130: Server Sent Events heartbeat race condition

dkerr64, 2024-03-18:
We have a race condition with the heartbeat... which exposes itself when browser client is on slow connection. Server allocates say.. `/rest/events/1` to a browser. Now the browser needs to open up an `EventSource` channel to that URI. But if `SSEheartbeat()` runs before the client has connected, then we get a client not listening error.

mitchjs, 2024-03-18:
@dkerr64 this condition, only happens when the browser is connected and actually viewing the web page of the device correct?
(which is not the case, 99% of the time)

dkerr64, 2024-03-18:
> @dkerr64 this condition, only happens when the browser is connected and actually viewing the web page of the device correct? (which is not the case, 99% of the time)

yes, correct

dkerr64, 2024-03-18:
Reviewing the code, I see a number of problems.

Browser requests URL for SSE subscription by fetch() to `/rest/events/subscribe` it expects a URI in response that looks like `/rest/event/1` and the browser then opens an `EventSource` channel to that.

The server `handle_subscribe()` function allocates a slot in `subscription[16]` array (so we allow for maximum of 16 client connections) it saves the client IP address, socket connection, and constructs a new Ticker() for it.

Problem... the `SSEheartbeat()` function is testing for this client socket being connected.  But that is the connection to `/rest/events/subscribe`. At this stage the client has not connected to `/rest/event/1`.  The `SSEHandler()` function overwrites the saved socket connection in `subscription[]` with the correct one for `/rest/event/1`.  We need a new way to test for connected EventSource

dkerr64, 2024-03-18:
Next problem... if a browser client never opens an `EventSource` channel... or its attempt to do so is blocked by a firewall or proxy (yes I have recreated exactly this scenario in my testing), then two things happen... 1) the browser (my javascript) notices that the connection failed and so it goes into error recovery mode... by trying again, and 2) the server keeps allocating a new slot in its `subscription[]` array, until it runs out (at 16 uses).

We need a timeout at the server side to notice that a client did not register at e.g. `/rest/event/1` before say 5 seconds and release its slot back into the pool.

dkerr64, 2024-03-18:
I think adding a boolean to the struct in `subscription[]` to track whether the EventSource channel is connected yet is a place to start.

dkerr64, 2024-03-18:
I have pushed a fix for the race condition.  I still need to look at adding a timeout to release a slot if it is never used, but that is a far less likely situation.

jgstroud, 2024-03-18:
@dkerr64 I did notice when watching the logs sometimes opening the status page opens and closes multiple channels.

>>> [65466519] RATGDO: Allocated channel on uri /rest/events/0
>>> [65466525] RATGDO: subscription for client IP 192.168.1.89: event bus location: /rest/events/0
>>> [65466545] RATGDO: Allocated channel on uri /rest/events/1
>>> [65466550] RATGDO: subscription for client IP 192.168.1.89: event bus location: /rest/events/1
>>> [65475083] HomeKit: [Client 1073700876] Disconnected!
>>> [65475089] HomeKit: [Client 1073700876] Closing client connection
>>> [65485597] RATGDO: SSEheartbeat - client not listening on channel 0, remove subscription
>>> [65485605] RATGDO: SSEheartbeat - client not listening on channel 1, remove subscription

jgstroud, 2024-03-18:
@dkerr64 Those logs were taken without this race condition patch.  Is this the same issue?

dkerr64, 2024-03-18:
I have seen that too.  And it could be related, but also could be just the nature of the error recovery on the javascript side.  Do you also see two requests for status.json ?

jgstroud, 2024-03-18:
I'm not sure if I saw the multiple status requests.  I really didn't spend much time looking at this issue.  I did notice one time yesterday I closed the web page on my phone and I saw it close 4!! connections.

jgstroud, 2024-03-18:
It doesn't seem to contribute to the instabilities, so at the moment I'm not all that worried about it.  The fact that it properly cleans up all the sessions on close makes me unconcerned.

jgstroud, 2024-03-19:
Fixed in #129

closed 2024-03-19 by jgstroud