ratgdo/homekit-ratgdo #138: I am never asked for my password

Dot-Dash, 2024-03-29:
I have added and saved a password and checked "require password", yet I have never been asked for my password. I can connect and control the GDO with no attempt at authentication.  

I am currently on version 1.1.0, but have been seeing this issue in the last few versions. I have never had any version ask for the password.

dkerr64, 2024-03-29:
Your browser has probably cached it.  Try changing the password and see if you are prompted for it then.

Dot-Dash, 2024-03-29:
I had gone through the steps to change the password many times, but always with the same password. Just now, I tried changing it to something different, and then was prompted to login when I tried to turn on the light.

dkerr64, 2024-03-29:
Yes, that is expected behavior.  I do not know how long it is remembered, it could vary by browser.

closed 2024-06-03 by jgstroud