ratgdo/homekit-ratgdo #140: Add support for Battery Status

jgstroud, 2024-03-29:
Battery Status command 0x09D.
https://github.com/ratgdo/esphome-ratgdo/blob/e3eacef44d6008aeb8a342a2fbb8e75d9cf6c98c/components/ratgdo/ratgdo_state.h#L67

dkerr64, 2024-04-02:
How useful is this?  Adding a battery accessory to the door should be possible, many HomeKit devices are battery operated and report state of battery.  But 99.99% of the time the battery is going to report fully charged, it only discharges if there is a power failure.

And if there is a power failure it is likely that the ratgdo will not be active (although I suppose you could power the ratgdo through a USB power brick that has passthrough capability?).  So it may be that there is only a short period after power is restored when the battery state may report something other than full.

What would be useful is a way to know if the battery needs replaced, but it is not clear that this is reported... based on the code at the above link.

All that said, I do see [09D] messages in my serial log flagged as unknown.  So as a start we could start decoding those and see what all is reported under various conditions.

jgstroud, 2024-04-02:
It's not really useful other than maybe indicating you have an issue with your battery.  It's more about having a cool new feature rather than adding something useful 🤣

dkerr64, 2024-04-02:
I'm all for cool features.  Let's start by decoding the messages and go from there.

dkerr64, 2024-04-03:
I regularly get the following in my log...

```
>>> [ 406800] RATGDO: reader completed packet
>>> [ 406804] RATGDO: DECODED  0001B10E 000000C0BC1A200A 0006209D
>>> [ 406810] RATGDO: PACKET(0x1A200A @ 0x1B10E) UNKNOWN - Unknown: [09D]
>>> [ 406817] RATGDO: Support for UNKNOWN packet unimplemented. Ignoring.
>>> [ 489615] RATGDO: reader completed packet
>>> [ 489619] RATGDO: DECODED  0001B10F 000000C0BC1A200A 0008C09D
>>> [ 489625] RATGDO: PACKET(0x1A200A @ 0x1B10F) UNKNOWN - Unknown: [09D]
>>> [ 489632] RATGDO: Support for UNKNOWN packet unimplemented. Ignoring.
```
So, can I interpret that as battery charging (0006) and then 82 seconds later, battery full (0008)

Still, not sure how we use this data.

jgstroud, 2024-04-04:
I dont know.  Maybe there is nothing to do here.

LilTrublMakr, 2024-04-07:
Person with a bad battery here. Lasts about 10 seconds without power (without using the motor). How can I help?

dkerr64, 2024-04-07:
> Person with a bad battery here. Lasts about 10 seconds without power (without using the motor). How can I help?

It would be interesting to see your log of all DECODED and PACKET where [09D] appear, as then we might be able to see if different values are returned when the battery is bad or failing.   But that currently requires a development environment (with your ratgdo connected to USB port).   In our next major release we may include a way to capture these logs into a browser console log.  When that is released, then it should be possible for you to capture those log lines quite easily.

Thanks.

LilTrublMakr, 2024-04-07:
Is there a way that I can send logs to a syslog server?

jgstroud, 2024-04-07:
Definitely no way to ssh to this device.  I've experimented with logging to a syslog server, and the proof of concept was promising, but it was never fully integrated.  You have 3 options available for capturing logs
1. connect a usb cable and capture the serial logs
2. build the code from source integrating PR #145 which allows viewing the logs through a browser
3. wait for the 1.3 build that will pull in #145.  probably sometime in the next week or so

LilTrublMakr, 2024-04-07:
Just realizing that I am running the esphome version and not the homekit version... But I can temporarily swap over in the name of science. I'll have time next weekend to play around so I will wait to see if the viewing logs from browser comes in.

Although, in my current web logs, I am seeing the following:
```
14:19:33 | [D] | [ratgdo:300] | Battery state=CHARGING
14:19:58 | [D] | [ratgdo:300] | Battery state=FULL
```
Probably doesn't help too much but it seems like we should be able to see if the battery is charging or full at the very least.

jgstroud, 2024-04-07:
Yeah, so these are just the same 2 commands we know about.  And only reports charging for 25s.  So probably nothing more we can do here.

jgstroud, 2024-04-30:
I don't think there is enough valuable status we can read from the GDO to make this a useful feature addition.

closed 2024-04-30 by jgstroud

jgstroud, 2024-06-19:
I just noticed my wall panel shows a battery charge level on the display.  there must be a way to see how charged it is.  I suspect the only way to get the various levels would be to discharge the battery and watch the messages while it charges back up