ratgdo/homekit-ratgdo #141: [Feature Request] Allow garage door to be designated as a switch or door

taisontan, 2024-04-03:
I have a gate and a garage for my home and only need CarPlay to suggest the gate for me to open when I get close. However, CarPlay’s Siri suggestion will sometimes suggest the garage. There’s no way for me to tell Siri which one to always suggest. It’s not consistent. 

Therefore I’d like to designate the garage with RATGDO as a door or switch so that Siri does not suggest it. 

Is this possible?

dmzimmerman, 2024-04-03:
Without saying anything about whether it's possible or not, I will say that I've had a similar problem with my two garage doors (I only want Siri to suggest one of them). I've found that "training" it, by manually telling Siri to open the one I want to open, a few times has resulted in Siri learning to always suggest that one (and that has stuck until a change in configuration, like when I switched from the MQTT to the ESPHome firmware, at which point I had to train it again because the HomeKit entities changed). Can't guarantee that'll work for you but it might be worth a try.

dkerr64, 2024-04-03:
I really don't think we should do this.  In HomeKit a garage door is a very different thing from a regular door which is different again from a switch.  Each have different states/characteristics.  I'm not saying it's impossible, but it would be a lot of work and, I think, unwise.

We have had a hard enough time getting what we have to fit within the constraints of the ESP8266 and to be stable and reliable.  Introducing additional complexity is unwise.  We are a garage door opener accessory, not a general purpose accessory.

jgstroud, 2024-04-30:
No plans to implement this.  You are certainly welcome to modify the code.  There may be other ways to do this with the home bridge plugin and dummy switches.

closed 2024-04-30 by jgstroud