ratgdo/homekit-ratgdo #144: No Response - Apple HomeKit

coryabooth, 2024-04-03:
I am having a strange issue....
Recently I noticed that my RATGD was unresponsive "No Response" in Apple Home.

I tried rebooting - no help
I reflashed firmware - no help
-----
I extended my Unifi AP by adding a 2.4Hz only Access Point, erased and reflashed the board
I was able to connect the board to the new AP
I was able to add the accessory to HomeKit - no problems
But IMMEDIATELY after the device registered and loaded into Apple Home - it went "No Response"
---------------------
I am able to do anything an everything you might expect from the web URL
It functions from the web URL

But Apple Home - despite adding it successfully as a new device - immediately shows "No Response"

jgstroud, 2024-04-03:
What version of the firmware are you on?  Do you have a HK hub?  Are you sure they are on the same network?  Seems silly, but I've seen 2 reports recently where the hub was actually the problem.  In one case the device and the hub were on isolated subnets causing this issue.

jgstroud, 2024-04-03:
Also, is your webUI still alive when this happens?

coryabooth, 2024-04-04:
Yes the web UI was alive

I am on 1.1.0

I only have one network and thus all my devices are on the same network.

My HK Hub is a wired Apple TV

I did read you shoukd wait 24hrs to readd a device in HomeKit?  Im in the process of this and see what happens.

jgstroud, 2024-04-04:
You definitely do NOT need to wait 24 hours.  While testing the pairing process, I've repaired my devices every couple of minutes.  You can remove the device in HomeKIt, reset the homekit pairing in the webUI and repair it.  You might also want to reboot your Apple TV.  One person on discord said they had a similar problem and it came alive after rebooting their Apple TV.

coryabooth, 2024-04-04:
I have power pulled all my devices to include all the network appliances

I went ahead and readded the RATGD and same issue

[image]

[image]

Then when it refeshes to App

[image]

jgstroud, 2024-04-04:
It's odd that your door shows up as "Garage Door" and not "Garage Door XXXX".  Maybe you changed that though the first time around?  I'm a little stumped at the moment why you are seeing this.  Any chance you can grab a snapshot of the webUI as well?

coryabooth, 2024-04-04:
I renamed the garage door while adding.  I rename everything I have for simplicy...

[image]

jgstroud, 2024-04-04:
Everything looks good there.  It seems like your AppleTV Hub isn't seeing the mDNS messages.  You said:
>  extended my Unifi AP by adding a 2.4Hz only Access Point with its own name
I'm not exactly sure how you did this, but maybe your mDNS packets aren't reflecting properly.

You can try installing the Discovery app on your phone https://apps.apple.com/us/app/discovery-dns-sd-browser/id1381004916?mt=12
Look for _hap_tcp and see if you see your garage door device.

coryabooth, 2024-04-04:
Extending.  what I meant is...

In my garage I have a Ubiquity AP mounted in the ceiling.  But I have a total of six AP throughout the house.

I read some people saying there were problems with RATGD switching APs

So I added a second WiFi Network - called it xxx2. Added it only to the Garage AP and set it to only 2.4Hz.  

This gives me a single named SID with only 2.4Hz and I connected the RATGD to this.

All this said, this was only done because of the issues this past week - I never had issue before without this isolated SID

coryabooth, 2024-04-04:
Yes on the scanner

[image]

coryabooth, 2024-04-04:
SUCCESS!

So you made me rethink a few things....
Now mind you - I changed nothing on my setup and I've had this running fine for a while now.
This all JUST hit a week ago - and despite the normal power cycle - nothing - I mean nothing would get this device talking to Apple TV Hub...

But you mentioned mDNS - I know I set that - but I had to go double check
It was set, but I reviewed my other settings...

IGMP Snooping
[image]
After reading that closer - I thought, hmmmm....  So I shut it off.

then I checked some other settings and I found:
Multicast Control
[image]
After reading that, I disabled it...

Now mind you, these settings were on before without any issues....
But after turning them off...  I now have connection to HomeKit

coryabooth, 2024-04-04:
[image]

jgstroud, 2024-04-04:
Sounds like it was that multicast control bit.  The description definitely explains your issue.  I just checked and it's disabled on my network.  I think if you enable that, you would need to add the Mac address to the exceptions list.  Typical unifi fashion, the help text leaves me with more questions than answers.  Sounds like it allows the multicast/broadcast over WiFi which is why you could see it from your phone, but it would not relay it over the wired network, which is why your AppleTV could not see it.  But I assume you have other HK devices that work.  So were they somehow unblocked?  Does unify provide some default exception list?

closed 2024-04-04 by jgstroud

coryabooth, 2024-04-04:
I do have Roku and Ecobee directly connected to Homekit - interestingly I had one Roku (out of seven) reporting failure but it was last on my list.  

I have Honebridge and Scrypted connected as well.

I was battling a problem with Scrypted where cameras loading on iPhone Home when on WiFi - but once leaving or setting to LTE they failed to load.

This fixed that as well.

coryabooth, 2024-04-04:
Do you have IGMP Snooping on?

jgstroud, 2024-04-04:
I have run with it both on and off.  On in theory is better.  I've heard of it actually helping in some cases.

coryabooth, 2024-04-04:
Just to close the loop here...

This morning I tackled the Basement Roku device (which was also "No Response") earlier when I had an issue.
With IGMP Snooping turned on...  I could register the device and add it - but it would goto "No Response"
With IGMP Snooping turned off....  No problems...

Again, IGMP has always been on with me (I tended to use this as the starting bible: https://evanmccann.net/blog/2021/11/unifi-advanced-wi-fi-settings)

But I will forever remain off from now on...

adammodlin, 2024-05-15:
I was hoping we could reopen this? I'm having this exact issue and I have IGMP snooping off. I'm also having the additional behavior of the web interface not being readily accessible and regularly timing out.

adammodlin, 2024-05-15:
Also, I'm not seeing my ratgdo device in the Discovery app, either.