ratgdo/homekit-ratgdo #150: Add setting to adjust WiFi TX Power

jgstroud, 2024-04-04:
Similar to the 802.11B/G/N setting, we should also allow adjusting the transmit power.  Don't know what the default is, but the internet seems to says it defaults to max.

/**
 * set the output power of WiFi
 * @param dBm max: +20.5dBm  min: 0dBm
 */
void ESP8266WiFiGenericClass::setOutputPower(float dBm) {

WiFi.setOutputPower(0); // this sets wifi to lowest power
WiFi.setOutputPower(20.5); // this sets wifi to highest power

see also
https://gitlab.com/painlessMesh/painlessMesh/-/issues/51

dkerr64, 2024-04-05:
Queued up in PR #145

Implemented as a range slider in the GUI with possible values from 0 to 20.  If 20 is selected then when setting power it is "rounded up" to 20.5.

jgstroud, 2024-04-30:
Fixed in #145

closed 2024-04-30 by jgstroud