ratgdo/homekit-ratgdo #154: Security+ 1.0 and Motion Sensor

drewcovi, 2024-04-07:
Not sure if I'm doing something wrong... should motion be reporting in the web ui? I've rebooted a few times, all other sensors seem to be reporting fine. I have the liftmaster 8500 so a bit of a hybrid in the security protocol space (or so I've read).

jgstroud, 2024-04-07:
I don't think the motion sensor is currently supported with sec+1. Probably easy to add with the correct opcode

drewcovi, 2024-04-07:
Yeah I thought I saw in the matrix that it's supported in mqtt and others, so it seems like it must be possible, just a small update. I'll cross reference and see if it's straight forward.

jgstroud, 2024-04-07:
I only have a sec+2 door, so I haven't looked at sec+1 at all yet. I suspect the person that added our sec+1 support doesn't have a motion sensor.  It should be easy to add though

PaulWieland, 2024-04-08:
Secplus 1 does not support motion sensing. 
The other firmwares report motion when the light is turned on since the motion sensor triggers the gdo light.

drewcovi, 2024-04-08:
So it would just be a matter of doing the same here then... triggering both on recognition of the light. although given they're somewhat interchangeable in regard to automation it's not likely a major showstopper for most.

Honestly not sure how I feel about it. Given the 8500 specifically externalizes the light assembly, and I previously disassembled that light unit to trigger a relay (and by extension turn on all garage ceiling lights), I'm personally treating it just like a motion sensor.

So yeah unless it was a configurable option in the webui to pick one or the other it's a bit redundant.

Either way absolutely love this project and thank you for the hard work here. Love the 2.53i and I do hope it's been rewarding. You've productized this in a way quinled has done and built an army of enthusiasts.

dkerr64, 2024-04-08:
So, we have two choices... remove motion from both the web page and also as a sensor in HomeKit when Sec+ 1 is selected as protocol, or fake it by reporting motion when the light is on.

I'm not sure the second is the same thing.  Light can be on for 10's of minutes, that does not mean that motion is taking place that whole time.

Personally I would choose to remove motion sensor accessory from HomeKit and the web page.

PaulWieland, 2024-04-08:
The way the motion sensors work is they send a motion detected message. That’s it. There’s no time out, no clear, etc. it’s just an event that says hey, motion was detected. 

So it’s useful for triggering automations, but not useful for showing a status.

dkerr64, 2024-04-08:
I think @PaulWieland is saying that it just does a notification, so there is never anything to clear.  So I now understand why using light turning on is a reasonable approach.

bkolaci, 2024-06-05:
I just got ratgdo installed and works great.  I noticed that the motion sensor no longer turns on the light nor shows up on the panel.  But I read that looking at the light is an approach, however my light never turns on (since installation).  Is it still supposed to?  I thought @PaulWieland said it wasn't supported.

jgstroud, 2024-06-05:
I can only comment on Sec+2 devices as that's the only setup I have.  With Sec+2 GDO, the motion sensor should still turn on your light and should report through homekit.  You light should still work as it did before and you should be able to control it from HK.

dkerr64, 2024-08-10:
@bkolaci please test with v1.7.0 (tagged as pre-release so select that checkbox on the firmware update dialog)

bkolaci, 2024-08-10:
Sorry, I actually replaced the whole brand new opener with latest security now and updated ratgdo to the latest.  It works great with the new opener, no issues.  The motion sensor works properly too.

dkerr64, 2024-08-10:
Will assume we're working in v1.7

closed 2024-08-10 by dkerr64