ratgdo/homekit-ratgdo #155: Omit obstruction when door is down

drewcovi, 2024-04-07:
Seems like a quick conditional could block this announcement. In Homekit and in the Home app, triggering an obstruction while the door is down causes the door to show as open temporarily. Technically its correct, but it doesn't have any consequence when the door is closed (door will open just fine even if obstructed).

https://github.com/ratgdo/homekit-ratgdo/blob/d08fdee8df518237020d2352529b98919a0b55ca/src/ratgdo.cpp#L136

jgstroud, 2024-04-08:
That's interesting.  The door is still reported as closed and HK still says it is closed but the icon shows an open door.  I've never actually noticed that before. To me this is a bug in HomeKit but I suppose we can change it in firmware.

jgstroud, 2024-04-08:
The eve app handles this correctly. I'm undecided if we should change anything or not.

jgstroud, 2024-04-08:
after giving this a bit more thought, its a harmless change and makes the experience better.  I'll include a patch for this in the next release.

drewcovi, 2024-04-08:
Sweet, and yeah I'm an experience designer and js dev by trade so I totally see both sides of this one. For me it came back to whether or not the garage is really obstructed when closed or opening. 

Truthfully it's not obstructed unless open or closing. And the sensor hardware could very well have been disabled when the door is down without impacting their purpose. 

Classic case of the garage sending TMI, which is cool, but maybe not useful.

drewcovi, 2024-04-08:
That's clever AF. Ever have any false positives?

dkerr64, 2024-04-08:
> I use the obstruction status so I can open the door with my foot after 2 consecutive breaks. That’s probably beyond what most hk firmware users will do though.

So, is that like me kicking my feet under the rear of my car and the trunk pops open for me?

Do you implement this in ratgdo firmware, or is it a HomeKit (or ESPhome) automation?

BTW... I immediately pictured a rodent scurrying around inside my garage and inadvertantly opening the door!

PaulWieland, 2024-04-08:
100% of my automations and even hk devices themselves are handled by NodeRed.

PaulWieland, 2024-04-08:
> That's clever AF. Ever have any false positives?

No, I programmed the timing so it has to be in a time window and there has to be presence detected in the garage.

drewcovi, 2024-04-08:
> That’s probably beyond what most hk firmware users will do though.

I think this part is probably spot on. I think power users have mqtt and esphome, while this integration is more turnkey and simplistic.

jgstroud, 2024-04-08:
So, @PaulWieland would agree with my original assessment that the obstruction should be reported regardless of door state.  My opinion is bouncing back and forth on this one.  Its just a HK cosmetic bug we would be working around.

drewcovi, 2024-04-08:
Is it possible to make it configurable? I guess I still see this as a report that's inaccurate as a true obstruction, based on the garage behavior.

jgstroud, 2024-04-08:
I dont think that this is something we should make configurable.  I also don't think it's inaccurate.  It's correctly reporting the status of the obstructions sensors.  That part is correct.   Homekit still has the door state as closed and shows the obstruction status.  The only thing I see that is inaccurate is the icon/animation that homekit shows.

drewcovi, 2024-04-08:
Fair enough. They could very well fix the app and provide those details without altering the UI. It definitely got me thinking about "what is an obstruction" and in some ways it doesn't seem like it's obstructing **door travel**. It is obstructing the laser, so just a different take on what the intent is.

And I was just looking to replicate behavior the garage seems to draw from, which is to say it doesn't actually do anything unless the door is closing.

jgstroud, 2024-04-08:
Consider a scenario where you check the status and it shows no obstruction, so you remotely open the door for someone and then find that it is obstructed and cannot close it without manual intervention.

drewcovi, 2024-04-08:
Well sure, but couldn't they just move it then? 😉

Totally get it though. Happy to close this one if there's a sense that Apple will fix it. Seems like when I've been working on homebridge work, my effort is often to mimic the behavior of the physical device, and the typical consumer here might be more likely to be confused until Apple gets around to finding a better solve in their ui.

closed 2024-04-08 by drewcovi