ratgdo/homekit-ratgdo #156: ToDo:  Add NTPclient to the firmware

dkerr64, 2024-04-08:
For a future release...

I'd like to add [NTPclient](https://github.com/arduino-libraries/NTPClient) to our firmware.

While there are probably many things it is helpful to know the current time for, what I would like is to log the time of most recent door activity (open or close) to report that in the web page and have that logged time survive a reboot.  The library looks pretty lightweight.

jgstroud, 2024-04-08:
I see the benefit here and it's something I have considered as well, but I'm hesitant to add more network stuff at the moment.

dkerr64, 2024-04-08:
Yes.  Let’s get the next release out before trying this.

jgstroud, 2024-04-30:
Decided not to implement this.   We can reopen if we change our minds.

closed 2024-04-30 by jgstroud