ratgdo/homekit-ratgdo #158: Unable to sync to button

jonathanjacob, 2024-04-20:
I was running the mqtt version (using ratgdo 2.0) and wanted to try the homekit version instead. I erased/flashed successfully. I am able to see the door status on the website, but it I select any options(light/door close/open) I see "Collision detected, waiting to send packet" in the logs. How do I force a resync of the rolling code? I tried pressing and holding learn for 6 seconds, I also tried to click the wall button 3 times (when its flashing). Nothing seems to do it.

Thanks

jgstroud, 2024-04-21:
Did you rewire it in the process?  The collision detected is referring to the serial line.  If you are always seeing this, then it thinks something is driving the serial line.  This can also happen when the serial line is disconnected.  Maybe the wire was pulled out.

jgstroud, 2024-04-21:
Ok. Well if you want to give it another try I can try to help figure out what the problem is.

closed 2024-06-28 by jgstroud