ratgdo/homekit-ratgdo #159: Security+ 1.0 state not syncing

crockpotveggies, 2024-04-24:
Purple square learn button. Ran MQTT firmware for 3 months stable. Switched to HomeKit firmware. Haven't rewired.

After switching door protocol in settings, the only function I'm able to perform is close the garage door. However the state remains on. Additionally, the light switch is not updating and remains off and I'm unable to control it.

I don't see any prompts to view logs, so I assume there's no crash? How do I debug this?

[image]

[image]

jgstroud, 2024-04-24:
So this is a security+ 1.0 door.  maybe @mitchjs knows what might be going on?

jgstroud, 2024-04-24:
Also, in the current build, the logs are only available in the serial console.

crockpotveggies, 2024-04-25:
@jgstroud is there a link or wiki page that describes how to set up the serial console?

mitchjs, 2024-04-25:
@crockpotveggies, Do you have a wall panel?
if not it has to go in to emulation mode.... not as well tested, as 99% of the people got a wall panel
(and not a plain old push button)

100ideas, 2024-06-13:
Just shared my similar issues with console logs over at https://github.com/ratgdo/homekit-ratgdo/issues/132#issuecomment-2165326762

mitchjs, 2024-06-26:
good news, i found the issue (my fault)... i will fix with a PR soon

dkerr64, 2024-08-05:
Fixed in version 1.6.1

closed 2024-08-05 by dkerr64