ratgdo/homekit-ratgdo #161: Motion sensor stopped working after upgrading from 0.8.0 to 1.2.1 (Sec+ 2.0)

robertlacroix, 2024-04-27:
I upgrade from 0.8.0 to 1.2.1 and since then motion sensor stopped working — the web UI shows 

> Motion:	false

no matter if it's triggered or not. It was working fine on 0.8.0. My GDO is using Sec+ 2.0

jgstroud, 2024-04-30:
In 0.10.0 we changed the code for the motion sensor.  It's no longer automatically added so users that do not have one never see the device.  Now, the motion sensor characteristic should be automatically added immediately after it first detects motion.  That attribute is forever saved.  If you were detecting motion before, the attribute really should be getting enabled.  Any chance you can retrieve serial logs?  If not, hopefully I'll have a new version released soon that lets you retrieve the logs over the network.

jgstroud, 2024-06-17:
This may be fixed in #204.  Several characteristics were failing to update properly

jgstroud, 2024-06-28:
I just realized user reported WebUI is not working either.  So this is not the issue that was fixed in the HomeKit library.  Mostly likely cause is that the ratgdo is never seeing motion detected messages. If we never see the message, there is nothing we can do.  Next course of action would be to use the viewlog.sh script to capture device logs while triggering the motion sensor.

dkerr64, 2024-08-10:
@robertlacroix please test with version 1.7 and advise if still a problem.  Release is currently tagged as "pre-release" so you will need to select that checkbox in the update firmware dialog.

dkerr64, 2025-09-13:
Please test with pre-release v2.0.0 (select pre-release box on update dialog).  Closing this issue, please open a new issue if problems found.

closed 2025-09-13 by dkerr64